LoupHC/controleur-CAPE #4: 'class Rollup' has no member named 'setStageParameters'

palexp2, 2018-01-05:
Copie du message d'erreur

```
/home/pa/Documents/controleur-CAPE/Greenhouse_prototype_CAPE_v1.1/Greenhouse_prototype_CAPE_v1.1.ino: In function 'void setup()':
Greenhouse_prototype_CAPE_v1.1:155: error: 'class Rollup' has no member named 'setStageParameters'
   rollup1.setStageParameters(1,R1_S1_MOD, R1_S1_TARGET);
           ^
Greenhouse_prototype_CAPE_v1.1:156: error: 'class Rollup' has no member named 'setStageParameters'
   rollup1.setStageParameters(2,R1_S2_MOD, R1_S2_TARGET);
           ^
Greenhouse_prototype_CAPE_v1.1:157: error: 'class Rollup' has no member named 'setStageParameters'
   rollup1.setStageParameters(3,R1_S3_MOD, R1_S3_TARGET);
           ^
Greenhouse_prototype_CAPE_v1.1:158: error: 'class Rollup' has no member named 'setStageParameters'
   rollup1.setStageParameters(4,R1_S4_MOD, R1_S4_TARGET);
           ^
Greenhouse_prototype_CAPE_v1.1:162: error: 'class Rollup' has no member named 'setStageParameters'
   rollup2.setStageParameters(1,R2_S1_MOD, R2_S1_TARGET);
           ^
Greenhouse_prototype_CAPE_v1.1:163: error: 'class Rollup' has no member named 'setStageParameters'
   rollup2.setStageParameters(2,R2_S2_MOD, R2_S2_TARGET);
           ^
Greenhouse_prototype_CAPE_v1.1:164: error: 'class Rollup' has no member named 'setStageParameters'
   rollup2.setStageParameters(3,R2_S3_MOD, R2_S3_TARGET);
           ^
Greenhouse_prototype_CAPE_v1.1:165: error: 'class Rollup' has no member named 'setStageParameters'
   rollup2.setStageParameters(4,R2_S4_MOD, R2_S4_TARGET);
           ^
exit status 1
'class Rollup' has no member named 'setStageParameters'
```

LoupHC, 2018-01-06:
Oups! Oublié d'updaté la librairie greenhouseLib!

LoupHC, 2018-01-06:
J'ai updaté la librairie, normalement ça devrait compile!

closed 2018-08-06 by LoupHC