LoupHC/controleur-CAPE #6: Décision current sensor SM ou sur connecteur

LoupHC, 2018-01-07:
Pour le montage des sondes de courant, ça me gossait que la seule chip existante soit surface mount, mais j'ai trouvé dernièrement ce socket, qui pourrait supprimer le issue de comment remplacer un sensor brisé.
https://www.aliexpress.com/item/SOP16-TO-DIP16-SOP16-turn-DIP16-SOIC16-to-DIP16-IC-socket-Programmer-adapter-Socket-150mil/1107288077.html?spm=2114.search0104.3.2.gGPdb3&ws_ab_test=searchweb0_0,searchweb201602_5_10152_10151_10065_10130_10068_10344_10547_10342_10343_10340_10548_10341_10084_10083_10139_10307_10059_10314_10534_100031_10604_10103_10142,searchweb201603_25,ppcSwitch_5&algo_expid=7ae05777-db5d-442c-b6c0-5fc18b6fb81e-0&algo_pvid=7ae05777-db5d-442c-b6c0-5fc18b6fb81e&priceBeautifyAB=0
La chip s'enlève en appuyant sur les ressorts, comme ça :
https://www.youtube.com/watch?v=XixbJe8b-Qw
L'enjeu c'est qu'il y a aucun datasheet donc j'ai aucune idée des ratings de courant, mais surement que ça le fait pour 1A.
Aucune idée à quel point le contact est solide non plus...
Une autre possibilité c'est de mettre un limiteur de courant avec la chip, genre fusible réarmable, pour éviter de buster les ratings du capteur avec un faux contact.

jayserf, 2018-01-17:
J'ai vu passer le courriel, je l'ai tu repondu?

c'est vraiment pour la programmation et connection temporaire. Je le mettrais pas dans une installation permanente exterieur.
Je crois que perdre un chip et avoir a le dessouder surface mount pour le remplacer est moins une probleme que le socket peut avoir. (meme si les contacts sont en or)

deux questions:
1. est-ce que ca risque reelement de mourrir ce chip? (j'en ai aucune idee)
2. est-ce que si ca brule metons, le courant passe quand meme mais tu n'a juste pas le lecture de courant, ou le moteur sera mort?

LoupHC, 2018-01-17:
Bon, après relecture du datasheet, l'isolation du capteur peut handle jusqu'à 420V de tension DC/des spikes de 2400V, je pense pas que ça soit un risque trop trop. 
L'autre risque c'est si le supply voltage excède 6V, avec le crowbar circuit ça devrait empêcher ça d'arriver.
Ok fuck les connecteurs à ressort. Je vais juste apprendre à souder du SM. :)

closed 2018-01-17 by LoupHC

jayserf, 2018-01-18:
Cool! OK d'accord