LoupHC/controleur-CAPE #14: Ajouter copyright et frame Creative Commons BY-SA au schematics du relay driver

LoupHC, 2018-01-08:


yoangp, 2018-01-09:
Autre chose à rajouter? J'upload la cette version des fichiers KiCad à l'instant.

[image]

LoupHC, 2018-01-09:
C'est cute au boute!
Pour les schematics, est-ce que c'est possible d'avoir un frame comme ça (avec la mention de licence)?
[Weather Shield_V12.pdf](https://github.com/LoupHC/controleur-CAPE/files/1615730/Weather.Shield_V12.pdf)

LoupHC, 2018-02-25:
Salut Yoan!
J'ai fait des logos CAPÉ et creative commons pour ajouter sur nos pcbs, je peux également les ajouter sur le pcb relais.

yoangp, 2018-02-27:
Salut Loup,

Gêne toi pas si tu veux ajouter cela sur le PCB des relais! :-)

Yoan

closed 2018-03-23 by LoupHC