LoupHC/controleur-CAPE #15: Décaller les trous d'adapteurs DIN du relay driver légèrement vers la droite

LoupHC, 2018-01-08:


yoangp, 2018-01-09:
J'ai du refaire le pcb pour corriger mon problème de routage. Pour l'instant les pièces sont placés approxiativement dans le fichier .kicad_pcb.
Est-ce que tu veux juste dire d'alligner les trous pour que tout soit centré ou tu veux que les trous soient plus à droite pour laisser plus d'espace dans le boitier?

yoangp, 2018-01-09:
J'ai regardé les dimensions des sockets de relai DIN, ils font 78mm de long selon la datasheet que j'ai trouvé. Pour rester à peu prêt dans la même largeur, est-ce qu'il serait préférable de tourner les circuits de drivers de 90 degrés en gardant les mêmes dimensions de 70mm x 100mm? Je n'aurais qu'a repositionner les pièces du PCB avec cette nouvelle orientation.

[image]

LoupHC, 2018-01-09:
Ouen pas très clair mon affaire désolé... Je voulais surtout dire décaler le PCB vers le centre pour dégager de l'espace entre la paroi du boitier et les terminaux à vis. 
Effectivement on peut aussi faire comme ça! Par contre si c'est possible  j'orienterais les terminaux à vis paralèlement aux connecteurs pour diminuer le stress sur les câbles.

LoupHC, 2018-01-09:
Par rapport aux sockets, j'ai trouvé la mention suivante sur Aliexpress : 
"Note:
Please allow 1-3mm error due to manual measurement. pls make sure you do not mind before you bid."
À considérer peut-être pour la dimension des trous.

yoangp, 2018-01-16:
1. J'ai repensé la surface du PCB de driver de relai, 8cm x 8cm semble être plus intéressant. Les sockets de relai DIN rail font 7,8cm de large donc on garde senseiblement la même largeur tout en prenant un peut moins de place en longueur sur la rail. Avec des pièces de même largeur sur la rail on va très probablement pouvoir faire un montage plus compact dans la boîte.

2. Pour les connecteurs pluggables, j'aimerais utiliser une version avec les fils qui partent vers le haut (exemple: https://www.digikey.ca/product-detail/en/phoenix-contact/1755736/277-1150-ND/260518). Je pense que cela va permettre de libérer un peut de surface dans la boîte et éviter de stresser les fils. Voici un dessin pour mieux comprendre:
[image]

3. Pour l'erreur possible sur les trous du support pour DIN rail, on peux faire des trous un peu plus gros. J'ai aussi trouvé une pièce équivalenete sur Digikey et les dimensions étaient bien 25mm centre à centre.

Qu'est ce que vous en pensez?

LoupHC, 2018-01-16:
J'aime bien ça arrangé de même! 
Et bon point pour les connecteurs, notez qu'on a non seulement le choix de l'orientation du connecteur mâle soudé, mais aussi du connecteur femelle!