LoupHC/controleur-CAPE #19: Partie relais avec ou sans PCB?

yoangp, 2018-01-09:
J'ai mis à jour le BOM pour la partie relai en 2 sections, une sans PCB l'autre avec PCB (voir feuille "Budjet détaill - boite relais pas étanche" dans https://docs.google.com/spreadsheets/d/1WUM-zMhos570SmZMH1RuJZH5EpLDn4kmKrJe-_xwwi0/edit#gid=1792563555)

En gros, le PCB fait monter le coût vraiment beaucoup. Je n'ai pas optimisé tous les coûts, mais ça semble tourner autour de 4000$ sans PCB (200$ par boîte) et 5500$ avec PCB (275$ par boîte). Ça me semble un gros défi de trouver une façon de garder les coût autour de 200$ par boîte avec un PCB.

Je trouverais ça plate de laisser tomber ce PCB, mais c'est pas si grave que cela non plus s'il faut rester avec la version 100% pièces sur DIN rail.

yoangp, 2018-01-09:
Après révision de mon erreur de coût de fabrication de PCB, j'ai recalculé le coût dans le budget et ça fait du sens avec PCB. Je suis à 205$ par boîte pour l'instant et surement encore capable de descendre un peu ce coût. Donc on reste avec l'option PCB.

closed 2018-01-09 by yoangp

LoupHC, 2018-01-09:
Nice!
Si ça te va je vais updater tes chiffres dans cette version du BOM : 
https://docs.google.com/spreadsheets/d/1U0wZ03mb_SrtCEw6ppra8F-SvSYpekVIvqOV3GDfxJs/edit#gid=0
C'est la plus à jour pour le contrôleur et ça va impliquer quelques modifications à peine pour le boitier relais. I'm on it!

Loup

yoangp, 2018-01-09:
Pas de trouble, tu peux bouger mes chiffres dans ce nouveau BOM, c'est lui que je vais utiliser aussi à partir de maintenant.