LoupHC/controleur-CAPE #20: Alimentation dans ou hors de la boîte?

yoangp, 2018-01-09:
Si on met l'alimentation 24V et 5V hors de la boîte on peut avoir une plus petite boîte. Est-ce que ça vaut la peine pour réduire les coûts. J'aime bien l'idée de Jason d'acheter un power supply 24V fermé, et de faire entrer dans la boîte juste du 24V. C'est plus safe que jouer avec des fils 120V et il me semble que Jason disait qu'il y a aussi des problèmes de règlemenation lorsque qu'il y a des branchements 120V.

LoupHC, 2018-01-09:
Ouen, l'intérêt de faire rentrer le 120V dans la boîte c'était de pouvoir éventuellement actionner des contacteurs avec les deux relais de spare. Je pense que le issue s'appliquait surtout aux assurances mais maintenant on sait que les serres sont jamais assurées... 
Ça fait aussi moins de filage à gérer à l'extérieur de la boîte je trouve. Un seul cable à pluger dans le 120V merci bonsoir.
Au niveau de la boîte, je vais essayer d'évaluer les économies possibles sachant que sans les PS on a quand même besoin au minimum d'une boite 30cmx30cm.

jayserf, 2018-01-17:
OK decision =

**AC/DC 24V 6A Waterproof externe**
https://www.trcelectronics.com/View/Mean-Well/CLG-150-24A.shtml

**https://www.trcelectronics.com/View/Mean-Well/ELG-150-24.shtml**
(je ne sais pas pourquoi lui est quand meme moitie prix... wtf is classified location use?)

https://www.trcelectronics.com/View/Mean-Well/HLG-150H-24.shtml

**AC/DC 5V 1A din rail (ou non) interne:**

din rail (meanwell real model)

Non Din model (half price, IP44 = water spray from any direction, vs no rating against humidity for din model)

**https://www.trcelectronics.com/View/Mean-Well/APV-16-5.shtml**

J'ai personellement (au travail) utiliser les APV et les HLG/CLG dans nos equipments et ils ne semblent pas lacher souvent, si jamais

closed 2018-01-18 by LoupHC