LoupHC/controleur-CAPE #21: Alimentation du contrôleur

LoupHC, 2018-01-11:
On s'était dit qu'on voulait skipper le voltage regulator intégré vu qu'il chauffe, est peu efficace énergétiquement, et peu fiable selon certains témoignages.

J'ai fait des recherches concernant l'alimentation du arduino via la pin 5V, je me rend compte qu'il y a au moins un enjeux : on fait tomber une certaine protection contre les survoltage. Selon les specs de la chip atmega2560, son operating voltage est compris entre 2,8-5,5v, et elle tolère des peaks jusqu'à 6v. En haut de ça on l'endommage.

J'ai fait le tour de pas mal d'options hier, voici le mieux auquel j'arrive à penser : 
**Protection contre le sur-voltage (crowbar circuit, j'aimer vraiment le nom) :** 
- le principe : en haut d'un certain voltage, le circuit s'auto-shortcut et fait sauter une fuse. 
https://en.wikipedia.org/wiki/Crowbar_(circuit)

C'est simple de chez simple! Sauf que... si on set le circuit pour s'activer à 5,6V mettons, ça risque de clamper tout le temps vu que l'écart entre notre load voltage et cutoff voltage est de 0.6V, et on est pas plus avancé si on doit changer la fuse tout le temps.
Alors... c'est pas une si bonne solution sauf... 

**Régulateur de voltage (LM2576-5):**
...si on intègre le circuit de protection avant un autre régulateur de voltage, plus efficace et moins tous les défauts de celui du arduino.  J'ai sélectionné le LM2576-5 parcequ'il nécessite pas trop de pièces externe, et permet d'alimenter le arduino entre 7-24V, avec un maximum input de 40V. C'est un régulateur à découpage, donc normalement ça chauffe pas, et c'est plus efficace (80ish% de conversion).
https://hackaday.com/2016/06/04/how-does-a-buck-converter-work-anyway/

Là dessus on peut intégrer un crow bar circuit à l'entrée du régulateur pour le protéger des grosses sur-tensions, ex, contact accidentel avec le 120VAC, et un après juste pour être sûr si, exemple, une goutte d'eau vient toucher le VIN et le VOUT.

Ça ressemblerait donc à ça : 
[image]

LoupHC, 2018-01-11:
Vous en pensez quoi?

LoupHC, 2018-01-11:
Parenthèse, le transistor en bas à droite, c'est pour éteindre manuellement les pull-ups et l'alimentation des sondes. Ça sera (peut-être) éventuellement utile si on adapte (un jour) le contrôleur pour fonctionner sur batteries.

yoangp, 2018-01-16:
Mon conseil Loup serait d'utiliser plutôt un régulateur 5V déjà tout fait. Je ne sais pas si tu as commencé la partie pcb, mais c'est pas de la petite bière. Il y a un tout un paquet d'affaires à vérifier et le moins que tu en as à faire, le mieux ça sera. Je sais que tu es super méga motivé et c'est très cool!! Mais tu va te sauver bien du travail et des tracas dans les prochains mois si tu n'as pas à t'occuper de cette partie du circuit. Dans tous les projets que j'ai fait jusqu'à maintenant en tant qu'ingénieur, ce qui est récurant c'est de ne pas avoir assez de temps pour compléter le projet à la date prévue. Malgré tout, je respecterai évidemment ton choix si tu veux quand même un circuit de régulation 5V sur le pcb du contrôleur :-)

LoupHC, 2018-01-16:
Ça a du sens... bon même si j'ai eu vraiment beaucoup de fun à monter ce circuit hahah... Je vais y aller au plus simple, et éventuellement, quand on aura tout terminé parfaitement dans les temps et même quelques jours d'avance, je verrai si je m'avance. 
N'empêche, régulateur ou pas, qu'est-ce que tu penses d'intégrer un crowbar circuit? Ça serait poche de buster notre contrôleur parceque la foudre tombe pas loin de la serre (bon j'ai aucune idée si je dis n'importe quoi en ce moment).

LoupHC, 2018-01-25:
Je voulais juste avoir votre avis sur le circuit final : 
[image]
Protection : 
- En cas de survoltage (5,6V et +), la diode zener actionne le SCR et short le circuit, ce qui fait sauter la fuse. (Je l'ai testé à 12V, ça marche super bien!) Théoriquement le SCR peut tolérer 200V et 4A, et avec le fusible ça dure vraiment juste une fraction de seconde.
- En cas d'inversion de polarité, la diode 8 devient conductrice et short le circuit, ce qui fait sauter la fuse. Le rating de la diode est 3A, vs la fuse qui est 1.5A (la même que les moteurs!)
- Avec une limite de 1,5A, on peut penser à toutes les sondes qu'on pourrait vouloir ajouter, on se rendra jamais là je pense. Si on short une sonde, encore là, ça fait sauter la fuse.

Ce que je me demande : est-ce qu'un circuit comme ça peut supporter si on plug le 120v par accident dans le contrôleur (shit happen)? Est-ce que vous voyez d'autres sources de problème?

LoupHC, 2018-01-25:
Tant qu'à faire, voici les protections que j'ai mis pour l'alimentation des relais :
[image]

- En cas d'inversion de polarité : la diode schotky empêche le courant de circuler, faible drop de voltage, donc peu de chaleur dégagé, max 30v par contre.
- Fuses réamorçables pour chaque sortie de relais, au cas où quelque'un connecte le 24V direct sans passer par les relais. Après 500mA ça trip et empêche le courant de passer, à part un peu de leak acceptable par les transistors.

Ce que je me demande : j'ai pas de protection contre les survoltages, est-ce que je devrais, est-ce qu'il y a une manière facile de le faire? Surtout si ça partage la même alim que les moteurs. Je trouvais ça overkill d'inclure une autre fuse et un autre crowbar circuit, est-ce que j'ai raison?

LoupHC, 2018-01-28:
Finalement, sur driver de relais on a décidé de remplacer les fuses/canal individuelles par une seule fuse sur le common ground. Comme ça, si ya un court circuit quelque part, tout arrête et l'alarme part!

LoupHC, 2018-01-28:
Et tant qu'à ajouter une fuse, j'ajouterais une zener/varistor pour protéger les transistors contre les surtensions.

LoupHC, 2018-02-04:
Guys! Je reviens de ma révision de circuit. Le résultat est malade!
A fuckin' rugged circuit!
Pour l'alimentation 5v, j'ai remplacé complètement le crowbar circuit par un low-drop voltage regulator, lequel est protégé dans ses trippes contre les courts-circuits, les spikes de voltages et la surchauffe. Pour protéger plus complètement j'ai mis une varistance en amont pour éviter de toaster le régulateur avec des vraiment hauts voltages (même s'il peut prendre jusqu'à 35v). Ça permet en bonus d'alimenter le contrôleur avec n'importe quoi entre 5v et 18v... pas pire!
[image]

LoupHC, 2018-02-04:
Pour l'alimentation 24V, j'ai changé les transistors par d'autres vraiment beefs (60V input, 80v output, 1,5 AMPÈRE par sortie!). Ça rajoute des pièces mais là on est carrément indestructible. On calculait même qu'on pourrait contrôler un moteur direct sur le contrôleur.
Comme on utilise pas tant d'ampérage pour les relais, on s'est dit qu'on mettrait une fuse 1.5A sur le common ground comme ça si une entrée saute, tout arrête et ça fait partir l'alarme, sans abîmer les transistors.
Là aussi, varistor et diode pour protéger des spikes de tension du moteur, même si à 80v de tolérance on est vraiment safe. Ça protège aussi contre le 110VAC.
[image]

LoupHC, 2018-02-04:
Ah oui et il pourrait nous prêter un oscilloscope et un bench PS pour nos tests. :)

jayserf, 2018-02-06:
moi aussi je peut en preter du travail

LoupHC, 2018-02-06:
Dernière implémentation : 
- ajout d'un power resistor de 20-40 Ohm/15W sur le ground commun des transistors.
Comme ça en cas de short des relais, le courant est limité à (roughly) : 
24/20  = 1,2A max (tolère spikes jusqu'à 34V)
24/40 = 0,6A max (tolère spikes jusqu'à 68V)
Et on a juste à mettre une fuse 0.5-1A en série pour éviter que ça chauffe longtemps.
En attente des pcbs pour tester ça!!!
[image]

yoangp, 2018-02-06:
Parfait, ça va aider pour les tests c'est certain!

Yoan

LoupHC, 2018-02-25:
[image]

closed 2018-02-25 by LoupHC