LoupHC/controleur-CAPE #22: Faire schematics du contrôleur sur KiCad

LoupHC, 2018-01-12:


LoupHC, 2018-01-12:
Je crois que j'ai réussi à tout mettre ensemble pour le schematic du contrôleur! Jettez un oeil : 
[image]

LoupHC, 2018-01-14:
Version updatée : 
[image]

yoangp, 2018-01-16:
Cool! Méchante grosse job!
-Tu a changé la PIN COM du ULN2803 du GND au +24V ? Je suis à peut près certain que ça ne marchera pas. Il faut qu'on rediscute de cette chip de toute façon, car après réflexion je vois plusieurs problèmes.
1. Toutes les sorties du ULN2803 sont connectées ensembles sur la pin COM, il n'est donc pas possible d'interrompre le lien du COM avec une switch pour passer en mode manuel car plus rien ne va fonctionner (fan, chauffage, etc.). De plus, avec 2 module de contrôle des côtés, il ne sera pas possible d'être en mode manuel juste d'un côté car la switch manuel-automatique du deusième module va continuer de maintenir un lien sur la pin COM. Il y a donc 2 options à première vue: Modifier le circuit du driver de relai ou modifier le circuit du contrôleur.

*Option modification du circuit du driver de relai*
La PIN COM de l'ULN2803 reste toujours connectée au GND. En mode manuel, la switch doit couper le signal OPEN et le signal CLOSE. Ceci est possible avec une switch 2P4T (2 pôles, 4 positions) 1 pôle pour OPEN, un pôle pour CLOSE, et 4 positions: OPEN-CLOSE-OFF-AUTO. Mais c'est difficile de trouver une switch comme cela. Il n'y a pas mille modèles comme une simple switch 1 pôle 3 positions. L'autre problème c'est que ce n'est pas compatible avec le design de la boîte de 2016. Mais de toute façon, on ne peut pas rendre notre montage compatible au montage de 2016 (le signal de contrôle est +12V). Le montage de 2016 fonctionne juste sur une sortie de mini-relai (comme à peu prêt 100% des contrôleurs de serre). Je me dit qu'il n'y aura probablement pas trop de monde qui veulent le contrôleur Arduino avec la boîte de relai de 2016. Et s'il y a quelqu'un avec ce problème, eh bien nous ferons quelques modifs à sa boîte relai 2016 pour la rendre compatible.

*Option modif du circuit du contrôleur*
Il faudrait rempacer le ULN2803 par des transistors Darlington NPN individuels. Ainsi chacune des 8 sorties sera 100% indépendante. Cela rend aussi le contrôleur plus facilement compatible avec d'autres boîtes de relai existantes (commerciales et celles de 2016).  Je pense que cette solution est plus souhaitable car elle est plus universelle.

Bon, je sais pas si c'était tout clair.. On s'en reparle au téléphone.

LoupHC, 2018-01-16:
En fait, sur le pinout de ULN2803, GND correspond au ground commun de tous les transistors, et COM connecte à un circuit de flyback diodes, qui doit être connecté à une tension pour protéger les transistors. J'ai fait quelques tests sur un prototype soudé chez nous et connecter COM au ground faisait que toutes les sorties étaient actives tout le temps. 

Pour bypasser le circuit de transistors, est-ce qu'on pourrait faire quelque chose comme ça tu penses?
On aurait besoin d'une switch DPDT et d'une switch SP3T on-off-on par côté ouvrant.
[image]

LoupHC, 2018-01-16:
Le boutte qui indique comment brancher la chip : 
[image]

jayserf, 2018-01-17:
WOW!

Je pensais pas que tu etais si avance! On est plus proche au proto finale
en gros:

- [ ] Finalise le BOM
- [ ] Change les pieces et finalise le schematique
- [ ] bouge tout a la bonne place
- [ ] change les footprints et trous et tailles de peices comme Yoan fait

Pour vrai, dependant du niveau d'apprentissage et de fierte de le faire sans aide tu veut avoir, j'en ai plusieurs amiss/collegues qui seront interesses a nous aider. Ca demeure une option pour accelerer.

Peut importe (si vous sont d'accords) je montrerais le pcb fini a un de ces gars pour un check d'erreurs de traces de debutant. (Peer review / fresh point of view) (peut nous sauver une revision de pcb!)

LoupHC, 2018-01-17:
Vu qu'on a commencé avec Kicad, est-ce que ça pose problème pour tes chummys? Est-ce que ça limite le niveau d'aide qu'ils pourraient nous apporter (exemple, retouches directement sur le fichier Kicad vs commentaires à l'oral)?
Dans tous les cas, c'est CLAIR que je suis down pour une révision!

yoangp, 2018-01-17:
Même chose pour moi Jason, c'est super si tes collègues peuvent faire une révision des fichiers kicad et du prototype de PCB du driver de relai.

jayserf, 2018-01-30:
Hey Guys,

J'ai faite une revision aujourd'hui avec un collegue au travail.

1. En gros, majeure commentaire: faut ajouter une colonne "PCB REFERENCE" dans les 2 boms pour mettre sur chaque ligne tous les places sur le PCB ou la piece se trouve.
ex: 10ohm resistor, R3,R7,R10,R11,R12 | 1uf Cap, C2,C7,C8 

comme ca n'importe qui qui assemble (ici ou en chine) est capable de bien faire correspondre les pieces du BOM avec sa place sur le PCB. (J'ai carrement oublie ce detail)

2. Il parait que les schematiques sur GutHub ne correspondent pas avec les Gerbers exportes que je l'ai montrer. Est-ce que c'est possible de remettre les plus recents et si oui est-ce qu'ils sont a jour avec le pcb (autogeneres par kikad) ou sont fait manuellement alors pas a jour encore car c'est long?

Alors sans les schematiques correspondants on a arrete le review apres 45 mins.

Sinon pas de flags majeures a premiere regard surtout si le Motor Driver est teste et confirme deja. Mais avec des schematiques a jour ca sera pas long a faire le tour complet du controlleur (et n'importe quelle questionnement plus technique je vais vous demander)

closed 2018-02-06 by LoupHC