LoupHC/controleur-CAPE #24: Faire routing du contrôleur sur KiCad

LoupHC, 2018-01-14:
Ça y est, on rendu là, faire le ménage de tout ce beau bordel :

LoupHC, 2018-01-14:
[image]

LoupHC, 2018-01-22:
Ça commence à ressembler à qqchose!
[image]

jayserf, 2018-01-22:
Fuck yeah! En 3d en plus 🚀

LoupHC, 2018-01-23:
Yeah, encore plus maintenant!!
[image]
[image]

yoangp, 2018-01-24:
Coooooooooool!!! :-)

LoupHC, 2018-01-24:
Guys! I made it!!!!!!!
[image]

yoangp, 2018-01-24:
Woooowwww! T'as même mis un ground plane! C'est la classe! ;-) Je vais essayer de jetter un coup d'oeil plus approfondi un peu plus tard. Quand prévoies-tu d'envoyer la commande de pcb?

En passant, quand j'ai fait ma commande de pcb, j'ai choisi revètement blanc avec texte noir et pas de plomb dans le revètement. C'était plus cher de prendre sans plomb, mais sans plomb blanc était moins cher que sans plomb vert (!?).

LoupHC, 2018-01-24:
Je vais réviser 2-3 trucs avant de l'envoyer, réviser toutes les footprints et les largeurs de traces.
Est-ce que Jason tu pourras montrer ça à tes chummys pour avoir leurs commentaires? J'update le fichier Kicad à jour live. 
Je pourrais envoyer ça peut-être d'ici le début de la semaine prochaine? Vendredi si je suis efficace.
Hey pis sont swell en tah les pcb que t'as monté!!!

LoupHC, 2018-01-26:
Et juste au moment ou je pensais avoir fini.... noooo! J'ai utilisé des connecteur 3.5mm au lieu de 5mm. 
Ça va être un peu plus long que prévu. :P

yoangp, 2018-01-26:
5.08mm ... ;-)

Moi aussi j'ai fait quelques erreurs comme cela.

Ciao, je te répond aux autres messages ce soir si tout va bien.
Vraiment cool que notre objectif d'inscription soit atteint!

Ciao!

Yoan

LoupHC, 2018-01-27:
Et pour finir en beauté, je me suis permis une petite coqueterie : 
[image]

yoangp, 2018-01-27:
> final](https://user-images.githubusercontent.com/24371752/35468160-9e8b9bd0-02e6-11e8-9818-1ecbe6f9d2c4.png)

C'est trop cute! :)

jayserf, 2018-01-28:
> 24371752/35468160-9e8b9bd0-02e6-11e8-9818-1ecbe6f9d2c4.png)
> issuecomment-360955863

Super nice!
Really like the drawing but that's secondary to the routing!
I'll try to get feedback on Monday so we can send it super quickly

jayserf, 2018-01-30:
Ignore comments re: SPI connector in other issue. this is good. 
OK now is much clearer to see the board with the real markings and connectors.

J'imagine c'est bon au niveau trous de montage?

LoupHC, 2018-02-06:
Version finale (on attend les prototypes!) : 
[image]

closed 2018-02-06 by LoupHC

LoupHC, 2018-02-25:
Version finale (après révision) : 
[image]
[image]
[image]