LoupHC/controleur-CAPE #26: Preparer doc pour soumission de PCBA

jayserf, 2018-01-17:
Needed info

Quantities: high & low estimate for **protos** and for **production** orders 
(can put in fab info text file)
PCB size (LxW in mm) (can put in fab info text file)
Boms (one file for each PCB)
Fab info:  (see below example from Eagle production pcb)

fabrication_parameters.txt
-------------------------
* 2 layers FR4 
* Finish board thickness 0.7874mm(0.031")
* TOP and BOTTOM copper thickness (1.0oz)
* White solder mask on both side
* Black silkscreen on both side
* HASL finish
* tolerance on drills (3mils)
* minimal drill size (12mils)
* minimal isolation (5mils)
* minimal trace width (5mils)
* PTH vias
* Manufacturer code accepted on PCB
* Tab-Routed panel

CAD Package
-----------
CadSoft EAGLE 5.12 Professionnal 

Gerber files format
-------------------------------------------------
* RS-274X format (Embedded D-Codes)
* 2.4 digits english
* All files are gerber-X format, except drills (.DRILL)
* origin at 0,0

Excellon files format (*.DRILL)
------------------------------------
* Excellon format (Embedded Drill Rack)
* Dimensions 2.4 digits OMIT LEADING ZEROS

included files
---------------
*.DRILL : 	drill file
*.GTL : 	copper layer TOP
*.GBL :  	copper layer BOTTOM
*.GTO : 	silkscreen layer TOP
*.GBO :		silkscreen layer BOTTOM
*.GTS : 	soldermask layer TOP
*.GBS :		soldermask layer BOTTOM
*.GTP :		stencil layer TOP
*.OUT :		board outline layer
*.GD1 :		drill symbol layer
*.MOUNT :	pick and place coordinates
*.png	:	extra pcbs information

LoupHC, 2018-01-18:
BOM controller components complet!
https://docs.google.com/spreadsheets/d/1U0wZ03mb_SrtCEw6ppra8F-SvSYpekVIvqOV3GDfxJs/edit#gid=0

LoupHC, 2018-01-18:
J'ai ajouté des détails sur celui des motor drivers aussi.

jayserf, 2018-01-18:
NIce!

Let me know if you need help with any of the fab stuff

jayserf, 2018-01-22:
> Ok là je pense qu'on a toutes les infos pour avoir une idée du prix (pour
> le PCB du contrôleur).
> Est-ce que t'aurais le temps de faire la soumission cette semaine?
>
> Dimensions : 4.2'' x 6.5''
> Proto : pas plus que 5
> Production : 20?
> TH/PCB : 90
> SMT/PCB : 34
>
> BOM : https://docs.google.com/spreadsheets/d/1ETJ_
> UvwiGVeKPeL4h7JOTw4AD7blGYYILlhRYPq2OK8/edit#gid=0
>
> Pour le layout du PCB, j'ai presque fini de placer les composantes, je me
> donne jusqu'à la fin de la semaine avant de voir si j'ai besoin de plus
> d'aide qu'un checkup final. Ça se pourrait, c'est long en titi placer tout
> ça!!

Oui bien sûr je peut envoyer la soumission cet semaine.
Pas aujourd'hui car c'est ma fête mais demain for sure

yoangp, 2018-01-23:
Bonne fête Jason!!

:-)

Yoan

LoupHC, 2018-01-23:
Bonne fêteee!!!!!!

jayserf, 2018-01-26:
thanks guys! I had a good one.

Quotes sent to PCBWay and More-Suns. We'll see what they return and I have some that I know well but definitely charge setup fees so dont want to msg them if we will for sure not use them as they dont specialize in lower volume, but would do a flawless job.

closed 2018-01-29 by LoupHC