LoupHC/controleur-CAPE #27: SPI channel  : logic level shifter??

LoupHC, 2018-01-18:
En lisant sur internet, je me rend compte qu'on peut pas directement plugger les pins de communications à n'importe quelle chip, particulièrement les chips 3.3v (le module LoRa ou ESP8266 nottament). Pour ces modules là, on doit utiliser un Logic Level shifter, qui ajuste le voltage d'un module à l'autre. Détails ici : 
https://learn.sparkfun.com/tutorials/bi-directional-logic-level-converter-hookup-guide
Certains modules tolèrent les signaux de 5v (comme la nrf24 que Jason utilise) mais dans tous les cas, c'est préférable. 

On pourrait se dire qu'on connecte le LLS à l'extérieur du contrôleur, mais ça impliquerait d'ajouter une 9e pin pour le canal SPI, pour avoir une sortie 3.3V et une sortie 5v. J'aimerais mieux pas, pour que ça rentre dans un cable cat-5.

À la place, je propose plutôt d'intégrer ce circuit à notre PCB, qui est ultra simple : 
https://cdn.sparkfun.com/datasheets/BreakoutBoards/Logic_Level_Bidirectional.pdf
Je réutiliserais exactement les mêmes composantes SMD que le module de sparkfun, pour pas risquer de manquer un paramètre critique.

jayserf, 2018-01-18:
damn, the radios I use NRF24 are 3.3 but the logic does not need to be shifted, so i've been using 3.3V regulator adapters and the signal stays as direct connections to the arduinos.

I actually thought esp and lora radios were 5 so I wasn't asking for 3.3...
(I have a bunch of those LLS boards but i dont use them)

ok to add the parts I say if we can more easily use most common radio modules.

but why the CAT5 cable?

we can connect the radio with an 8 pin header on the board. the box is plastic and worse case you buy an SMA antenna and keep it in the box. or make a hole and use an extension wire. (Especially LoRa wont have a problem, but even NRF works really well.) 

If I understand you have a cat5 cable to connect a small radio board outside the main box?

LoupHC, 2018-01-28:
Finalement j'ai juste copié le pcb de Sparkfun, ya pour 3$ de pièces au total. 

Pour le cat5 j'imaginais que ça serait intéressant si on connecte plusieurs modules sur le channel SPI, ça permet d'avoir les modules wireless en dehors de la boîte et de garder un layout simple au lieu d'avoir plein de cables lousses dans le boitier.
C'est mon côté un peu maniaque et rangé qui s'exprime, mais anyway là je rajoute pas de connecteur extra, ça va tout décaller et j'ai déjà tout refait les traces!

Mais j'avais oublié qu'on voulait mettre un connecteur pour plugger direct une radio comme t'aimes! Je vais essayer de faire fitter ça, ça pourrait je pense!

jayserf, 2018-01-30:
Ah OK, good pour le PCB de sparkfun.

Merci pour le connecteur si possible, sinon c'est des fils mais juste une petite radio n'est pas si pire. Pour le range ca se colle sur un mur de la boite, le platique derange pas trop les radios si l'antenne st tenu un peu loins des PCBs.

Sorry I wasnt more helpful on the LLS debate

closed 2018-02-06 by LoupHC