LoupHC/controleur-CAPE #28: Nouvelle pub sur le RJME

yoangp, 2018-01-19:
Salut!

Loup, est-ce que tu peux nous dire quand tu pense que ça peut se clarifier le lieu de l'atelier et la subvention pour les formations?

Éléments à penser:
-Établir concrêtement comment des gens puissent commander sans être présents.
-Ça serait le fun de montrer une photo de comment ça avance, mais s'il fait attendre le premier prototype monté, ça me semble trop long. Il y a toujours les fichiers de pcb, mais c'est juste bon à exciter les geeks ces affaires là..
-Expliquer comment va se passer la fin de semaine (hébergement, bouffe, etc.)
-Intégration de la vente de moteurs..
-Autres..

LoupHC, 2018-01-19:
Ça gosse, ça a été délégué au nouveau dg de la CAPÉ, et ils ont l'air d'être en branle-bas de combat en ce moment. Pour l'instant je dirais que ça va avoir lieu "aux environs de Montréal" vu que les deux options c'est Repentigny et Saint-Jean-sur-Richelieu.

LoupHC, 2018-01-19:
Est-ce que ça fait le tour des infos à transmettre?

Un petit rappel pour dire qu'il reste une semaine pour vous inscrire à l'atelier d'autoconstruction d'automates de serre.

Date limite d'inscription : 26 janvier
Date de l'atelier : 14-15 avril
Lieu : à déterminer (aux environs de Montréal)

De l'hébergement sera disponible sur place, et un système de remplacement sera mis en place pour ceux.celles qui ne pourraient pas être présents, prioritairement à ceux.celles qui viennent des régions éloignées.

Inclus dans le programme :
- Boîte relais avec contrôle indépendant des rollups et overrides manuels
- Contrôleur pour le contrôle des roll-ups et 4 sorties supplémentaires pour le contrôle des fournaises et de la ventilation
- Possibilité d'intégrer une alarme en cas de coupure de courant ou de surchauffe
- Initiation à l'électromécanique, troubleshooting électrique, l'électronique et la programmation.

Et un scoop avec ça :
Le comité d'autoconstruction s'associe avec l'entreprise Orisha pour offrir une interface  de contrôle et de monitoring du contrôleur sur le cloud. Le service sera mis en place dès cet été.
Coût de l'abonnement : 12-25$/mois, peu importe le nombre de serres connectées

Pour s'inscrire : https://www.fermierdefamille.com/fr/capeautoconstruction/catalog/72/
Infos sur le contrôleur : https://github.com/LoupHC/controleur-CAPE/wiki
Écrire à Loup pour suivre les échanges du contrôleur, via liste de courriel dédiée.

LoupHC, 2018-01-19:
Ah ouais pis : Amenez votre lunch!

LoupHC, 2018-01-19:
Pour les photos, c'est vrai que c'est pas évident de trouver une bonne photo qui donne le goût... 
Je pense que je me fierais aux envolées lyriques de Yan pour le côté catchy...

yoangp, 2018-01-19:
-Ok pour le lieu, au moins ça donne une bonne idée.
-Comme tu le mentionnait à Reid, je penses effectivement qu'on pourrais intégrer la vente des moteurs sur la même page que les ateliers, je trouve que c'est clean.
-Est-ce qu'on peut contacter la liste des bénévoles inscrits pour leur demander s'ils sont prêts à faire le montage pour d'autres gens pas présents?
-Amenez 4 ou 5 lunchs...? Est-ce qu'on s'imagine que la plupart des gens vont rester coucher? Il y aura alors au moins diner, souper, déjeuner, diner. Est-ce qu'on fait cela style potluck vu que ça fait bien des lunchs à préparer sinon?

yoangp, 2018-01-19:
Dire que ça pourrait être connecté au Cloud cet été, c'est pas un peu tôt? Si on dit cet automne, ça donne un peu plus de temps et j'ai l'impression qu'il n'y  pas grand monde qui va avoir le temps de mettre cela en place au milieu de l'été. Qu'en penses-tu Loup?

LoupHC, 2018-01-19:
J'écris aux bénévoles, en attendant j'ai déjà deux amis qui m'ont dit qu'ils seraient ouverts à un échange, donc on peu en parler avant d'avoir la confirmation de tout le monde.

Pour le potluck, je sais pas, c'est quand même de la gestion, un potluck. Et on sait pas encore l'espace cuisine disponible, si c'est à Cadet Roussel ça se peut sans doute, mais chez Mario, on aurait sans doute juste une petite cuisinière. Comme c'est sur une ferme on peut dire qu'il y aura une chambre froide pour conserver la bouffe?

Pour le cloud ils m'ont dit qu'ils allaient se mettre là-dessus pour rendre ça disponible dès cet été. Mais la formulation "au courant de l'été" est peut-être à privilégier.

LoupHC, 2018-01-20:
All set : ça sera chez Cadet Roussel.
Et je rajoute : formule potluck pour les repas.

yoangp, 2018-01-20:
Cool!! Great job Loup!

jayserf, 2018-01-20:
Wow, beaucoup d'information vous avez rassembler! Nice job!

yoangp, 2018-01-22:
Concernant les gens qui veulent acheter un kit, mais qui ne peuvent pas venir:
-Est-ce qu'on limite cette option à ceux qui sont vraiment trop loin (car on a un nombre limité de bénévoles pour nous aider)? Ou bien, on met simplement cette option dans le catalogue selon le nombre de bénévoles qui veulent s'impliquer et c'est premier inscrit premier servi.
-Qu'est-ce qu'on fait avec les coûts de formation? On pourrait filmer l'événement (grâce à un bénévole) et mettre cela en ligne par la suite. Autre idée?

closed 2018-02-06 by LoupHC