LoupHC/controleur-CAPE #30: Prototype driver de relai

yoangp, 2018-01-24:
J'ai fait un test avec le prototype de driver de relai et tout a l'air de fonctionner normalement avec un moteur de côté ouvrant branché dessus. Je n'ai à peu près pas de modifs à faire sur le pcb, sauf peut-être un trou dans le milieu du support de la fuse pour la solidité mécanique.

Loup, je monte le deuxième driver de relai et nous pourrons nous rencontrer en personne. Je vais regarder mon horaire, car j'ai un petit rush sur la ferme d'ici mercredi prochain. Ainsi, jeudi de la semaine prochaine serait bon pour moi. Est-ce trop tard pour toi? Si oui, je vais voir si je peux me libérer vers la fin de cette semaine.

En passant, ça m'a pris à peut près une heure à monter, donc x 2 pour les deux côté nécessaire à une boîte de relai. Évidemment, cela n'inclue pas le travail pour les connecteurs et fixer le matériel sur la boîte.

Ciao!

yoangp, 2018-01-24:
En passant Loup, est-ce que tu as vu que les drivers de relais sont maintenant 8cm de large par x 6cm de long. Ça économise encore un peu d'espace dans le boîtier.

LoupHC, 2018-01-25:
Nice non j'avais pas vu ça!!!
C'est vraiment excitant j'ai hâte de voir ça. Jeudi prochain moi ça me va. Sinon en fin de semaine je suis à Montréal, on pourrait se rencontrer là et ça permetterait à Jason d'être là aussi!
Je vais assembler mon circuit perfboard d'ici là, et envoyer les protos à imprimer en début de semaine prochaine. 
Ça t'a coûté combien le shipping recommandé et les prototypes? T'as reçu ça en combien de temps et t'en as commandé combien?

jayserf, 2018-01-26:
Cool Yoan! hate de voir ca aussi. Nice job pour la reduction de taille et PAS de revisions majeure!

Cet fin de semaine (demain) je suis au laurentides pour une fete. Vous parler de celui ci ou le 2-3-4fev?

LoupHC, 2018-01-29:
Faque tout est beau avec le motor driver, c'est clean, le form factor est parfait et ça fonctionne super bien.
Bravo Yoan!!!

closed 2018-01-29 by LoupHC

jayserf, 2018-01-29:
Bravo Yoan!