LoupHC/controleur-CAPE #31: Retour sur soumissions

LoupHC, 2018-01-29:
Faque on a reçu les deux soumissions pour faire imprimer/assembler les circuits. 
Soumission contrôleur : 
https://docs.google.com/spreadsheets/d/16Y3bVeKkrIgNWwjR_LlU5sF_WZEezp7sD9hzIpVireg/edit?usp=sharing
Soumission relais : https://docs.google.com/spreadsheets/d/1IUjBSyNJqgLbTRPqoz3maPh4fo51kFOpfDNz3UG8ut8/edit?usp=sharing

Je retiens 4 points critiques en checkant ça : 
1. On est limité pour les formats à 25 ou 50 circuits.
2. En fonction de ces chiffres-là, ça balance!!!  Et la CAPÉ dégage même un bénéfice de 0-3000$.
3. Le temps de fab est de 45 jours,

1. Par rapport aux quantités, je propose qu'on fixe le nombre maximal d'unités vendues à 25 boitiers relais et 25 contrôleurs. Comme ça on commande 25 contrôleurs (excluant les protos) et 50 motor drivers.  Les 3 tannants qui auront besoin de drivers relais extras (6) on pourra leur faire à partir des protos, ça en laisse 2 pour Yoan, 2 pour moi, et Jason aura ceux de sa ferme! Et on pourra leur vendre personellement ça va être moins compliqué que gérer ces détails-là via la CAPÉ. 
On est rendu à 21 vendus de chaque, et yen a deux autres réservés déjà, aucun soucis qu'on va réussir à tous les vendre d'après moi. 
Je pense que c'est risqué d'en commander le double, surtout qu'après 1 été il y aura peut-être des améliorations à apporter si jamais la CAPÉ veut refaire un atelier.

2. Pour le budget, j'ai calculé une grosse boîte pour tout le monde, pour simplifier l'assemblage.

3. Là on a un problème. Si c'est 1mois et demi fabriquer et shipper tout ça, à partir du 5 mars, en courrier recommandé, on arrive short en esti avec l'atelier. 
Est-ce que ça serait possible de leur demander de commander les pièces longues à shipper au moment de faire les protos, avant le nouvel-an Chinois, pour accélérer le processus quand on aura fait les révisions?

yoangp, 2018-01-30:
1. 100% d'accord d'y aller pour 25 contrôleurs, 50 driver de relai et 10 drivers de relais de plus montés à la main par nous avec les prototypes (dont 4 de côté pour Loup et moi).

2. Super, une grosse boîte pour tous.

3. Je suis d'accord qu'on a un problème. Une autre solution serait d'acheter nous même toutes les pièces dès que possible, les faire livrer au Canada, puis de tout envoyer en chine en courrier recommandé pour qu'ils aillent tout le nécessaire dès le retour au travail. C'est un peu stupide ce va et vient sur toute la planète, mais ça nous permettrait aussi d'acheter quelques pièces moins critique sur Aliexpress, moins cher que chez Digikey.

4. Jason, ma commande de PCB prototype a coûté 54$CA et j'ai du payer un extra de 17$ de frais de douane. 30% de frais de douane supplémentaire!! Est-ce que tu as une idée comment on peut éviter de payer 30% sur notre grosse commande?

jayserf, 2018-01-30:
1. OK mais on n'est pas limites en qte du tout, (sauf possiblement par multiple du panneau de pcb, x 2 x4 x6 etc) on peut faire 28 ou 30 ou 22, j'ai demander cela pour savoir autour de ces qtes la.
Je vais additionner ma commande perso aussi! pour 2-3 serres je crois.

2. OUIIII gross boite

3. Yoan ca marche pas malheureusement, l'import en chine est tough et il faut etre importateur et collaborer avec la compagnie destinataire pour que ca passe les douanes, souvent ca prends des semaines sauf s'ils sont habitues. MAIS je pose des questions ce soir pour reduire les delais si on commande les peices tout suite etc,. et aussi clarifier la 45 jours ca doit comprendre les vacances. le 5-8jrs d'achat pour les pieces est le plus long.

4. un peu mais le pcb assemblee n'est pas le meme code douane que pcb vierge. alors ca comprends des differents taxes. AUSSI c'est possible que 10$ sont des frais de service de douanes de DHL ou qqchose. faudra regarder la facture.
Je peut demander a une compagnie logistique avant de commander si on identifie differament ou qqs d'autre pour reduire la charge.

add 5. je trouve les frais d'assemblage tres haut je ne sais pas si c'est la qte ou le fait que c'est through hole alors plus manuel.

add 6. va falloir mettre une # de piece reference pour les les items 'generique' dans le bom.

add 7. pour reduire le temps de sourcing des composantes, on peut demander des alternatives ou les trouver nous memes, ils on deja suggere qqs unes. ca rentre en phase "bom optimization" pour reduire le temps de fab! (esperant ce n'est pas necessaire)

yoangp, 2018-01-30:
Autre idée pour sauver de temps. Loup, tu pourrais commander tes pcb de prototype de manière à ce que ça soit envoyé avant les vacances en chine. Peut importe si on a trouvé le bug de l'autre jour ou pas. On finira de trouver le bug pendant les vacances en chine. Au moins on aura quelque chose entre les main à tester. S'il faut faire quelques modifs du pcb de contrôleur, on devrait être assez certain que ça marche même sans faire de 2ième prototype.

yoangp, 2018-01-30:
Allo,

Est-ce que vous pensez qu'on pourrait lancer la fabrication finale du pcb de relai avant celle du contrôleur? Est-ce que ça fait une grosse différence de prix d'envoie? Au moins cette partie serait finalisée et ça serait ça de moins à se préoccuper.

LoupHC, 2018-01-30:
Ok, les inscriptions terminent le 3 février. Le 4 février on envoie les commandes finales de PCB moteurs et la commande de protos contrôleurs. Pour les PCBs moteurs, ils vont pouvoir commander les pièces manquantes avant le Nouvel-an et on devrait les avoir pas longtemps après qu'ils reviennent de congé. 
Pour le PCB contrôleur, les seules pièces qui demandent du shipping, ce sont les switchs TH, et les sensors de courant. 
Pour les switchs, est-ce qu'on peut s'informer s'ils ont des équivalents en stock et leur demander les dimensions? N'importe quelles switchs spdt TH feraient l'affaire.
Pour les sensors, est-ce qu'on peut s'informer s'ils ont ACS712 en stock? Sinon, si on leur confirme l'achat de 25 contrôleurs le 4 février, est-ce qu'ils peuvent commander les sensors tout de suite?

Par rapport au proto Contrôleur, j'ai commandé assez de pièces pour retester l'ancien prototype, figure out ce qu'on a fait de pas correct et éventuellement changer de stratégie. Ils m'ont appelé ce matin pour dire que c'était arrivé au Canada et ça devrait arriver aujourd'hui ou demain. 
Réalistement je peux arriver à qqchose d'ici le 4 février.

Je pense qu'on est dû pour se jaser au téléphone, jsais pas ce que vous en dites.

yoangp, 2018-01-30:
Bonne idée pour l'appel téléphonique. Je suis disponible en soirée,
n'importe quel jour.

closed 2018-02-12 by LoupHC