LoupHC/controleur-CAPE #32: Mise à jour du BOM

yoangp, 2018-01-30:
Allo!

Voici quelques détails sur la mise à jour du BOM:

1. Fuseholder: le numéro de pièece du manufacturier que tu avais mis Loup n'était pas le bon, c'était pour une fuse 20mm de long, ça nous prend 30mm de long avec le footprint que j'ai mis. J'ai pas trouvé de fuseholder 6mm x 30mm sur Aliexpress avec le même footprint que celui que j'ai acheté chez Digikey. Donc j'ai mis celui de Digikey.

2. Dans le feuilet de la boîte, si la switch "3PIN Waterproof Switch ON/OFF" c'est pour changer de ouvrir à fermer en mode manuel, ça devrait être une switch SPDT (ON-OFF-ON).

3. C'est pas évidant de faire des trous carrés pour les switchs que tu as choisis. Je propose: 
[Toggle-Switch-Screw-SP3T](https://fr.aliexpress.com/item/Toggle-Switch-3-Terminal-3Pin-ON-OFF-ON-15A-250V-Toggle-Switch-Screw-SP3T-Industrial/32817323925.html?spm=2114.search0104.3.1.32f278f5YOYFxk&ws_ab_test=searchweb0_0%2Csearchweb201602_1_10152_10151_10065_10344_10130_10068_10547_10342_10343_10340_10548_10341_10084_10083_10139_10615_10307_10313_10059_10314_10534_100031_10604_10103_10142%2Csearchweb201603_25%2CppcSwitch_5&algo_expid=64a902ed-9700-4ef7-9a1e-3428c4ee4cb4-0&algo_pvid=64a902ed-9700-4ef7-9a1e-3428c4ee4cb4&transAbTest=ae803_4&priceBeautifyAB=2) 
et
[On-Off-On-3-Position-DPDT-6-Screw-Terminals-Toggle-Switch](https://fr.aliexpress.com/item/On-Off-On-3-Position-DPDT-6-Screw-Terminals-Toggle-Switch-AC-250V-15A/32796952978.html?spm=a2g0w.search0204.3.13.d78c826y3pQ21&ws_ab_test=searchweb0_0,searchweb201602_1_10065_10344_10130_10068_10547_10342_10343_10340_10548_10341_10084_10083_10618_10139_10615_10307_10313_10059_10534_100031_10103_441_10624_442_10623_10622_10621_10620_10142,searchweb201603_2,ppcSwitch_4_ppcChannel&algo_expid=fad5d715-57c6-40c4-9c2f-74ae853cde3c-2&algo_pvid=fad5d715-57c6-40c4-9c2f-74ae853cde3c&priceBeautifyAB=5)
avec
[capote](https://fr.aliexpress.com/item/5-Pcs-Set-12MM-Toggle-Switch-Resistance-Cap-Cover-Rubber-Boot-Waterproof/32776791165.html?spm=a2g0w.search0204.3.145.74edb6e2bBLJOB&ws_ab_test=searchweb0_0,searchweb201602_1_10065_10344_10130_10068_10547_10342_10343_10340_10548_10341_10084_10083_10618_10139_10615_10307_10313_10059_10534_100031_10103_441_10624_442_10623_10622_10621_10620_10142,searchweb201603_2,ppcSwitch_4_ppcChannel&algo_expid=07d6410d-b808-47d1-b432-97056765a9a9-23&algo_pvid=07d6410d-b808-47d1-b432-97056765a9a9&priceBeautifyAB=5)

4. J'ai changer les plug-in screw terminals pour qu'ils soient droit vers le haut.

5. Il faut voir ce qu'on fait pour les fusibles, j'avais fait une étude théorique en 2016 qui me disait que c'était mieux slow blow, mais je comprend que ça serait bien toute le même fuse partout. Il faudrait essayer avec des fast blow pour voir si ça fait la job dans la réalité. Pour finir ma réflexion, il existe toute sorte de fuses et l'ampérage n'est qu'un paramètre, ça me rend un peu mal à l'aise d'acheter une fuse sur Aliexpress qui n'as pas toutes les caractéristiques voulues. Ça risque de nous apporter des surprises. Je propose qu'on y aille avec un fusible Digikey, même si c'est 0,4$ de plus par fusible.

jayserf, 2018-01-30:
Cool, merci d;avoir fait ca Yoan.

ET d'avoir incorporer la soumission de PCBWay en format excel, j'ai qqs questions mais pas tout suite.

1. OK good

2. pour les switchs OK cool, merci.

3. Yup Agreed.

4. nice

5. all good, j'aillais prendre d'une fournisseur qui liste tous les specs dans l'idee d'inclure plein de spares (genre 10) dans ch. kit, mais pas necessaire. 

Je suggere qu'on fait le tour du bom du boite wt commander une prototype pour la boite vers la fin de la semaine pour pouvoir montrer une vrai durant leur fermeture.

yoangp, 2018-02-01:
Je pense que j'ai fini avec le BOM, voici un résumé des changements:
-J'ai ajouté les références de pièces du pcb
-J'ai créé une nouvelle feuille afin de séparer les pièces à souder sur le pcb du driver de relai de celles qu'on a juste à plugger dessus (juste fusibles et relais).
-J'ai réviser les coût avec les quantités qu'on a à acheter
-J'ai mis des relai Aliexpress
-Anisi, le coût final des pcb complets et soudés tombe à environ 25$ par pcb. Je pense que ça a du bon sens.

Jason: Est-ce que tu peux donc envoyer le BOM à PCBWAY selon la feuille "PCB relai" stp? J'ai aussi uploadé le fichier KiCad à jour avec les fichiers Gerbers (https://github.com/LoupHC/controleur-CAPE/blob/master/KiCad%20Files/CAPE_Relay_schield.zip). C'est les mêmes fichiers que le prototype.

Loup: J'ai refait les calculs de prix des feuilles "PCB relais" et "PCB relais add". J'ai fait une partie de la mise à jour des sommes sur la feuille "Summary", mais je te laisse le reste pour la partie contrôleur.

yoangp, 2018-02-01:
UPDATE!:

J'ai supprimé la feuille "PCB relai add. Il y avait juste les fuses des drivers de moteurs (qui étaient déjà là dans la feuille "Circuit components") et les relais (qui étaient déjà là dans la feuille Main box du BOM).

Jason: Pour la commande de quicaillerie, oublie pas aussi les relais , j'ai changé ça dans la feuille Main box du BOM pour qu'on les achètent aussi en chine pour réduire les coûts. Loup, ton idée c'était 6 relais  total par boîte right (dont 4 pour les drivers de moteurs)?

Loup: Est-ce qu'on a les adapteurs [PCB à DIN rail](https://www.digikey.ca/product-detail/en/phoenix-contact/1201578/277-2296-ND/290934) dans la quicaillerie du BOM? Je vois cela nul part.

LoupHC, 2018-02-01:
Yep, 6 relais, dont un pour l'alarme. 
Et effectivement, je constate que les  adapteurs DIN rail sont pas inclus, oups.

J'ai juste un commentaire pour les switchs, j'avais eu une mauvaise surprise avec une switch très similaire à la switch On-off-on. Je pensais qu'elle gardait sa position mais en fait elle revenait toujours à Off quand on la lâchait. Garder une switch enfoncée pendant 2 minutes pour descendre un rollup, c'est long, faudrait vérifier si elle garde bien sa position toute seule. Sur la photo, ça a l'air par contre, la switch est visiblement à On et personne la tient. Mais je leur demanderais au cas.
Aussi, les connecteurs à vis sont pas très solides, ya presque pas de thread sur le boutte femelle, mais c'est pas trop grave si on crimp le fil. Mais sinon, avec un fil lousse, ça tient mal. 
Sinon elles font un click très satisfaisant et ont l'air overrall robustes. 
Et c'est une vraiment une bonne idée les capotes, est-ce qu'on est certain que c'est la bonne taille? 

De mon côté, UPS a encore chié et le colis est resté stallé à Montréal, je reçois ça demain mais demain je travaille. Bref, je vous donne des nouvelles vendredi pour le circuit.

jayserf, 2018-02-01:
Alors 6 relais = 3 x pcb relais par boîte?

Ok pour le switch. Non-momentary type.

Ok pour pcbway et merci pour le bom + références. Plus le cleanup des extra
feuilles. Plus la séparation des pièces électromécanique du bom pcb!

La seule redondance est les connecteurs ronds sur le boîtier, je dois
ajuster le bom des boîtes pour être la même que celui des câbles.
(nécessaire pour une validation de polarité)

Je ne vais pas oublier les relais!

Loup tu mets les DIN mount supports?

Too bad pour ups, that sucks!

Je pense c'est tout

jayserf, 2018-02-01:
Aussi, um, on est en train de vendre un kit de contrôle sans brains.

Pas de arduino inclus?

LoupHC, 2018-02-01:
Yep, arduino, rtc et autres sont dans la page Circuit components.
Pour les relais ya deux relais individuels sur des sockets din rail.

yoangp, 2018-02-01:
Loup, comment tu ajoutes la somme de la feuille "Circuit components" dans le summary du BOM?

Sinon pour les relais, ça demande un peu de clarification:

Option 1:
2 relais pour le driver de moteur du côté 1
2 relais pour le driver de moteur du côté 1
1 relais indépendant pour le chauffage
1 relais indépendant pour la ventillation ou l'alarme
Total = 6 relais

Option 2:
2 relais pour le driver de moteur du côté 1
2 relais pour le driver de moteur du côté 1
1 relais indépendant pour le chauffage
1 relais indépendant pour la ventillation
1 relais indépendant pour l'alarme
Total = 7 relais

Qu'est ce que vous en pensez? C'est sur que c'est plus le fun 7 relais, mais ça ajoute encore des coûts.

LoupHC, 2018-02-10:
Finalement j'ai merge Circuit components avec Main box pour clarifier. 
Pour le 7e relais à voir, j'aimerais mieux, mais on peut se garder ça pour la fin quand on aura les soumissions, voir si ça bust notre budget

closed 2018-02-10 by LoupHC

jayserf, 2018-02-13:
Agreed...

Dans nos serres le fournaise est un relais, ventilation intégrée au
fournaise  une autre et fans supplémentaires  au plafond une 3ieme...