LoupHC/controleur-CAPE #33: Plan de test

yoangp, 2018-01-30:
Identifier les tests à faire sur le driver de relai et le contrôleur.

À venir...

LoupHC, 2018-01-30:
Tests driver de relais : 
- capacité des transistors à s'actionner sans brûler avec un input de 24V
- comportement des transistors en cas d'inversion de polarité des relais
- comportement des transistors en cas de court-circuit
- comportement des transistors en cas de survoltage
- capacité des transistors à actionner 8 relais 24v en simultané
- tester fuse action rapide/action lente
- si ULN28003 est inadéquate, tester les quad darlingtons ULN2065, qui supportent 80v en input et 1,5A en output!!!! (4x le prix par contre : 6$(3,50$ vs 1,50$)

Tests contrôleurs : 
- compatibilité avec sondes
- tester communication SPI et sérielle
- protection survoltage/inversion de polarité/court-circuit

LoupHC, 2018-01-30:
J'ajouterais, tester si (worst case scenario) on branche 110V AC au input 24V ou 5v, quecé qui se passe.

yoangp, 2018-02-03:
Comment se passe les tests? Est-ce que tu as reçu les pièces finalement? Si tu veux qu'on s'en parles, on peut faire ça sur Hangout (Maike est partie avec le téléphone ce vendredi soir).

LoupHC, 2018-02-03:
On s'est manqué on dirait.
Oui, j'ai commencé à faire ça tantôt! À date, j'ai toujours pas compris ce qui s'est passé. 
1. Test des inputs :  j'ai appliqué 23v  à chaque entrée, sans charge entre émetteur/collecteur.
Résultat : aucun dommage
2.  Test des outputs : J'ai appliqué 23v à chaque entrée, et connecté un relais à chaque collecteur.
Résultat : aucun dommage
3. Test des diodes : J'ai fait le même montage avec les diodes sur les relais, dans la même direction que la dernière fois.
Résultat : aucun dommage
5. Test court-circuit : j'ai branché une sortie directement sur le 24v
Résultat : dommage très localisé, sur la sortie et celles d'à côté, pas de feu ou de bruit

Bref, je suis un peu mystifié. Je me demande si il y avait pas un short qqpart dans mes soudures, pourtant à l'oeil je vois rien.
En y réfléchissant bien j'ai pas testé l'inversion de polarité. Je vais aller essayer ça!

yoangp, 2018-02-03:
Donc si tu alimentes les 8 relais il se passe rien?
Essaye de faire clignoter les relais genre une seconde tous à ON, une seconde tous à OFF quelques minutes. Tu peux aussi essayer des mettre cumulativement à ON à tour de rôle à une seconde d'intervale. Juste pour tester la capiacité de faire des manoeuvres à répétition et sur plusieurs sorties sans que rien ne brûle.
Est-ce que tu peux aussi mesurer le courant total qui passe dans le 2803 avec ton multimètre?

LoupHC, 2018-02-03:
210mA , les 8 relais allumés (25-30 mA chaque).
J'ai essayé de les allumer à tour de rôle à chaque .5sec et tout est beau...
Je vais essayer le clignotement, mais à date, aucun problème en vue...

yoangp, 2018-02-03:
Ça a l'air que le voltage max sur les sorties du PS2501 est de -7 à 80V. Il faudrait voir si un spike lors de l'arrêt du moteur pourrait théoriquement générer une tension négative inférieure à -7V sur l'alimentation 24V..

J'ai regardé ton circuit de protection d'inversion de polarité sur l'alimentation 24V du contrôleur. Il semble que la diode schotky 1N5822 peut laisser passer un leek current de 2mA. Ça reste à voir si c'est suffisant pour créer une tension inférieure à -7V sur les sorties du PS2501.

LoupHC, 2018-02-07:
Finalement j'ai remplacé la schotky par une 1n4004. Pas vraiment de stress pour ce circuit au niveau du voltage drop et le leak current tombe à 30 uA.

LoupHC, 2018-02-07:
Pour jouer safe, j'ai refait le circuit avec les transistors plus beafs (ULN2075). On a maintenant un circuit qui peut largement supporter des spikes de 47v, jusqu'à 2A/chip et 1,5A/pin. Overkill! 
La fuse est pas assez rapide pour protéger des courts-circuits, la meilleure option que je vois c'est de mettre un power resistor en série pour limiter le courant, et rester comme ça largement en dessous des specs.
Je n'ai pas la résistance que ça me prendrait pour ce circuit mais j'ai essayé à plus bas voltage, short limité à environ 1,2A de courant et le transistor encaisse magnifiquement bien. On testera quand on aura les PCBs!

yoangp, 2018-02-07:
Parfait! J'ai bien hâte de voir ton pcb de contrôleur en vrai!

Tant mieux d'y aller avec des pièces overdesign, ça va nous protéger
un peu des imprévus. On est pas rendu à faire de l'optimisation fine.

Pour la résistance de puissance, je pense bien que ça peut faire
l'affaire, mais je ne suis pas sur que ça soit la stratégie
généralement utilisée pour limiter le courant. Sauf que là, je me dit
que le plus important c'est que ça marche. ;-)

Ciao!

Yoan

LoupHC, 2018-02-07:
Hey c'est compliqué les maudits circuits limiteurs de courant!
Le gros bémol de la resistance c'est que ça limite le potentiel d'utilisation des transistors... mais bon le jour ou quelqu'un voudra driver ses moteurs directs sur le contrôleur il aura juste à l'enlever et à se forcer pour pas shorter son circuit. :P.

LoupHC, 2018-02-07:
On va recevoir ça dans les temps! Lundi prochain si tout va bien!

jayserf, 2018-02-13:
Wow, j'ai complètement louper ce bout de conversation. Désolé mais j'ai pas
trop de Input sur les résistance de puissance!

Par contre testes fonctionnelles et qualité jai de l'expérience. Aussi
testes de boîtiers assemblés plus.

Idéalement une arduino qui fait actionner tous les outputs un par un et
aussi qui teste les sondes ou une setup de test au poste d'assemblage
finale.

A voir

LoupHC, 2018-02-14:
Test effectués : 
1. Allumage successif et simultané de 8 relais
- tension d'entrée : 24.5V
Observations : 
Ça clique sans problème. Lorsque les 8 relais sont allumés, la chute de tension aux bornes de la résistance 20 Ohm est de 4.6v, autrement dit, les relais fonctionne sur 19.6v. On reste en haut du 18v minimum d'opération d'après la datasheet.

2. Court-circuit des relais
- court-circuit d'un des relais
- fusible fast blow 500 mA
- test avec résistance 25Ohm/20W et 20Ohm/10w
Observations : 
Le court-circuit est réalisé en bypassant la bobine du relais.
J'avais d'abord fait le montage avec une résistance 25 ohm. Dans ce contexte-là, le court-circuit faisait chauffer le fusible mais pas suffisamment pour le faire fondre. La résistance dissipait cependant assez bien la chaleur.
J'ai changé la résistance pour une résistance 20Ohm. Quand j'ai refait le test, le fusible a brûlé après 3 secondes.
J'ai aussi réalisé que la résistance est collée au module d'horloge, qui dispose d'un capteur de température interne. Ce capteur pourrait servir de backup pour détecter les courts-circuits si le fusible ne lâche pas comme il devrait.

LoupHC, 2018-02-14:
3. Court-circuit des sondes
- jumper entre 5v et ground d'un connecteur de sonde
- laissé 5 secondes
Observations : 
Tout éteint (donc l'alarme part!). Le régulateur de voltage ne chauffe pas, (limite selon les specs le courant à 1.25A) les traces non plus. Après suppression du court-circuit, tout reprend normalement.

LoupHC, 2018-02-14:
4. 110VAC dans le input 24v
- input 24V : 
-varistor 39v 250A
- fuse 0.5A fast blow
Observation : 
La fuse fond instantanément.
Aucun dommage sur le circuit après avoir changé la fuse. 

5. 110VAC sur le input 5v
- varistor 27v 250A
- fuse 1.5a slow blow
Observation : 
La fuse fond dans un beau flash.
Destruction du varistor
Destruction du regulateur de voltage
Destruction du cap Vout du régulateur
Destruction du I/O expander
Destruction de la chip temp/humidité
Destruction du arduino
...FAIL

yoangp, 2018-02-15:
Salut Loup,

Je trouve que c'est déjà super que le 110V dans l'entrée 24V ait fonctionné.

Je pense qu'il n'y a pas trop de circuit commerciaux qui sont protégés
pour du 110V dans une entrée 5V...

En tout cas, beau test destructif!! Ça fait mal au coeur, mais on sait
à quoi s'attendre après.

Est-ce que tu as les pièces pour refaire ton circuit?

Ciao!

Yoan

jayserf, 2018-02-15:
> issuecomment-365783066

Ouch, coup de foudre! mais je suis d'accord avec Yoan,

deja que ;e circuit a du texte ecrit dessus pour les terminaux et qu'il
fait essayer fort (les connecteurs ne se branchent pas ensemble (male 3 pin
au male 4 pin) si on disait que le % de possibilite est de 10% et moins ou
qqchose similaire, personne sera dessus de pas avoir ce protection je pense.

LoupHC, 2018-02-15:
Non sti... pas de arduino de rechange.
J'en ai commandé un autre pour vendredi.
J'ai refait une simulation du circuit avec un transistor à la place du regulateur, et une fuse 1.5A fast blow, et cette fois, aucun dommage. Je l'ai même essayé 2 fois.
J'ai l'impression que le circuit a le potentiel de fonctionner si la fuse fond instantanément.
Peut-être que ça a un lien avec la capacitance qui absorbe une partie du choc jusqu'à saturation...
On s'en reparle vendredi!

LoupHC, 2018-02-16:
Pas de arduino de spare... mais j'en ai commandé un pour demain.

LoupHC, 2018-02-16:
La transplantation de cerveau s'est bien déroulée. Notre pauvre prototype est back on tracks.
Comme je pensais, la vitesse à laquelle la fuse fond joue significativement sur les chances de survie du circuit. 
J'ai refait environ 5 simulations de 120v dans le input 5v aujourd'hui, avec quelques variations dans le circuit. J'ai appris 2-3 petites choses au passage. 
D'abord, pour protéger convenablement notre circuit, il faut mettre une diode sur le Vin, mais aussi sur le ground. Dans le premier fail de mercredi, ce qui a tué le condensateur  d'après moi c'est qu'il s'est retrouvé avec sa charge statique de 5v sur sa borne positive, et 27v sur sa borne négative, donc -22v en pleine gueule. 
Ensuite, j'ai constaté que mettre un fusible action rapide permettait de sauver le circuit, mais que avec un fusible action lente ou attendre que le breaker lâche, résultait quand même à un bris de composantes, même si, théoriquement, il devrait survivre. Ya probablement une affaire de bruit là-dedans.

J'ai aussi testé avec une zener 25v, du côté protégé par les diodes, comme double protection du varistor. Je l'ai rajouté pour pouvoir utiliser un varistor plus gros capable d'opérer à 24v, parceque brancher 24v sur l'alimentation du contrôleur aurait pu fonctionner un certain temps, mais aurait dégradé le varistor sur le long terme. Avec la zener, si le voltage dépasse 25v, ça provoque un court-circuit qui fait sauter la fuse. Le bris survient donc sur une pièce enfoncée plutôt que soudée.
Dans un autre test, avec une fuse action lente, seulement la zener a brûlé, pas le régulateur de voltage! 
Comme la zener est moins critique que le régulateur, c'est une autre bonne raison qu'elle soit là je crois.

LoupHC, 2018-02-17:
Finalement, quand je met tout ça bout à bout, ça ressemble à ça, et  le résultat est très satisfaisant!!!
Testé 2 fois à 120VAC, protégé 2 fois.
[image]

yoangp, 2018-02-17:
Ouais, ça va être méchament protégé ton circuit! On va pouvoir brancher les
fils les yeux fermer sans avoir peur de faire sauter quelque chose ;-)

Yoan

LoupHC, 2018-03-01:
Finalement le triac sera remplacé par un SCR unidirectionnel. La diode anti inversion sera placée entre le Varistor et le SCR sur le GND.
Testé 3 fois avec Fuse 1.5A FB. Aucun dommage 3 fois.

LoupHC, 2018-03-03:
J'ai refait le test de courant des moteurs. Bonne nouvelle, la chip ACS723 fonctionne!!!
Et j'avais visiblement un bug avec mon multimêtre....
À vide, le moteur tire entre 0.3 et 0.5A... en oscillant (le flow de courant n'est pas constant)
En forçant sur une ratchet, j'ai réussi à monter ça à 1A.

J'ai aussi commencé à écrire du code pour tester les différentes sections du contrôleur.

yoangp, 2018-03-04:
Salu Loup,

C’est vraiment plus c à quoi je m’attendais les valeurs de courant que tu
viens de mesurer! Très cool que la chip pour le courant fonctionne!! :-)

Ciao,

Yoan

closed 2018-08-06 by LoupHC