LoupHC/controleur-CAPE #34: Sondage

yoangp, 2018-02-01:
Je vous as ajoutés comme collaborateurs au [formulaire du sondage](https://docs.google.com/forms/d/1WxO17ut5oeYTsKfFmCTYdPt4qe6uDdXuSNXFon4QN1Y/edit) . 

Loup, est-ce que tu veux jetter un dernier coup d'oeil avant qu'on l'envoie? Tu as accès en écriture au besoin. Est-ce que tu t'occupes de l'envoyer aux gens inscrits?

LoupHC, 2018-02-02:
Oui je m'en occuppe! Merci Yoan!

closed 2018-02-02 by LoupHC

LoupHC, 2018-02-02:
C'est envoyé!

LoupHC, 2018-02-02:
Je vais compiler ici les données du sondage au fur et à mesure que ça rentre : 
https://docs.google.com/spreadsheets/d/1j9LD_4gHe2kkoIPN0RChVHgnP19oJdYkSyDbvr8SMhM/edit#gid=0
En passant, on est sold out!!!! 30 boitiers relais et 25 contrôleurs commandés. Hoppa!!

yoangp, 2018-02-03:
Malade!! :-)