LoupHC/controleur-CAPE #35: STOP : ça balance pu

LoupHC, 2018-02-03:
Ok guys on l'a échappé. 
Ça balance plus. 
Faut trouver comment économiser minimum 70$ par boîtier.

LoupHC, 2018-02-03:
Pistes de solution (soumissions) : 
- budget connecteurs sans soudure : 2816$ à 5,60$/ch
De combien on peut baisser notre prix unitaire si on en commande 500?
- budget grosse boîte : 6300$ à 92$/ch
De combien on peut baisser notre prix unitaire si on en commande 34?
- budget power supply 24V : 2000$ à 65$/ch
De combien on peut baisser notre prix unitaire si on en commande 34?

Pistes de solution (coupures) : 
- acheter sonde température plutôt que temp/humidité : 400$ (-15$/sonde)
- supprimer un set de relais indiv./connecteurs : 320$ (-10$/boitier)
- acheter plus petites boîtes relais/contrôleurs seuls : 200$ (-20$/boite seule)
- assembler nous même les PCBs relais : 640$ (-20$/boite)
- mettre un PS 24v non scellé dans la boîte : 800$ (-25$/boite)

LoupHC, 2018-02-03:
Aussi :
- acheter 100 pcb relais (donc en stocker un peu) : (-10$/boîte)

jayserf, 2018-02-03:
Les connecteurs sont 5.60 pour le côté boîtier et côté fil, j'ai descendu
de 7.5 la paire ce matin.

Je crois que la boîte @ 90$ qui inclut le transport peu descendre pas mal
aussi.

Je suis chez ma mère et je vais possiblement être en train de manger à 8h.

Loup est ce qu'il y a juste 4 boites relais seulement?

Si oui est-ce qu'on peut enlever le 'format' qui inclut le transport et le
multiple pour les différentes 2 boîtes et les quantités commandes?

Faire tous les boms pour juste la boîte sans transport et fait la feuille
de calcul récapitulatif qui multipli la qte de chaque sous assemblage plus
transport.

Comme ça on élimine la possibilité d'erreurs de double calculs dans les
autres onglets

LoupHC, 2018-02-06:
Outcome : 
- réduction du prix de la boîte
- réduction du prix des PS
- réduction du prix des connecteurs

closed 2018-02-06 by LoupHC