LoupHC/controleur-CAPE #36: Défauts à corriger du proto contrôleur

LoupHC, 2018-02-14:
- Fusible F1 bloque le port USB du arduino
- Switchs 1-5 trop proches des connecteurs à vis
- Switchs 4-5 trop proches une de l'autre
- LED D16 bloquée par l'horloge
- Switchs 7-11 bloquent la chip Atmega du Arduino UNO (la chip du MEGA est ok, vu que surface mount)
- Switch P-MOSFET pour les allumer/éteindre les sondes = le circuit ne fonctionne pas, sort 2V seulement

jayserf, 2018-02-14:
Apart du p fet c'est juste du placement... Alors super!

Pourquoi on a ce circuit p fet? Pour protéger les Inputs de sondes?

LoupHC, 2018-02-14:
Pour pouvoir éteindre les sondes... javais mis ça au début, en me disant que ça pourrait permettre d'optimiser la consommation de courant du circuit. Je pense que je vais juste l'enlever.

LoupHC, 2018-02-14:
J'oubliais : 
- inversion de la position du 24V+/GND sur le contrôleur, pour être similaire avec le shield relais.

jayserf, 2018-02-15:
> Super! Bon travail Loup et Jason! Je vais essayer de vous écrire du
> feedback d'ici la fin de la semaine.

Awesome!

closed 2018-02-25 by LoupHC