LoupHC/controleur-CAPE #38: Préparation de l'atelier

yoangp, 2018-03-22:
Allo Loup et Jason!

Nous allons avoir pas mal de choses à préparer pour l'atelier, voici quelques éléments, à compléter..

1. Rencontre préparatoire à trois avant l'atelier, je propose le 7 avril à la Ferme Cadet-Roussel. Comme ça on va pouvoir voir ce qu'il y a sur place. (À confirmer avec Arnaud si c'est ok pour vous)

- Voir ou les gens vont pouvoir dormir, manger, est-ce qu'il y a assez d'espace de travail, besoin de chauffer une pièce de plus, etc.
- Nombre de tables requis et arrangement, place pour projecteur, prises électriques et rallonges 120V, 
- Révision de l'inventaire des pièces reçu et ce qu'on fait pour ce qui manque (Commande rapide Digikey, Addison, Abra, etc)
- Apporter déjà une partie de la commande de pièce si c'est trop pour entrer dans une auto en plus du reste du matériel pour l'atelier
- Vérifier l'accès à internet
- Organisation de la bouffe
- Plan de travail pour atelier efficace (chaine de montage, nombre de personnes par station, etc.)
- Horaire de l'atelier et temps alloué à chaque parties
- Plan du contenu des formations
- Avoir un prototype tel que sera construit
- Avoir un plan pour la modification des quelques boîtiers de 2016
- Penser et contacter quelqu'un pour présenter comment gérer le température des serres (Reid ou autre)
- Avoir une liste de tâches pour les gens assez bon en électronique pour nous appuyer lors de l'atelier (répondre aux questions, tester les montages, etc)
- Avoir un montage pour brancher aux relais les ventilateurs et une fournaise.

2. Matériel requis pour l'atelier

- Assez de tables
- Assez de chaises
- Projecteur
- 2 laptops
- Présentations offline sur clé USB
- Cables d'extension 120V
- Fers à souder
- Fils d'étain à souder
- Inventaire de pièces pour l'atelier
- Perceuses
- Mèches
- Cutters pour fils
- Pinces à dénuder fils
- Tournevis
- Multimètres
- Liste des participants pour les 2 jours (producteurs présents, bénévoles présents, bénévoles pour producteurs absents, etc.)
- Prototype assemblé pour faciliter les explications de montage

3. Message à envoyer aux participants une semaine avant l'atelier
- Rappel date, heure, lieu
- À apporter: Bouffe, outils de travail, tables, chaises, projecteurs, matelas et sac de couchage
- Paiement à apporter si pas fait avant?
- S'ils en ont déjà une pièce du montage, il serait mieux qu'ils l'apportent (leur contrôleur, leur boîte de relais de 2016 ou autre, leur moteurs si pas des 100W 24V)
- Demander qui parmis les participants a suffisament d'expérience pour répondre à des questions et tester les montages complets
- Apporter manuels d'installation et fiches techniques sur leurs ventilateurs et fournaises

LoupHC, 2018-04-10:
Pour m'amuser (et pour lâcher l'ordi 2 min) aujourd'hui j'ai fait deux plans de montage pour l'assemblage des sondes et du boitier d'interface (des éléments qu'on avait moins discuté ensemble).
Je pensais faire un plan similaire pour le code de couleur des fils dans les connecteurs du boitier

[image]
[image]
[image]

jayserf, 2018-04-11:
Great man!

Looks like da Vince sketchbook 😉

Très claire et efficace

yoangp, 2018-04-11:
C'est malade Loup! :-) Je travaille là-dessus cet après-midi et ce soir..

Ciao!

Yoan

closed 2018-08-06 by LoupHC