LoupHC/controleur-CAPE #39: Documents de formation pour l'atelier

yoangp, 2018-03-25:
Allo Loup et Jason!

Voici ce que j'ai jusqu'à maintenant, mais ce n'est pas final. J'ai refait les schémas dans Powerpoint, c'est pas mal semblable que Lucidchart et je n'ai pas de limitations..

[Formation Contrôleur de Serre V1.pdf](https://github.com/LoupHC/controleur-CAPE/files/1844964/Formation.Controleur.de.Serre.V1.pdf)
[Formation Contrôleur de Serre V1.pptx](https://github.com/LoupHC/controleur-CAPE/files/1844965/Formation.Controleur.de.Serre.V1.pptx)

La suite va avancer moins vite, il faut que je retourne à mes affaires de ferme!

LoupHC, 2018-03-25:
De mon côté j'ai mis à jour l'ordre du jour "officiel" de l'atelier, pour le 14 je me suis inspiré de ce que t'avais mis dans ton powerpoint : 
[Plan de cours - autoconstruction contrôlleur de serre.docx](https://github.com/LoupHC/controleur-CAPE/files/1845493/Plan.de.cours.-.autoconstruction.controlleur.de.serre.docx)
Vous me direz si vous trouvez que c'est un ordre logique

yoangp, 2018-03-30:
Allo!
Nouvelle version, suffisament correct d'ici à notre meeting la fin de semaine du 7 avril:

[Formation Contrôleur de Serre V2.pptx](https://github.com/LoupHC/controleur-CAPE/files/1864466/Formation.Controleur.de.Serre.V2.pptx)
[Formation Contrôleur de Serre V2.pdf](https://github.com/LoupHC/controleur-CAPE/files/1864467/Formation.Controleur.de.Serre.V2.pdf)

Loup, j'ai repris ton horaire pour le samedi, mais j'ai fait quelques petits ajustements.
Aussi, est-ce que tu peux ajouter une section malfonctionnement et problèmes dans ta présentation du contrôleur? Ex: Quoi faire si le contrôleur ne fonctionne pas?

Je suis toujours en attendre d'un retour d'Arnaud pour confirmation qu'on peut venir le 7 avril..

jayserf, 2018-04-04:
Super nice doc!
Hâte de voir et discuter ce samedi!

yoangp, 2018-04-13:
Merci Loup, j'ai corrigé l'erreur
[Formation Contrôleur de Serre V2.pptx](https://github.com/LoupHC/controleur-CAPE/files/1907713/Formation.Controleur.de.Serre.V2.pptx)

yoangp, 2018-04-21:
Version comprenant juste les plan de montage plus à jour publiée sur le google drive public :
[CAPE - Automate de serre - Plans de montage de la partie puissance et moteurs.pptx](https://github.com/LoupHC/controleur-CAPE/files/1935034/CAPE.-.Automate.de.serre.-.Plans.de.montage.de.la.partie.puissance.et.moteurs.pptx)
(modif du montage pour ventilateur et corrections montage tubes de supports des moteurs)

closed 2018-08-06 by LoupHC