LoupHC/controleur-CAPE #41: Documents techniques

LoupHC, 2018-04-11:
Ceci est une ébauche concernant les documents techniques à produire pour l'atelier. 
J'en ai fait 4 aujourd'hui que je vous présente, et qui m'on permis de commencer à définir une terminologie qu'on pourrait utiliser pour se comprendre à l'atelier.

LoupHC, 2018-04-11:
Premier document, l'assemblage des connecteurs sur le boitier. La terminologie utilisée ici : 
- les connecteurs sont numérotés de C1  à C14 un peu comme les pins d'une chip électronique, en partant d'en haut à gauche, en descendant et en remontant par la droite.
- un M veut dire connecteur mâle et un F connecteur femelle
- en plus du numéro de connecteur, chaque pin est numérotée, donc si on parle de la pin 3 du connecteur #4 on dira C4-3, etc.
- les relais sont numérotés de 1 à 3 et chaque pins sont numérotées de 1-8, donc la pin 6 du relais 3 = relais3-6
- pour le reste, j'ai utilisé les inscriptions marquées sur les pcbs
[image]

LoupHC, 2018-04-11:
Deuxième document, l'assemblage des switchs d'override sur la porte.
- Ici, les switchs d'override sont numérotées de  S1 à S6 dans le sens de la lecture (gauche-droite/haut-bas)
- Sauf la switch d'alarme, qui s'appelle ALARM-S
- Le bouton d'urgence s'appelle STOP(vis : STOP-IN et STOP-OUT)
[image]

LoupHC, 2018-04-11:
Et voici deux tutoriels : un pour l'assemblage de la sonde et un pour l'assemblage de l'interface.
[image]
[image]
[image]

LoupHC, 2018-04-11:
Voici ce que je pense faire d'ici l'atelier en plus de ça : 
- longueurs et couleur des câbles
- tableau résumant les connexions (comme ce que t'avais fait Yoan en 2016)
Avez-vous d'autres suggestions de documents à faire?

jayserf, 2018-04-11:
Il faut que j'y pense plus...

LoupHC, 2018-04-11:
Document sur la longueur des fils : 
[Longueurs de fil.docx](https://github.com/LoupHC/controleur-CAPE/files/1900225/Longueurs.de.fil.docx)

yoangp, 2018-04-11:
Wow, great job Loup!

jayserf, 2018-04-12:
Vraiment! Travaille de fou

yoangp, 2018-04-13:
Document en cours sur les étapes de montage:
https://docs.google.com/spreadsheets/d/1Ph90L6zSw53EDh-UzOvEIbaZvkrDvfvorQicZrnXrpg/edit#gid=0

closed 2018-08-06 by LoupHC