LoupHC/controleur-CAPE #48: Bug situation réelle #1 - Power supply 24V sauté - fuse F1 du contrôleur sautée.. Surtension?

yoangp, 2018-05-07:
Allo,

Je vais commenter le bug que j'ai en ce moment sur mon contrôleur, peut-être ça sera utile pour quelqu'un d'autre dans le futur..

Soiré d'orage suivie par power supply d'addison 24VDC 12,5A sauté et fuse F1 du contrôleur sauté sur le 5VDC. Peut-être tout ça a sauté à cause d'une surtension..

Loup, la fuse F1 sur le contrôleur que j'avais était de 1A, mais sur le schéma c'est 1,5A. Tu avais fait des tests à ce sujet, est-ce que je devrait vraiment utiliser 1A pour éviter tout domage, ou 1,5A sera ok?

Merci!

LoupHC, 2018-05-07:
Effectivement la fuse F1 saute seulement dans les situations de surtension... grace à elle tu viens de sauver toute l'électronique du circuit...

La fuse doit bruler le plus rapidement possible pour ne pas endommager le varistor et la zener de protection. Entre 1A et 1.5A la différence nest pas grande mais c'est important quelle soit de type fast blow.

LoupHC, 2018-05-07:
Pour comprendre le circuit de protection en cause :
https://www.sunpower-uk.com/glossary/crowbar-protection/

jayserf, 2018-05-07:
Le ps a sauté aussi?!

yoangp, 2018-05-20:
Update:

La surtension a sauté le power supply 24VDC et la fuse sur le circuit de sécurité du controleur sur le 5VDC. J'ai remplacé les 2 et depuis tout marche normalement! Yé!!

Mesure préventive: J'ai installé une power bar avec protection contre les surtensions, pas la top top, mais protection jusqu'à 1500 joules d'énergie, 15$ chez addison.

Je pense attendre que les autres bugs en cours soient réglés avant de parler de cela au groupe projet..

yoangp, 2018-08-05:
Allo les gars! J'espère que la saison va bien pour vous! Hmm, côté contrôleur; 2ième orage, 2ième power supply sauté! Shit! Les 2 étaient des power supply de led 12A de chez Addison. Je viens de commander un Mean well 6A comme au dernier atelier et un module contre les surtensions Mean well (https://www.digikey.ca/product-detail/en/SPD-20HP-277S/1866-5480-ND/9451829?utm_medium=email&utm_source=oce&utm_campaign=_OCE18RT&utm_content=productdetail_CA&utm_cid=604405&so=56027421&mkt_tok=eyJpIjoiT0dZME1ETXhabUpqTXpabCIsInQiOiJYbjd1QytkZnBiSG5ubjgyWDZtKytnMTFpelBlK0E3REZVYzhpaEs5ZHpsM0YxNU9xSkcrd2pDTUI1OVc5NUdoUktXMWM3YTVyQmM2S2Yyc21rSkh5ckFEajZ5eHNZWnRWYTgyUXBpckhlb1cxRlVrWHR2d3NMQ0xRdndLY1l6RSJ9). J'espère que je vais régler le problème ce coup ci!! La seule affaire c'est que le power supply de 6A sera pas suffusant pour mes besoins, il va falloir que je joue avec le code pour pas que les moteurs démarrent un après l'autre. À suivre...

LoupHC, 2018-08-05:
Hey si yest pas trop tard pour annuler ta commande, j'ai des power supply 24v à tension ajustable en trop ici (+- 2v genre)!
Je me demande si ça pourrait palier le problème des relais.
Sinon j'ai proposé à Fred Verville d'ajouter des thermistors en série avec ses moteurs et ça a reglé le problème chez lui!
Bref, si tu veux je peux t'envoyer ça cette semaine.
Très belle saison ici, l'étang est pas encore sec donc tout va bien!

LoupHC, 2018-08-05:
Aussi, si tu décides d'y aller avec le software fix, je te suggère d'y aller avec un délais utilisant millis() plutôt que delay(), qui va freezer le programme.
Un exemple avec la librairie elapsedMillis : 
https://playground.arduino.cc/Code/ElapsedMillis

yoangp, 2018-08-05:
Cool, merci Loup pour les précisions sur le code. Je vais garder ma
commande de power supply.

Je pense que c'est une bonne chose que tu en ailles quelques un en stock
pour d'autre monde..

Bonne semaine,

Yoan