LoupHC/controleur-CAPE #49: Bug alarme qui part lorsque les roll-up s'activent

yoangp, 2018-05-20:
Problème:
-Étienne Goyer mentionne que l'alarme part pour un court moment lorsque les roll-ups s'activent.

Hypothèse: 
-Le courant lors de l'activation des relais et des moteurs des roll-up crée une chute de tension qui désactive momentanément le relai de l'alarme.

Pistes de résolution de problème:
1-La résistance de protection introduit une chute de tension trop élevée. Voir si la mise en place des jumpers pour contourner le résistance de protection règle le problème.

2-Le démarrage des moteurs tire trop de courant du power supply 24VDC et se traduit pas une baisse de tension suffisament grande pour que le relai de l'alarme (et les autres relais aussi surement..) deviennents momentanément inactifs. Ça m'étonnerait que ça soit ça le problème, car le montage est supposé consommer moins que la moitié de la capacité du power supply 24VDC, mais le courant de démarrage peut poser problème. Les thermistors sur les driver de relais sont là pour minimiser ce problème, mais je n'ai jamais pu vérifier que tout est ok par manque d'équipement de mesure.

Observation: J'avais déjà noté lors de l'atelier que les relais indépendants, avaient l'air de ne pas n'activer fermement (même sans moteurs branchés). En premier, la led de certains relais ne reste pas allumée (signe d'un manque de tension). Ensuite, les relais cliquent plus faiblement. C'est peut-être aussi un effet cummulatif; le premier relai qui s'active ne représente qu'une petite charge, mais rendu au 5ième relai et plus la baisse de tension est significative. Je suspecte grandement la résistance de protection. Loup, avec l'oscilloscope, il serait possible de voir à quel point la chute de tension est importante lorsque les relais s'activent. Il faut brancher l'oscilloscope sur la tension 24VDC juste avant les relais (une fois passé de l'autre côté de la résistance de protection), activer le test 5555, puis mesurer la tension avec l'oscilloscope lorsque chaque relai s'active.

LoupHC, 2018-05-26:
C'est probablement un peu des deux. 
Avec deux moteurs à vide, en incluant les thermistors, le courant (mesuré au multimetre) est de 1,8 à 2,5A au démarrage, et de 0,4 à 0,8A en temps normal. (sans thermistor ça monte à 4A et probablement même plus si on pouvait faire une lecture très précise).
 
À 2,5A ça n'impacte pas le power supply, c'est pour ça que je l'ai pas vu passer, mais possiblement qu'avec 100 pieds de plastique et de tube galvanisé ça augmente significativement la charge au démarrage.
La cause du problème serait donc : probablement que le power supply dépasse son plafond de 6A et fait dropper la tension en conséquence. La résistance a un effet cumulatif vu qu'elle fait déja dropper la tension de 0,7V/relais actif.

Le problème semble survenir en PM, donc les rollups sont probablement déjà en partie enroulés (la charge est au maximum) et le relais de ventilation est possiblement aussi démarré : 
- ça fait 4 relais actif donc 2,8V de chute
- si la tension chute encore de 15% à cause de la protection contre les surtensions on commence à perdre des relais.

En plus de bypasser la résistance, un autre piste de solution serait d'ajouter un condensateur pour amortir les pics de tension au démarrage. Peut-être même que ça résoudrait tous les problèmes et qu'on pourrait garder la protection que représente la résistance (qui a quand même son utilité).

LoupHC, 2018-08-06:
Solution testée à la ferme du Coq à l'âne : 
- un thermistor supplémentaire en série avec chaque moteur
- bypass de la résistance de protection
Et ça marche! Plus de chute de tension déclarée.