LoupHC/controleur-CAPE #52: Empêcher d'activer les 2 roll ups en même temps

yoangp, 2018-08-30:
Allo Loup, pour garder une trace de notre échange...

Objectif:
Il faut empêcher d'activer les 2 rollups en même temps pour réduire le courant tiré du power supply (c'est à dire une charge de max 1 moteur plutôt qu'une charge max de 2 moteurs)

Une bonne partie de la solution:
Dans la loop du fichier Greenhouse_controller_CAPE, remplacer:

#if ROLLUPS >= 1 && defined(ROLLUP1_DESHUM)
R1.routine(rollup1Deshum, coolingTemperature, greenhouseTemperature.value());
#elif ROLLUPS >= 1 && !defined(ROLLUP1_DESHUM)
R1.routine(coolingTemperature, greenhouseTemperature.value());
#endif

#if ROLLUPS == 2 && defined(ROLLUP1_DESHUM)
R2.routine(rollup2Deshum, coolingTemperature, greenhouseTemperature.value());
#elif ROLLUPS == 2 && !defined(ROLLUP1_DESHUM)
R2.routine(coolingTemperature, greenhouseTemperature.value());
#endif

PAR:

 if(R2.isMoving() == false && R2.isWaiting() == false){
  #if ROLLUPS >= 1 && defined(ROLLUP1_DESHUM)
    R1.routine(rollup1Deshum, coolingTemperature, greenhouseTemperature.value());
  #elif ROLLUPS >= 1 && !defined(ROLLUP1_DESHUM)
    R1.routine(coolingTemperature, greenhouseTemperature.value());
  #endif
  }

  if(R1.isMoving() == false && R1.isWaiting() == false){
  #if ROLLUPS == 2 && defined(ROLLUP1_DESHUM)
    R2.routine(rollup2Deshum, coolingTemperature, greenhouseTemperature.value());
  #elif ROLLUPS == 2 && !defined(ROLLUP1_DESHUM)
    R2.routine(coolingTemperature, greenhouseTemperature.value());
  #endif
  }

yoangp, 2018-09-05:
Mise à jour:

J'ai mis en service mon contrôleur avec la modif. À première vue tout fonctionne. Lorsque j'allume le contrôleur, la fermeture des rollups pour la calibration se fait un côté à la fois. Ensuite, les côtés s'ouvrent à tour de role, jusqu'à 100% des deux côtés lorsqu'il fait chaud:

[image]

Mais après un petit bout de temps un bug étrange apparait. Il fait toujours aussi chaud, le contrôleur ne devrait rien faire. Mais il se met à afficher ceci:

[image]

Pis là ça coince, car il ne se referme plus lorsque la température redescend (j'ai mis la sonde de temp dans l'eau glacée et rien ne se passe).

Questionnements:
-Pourquoi le contrôleur se met à afficher OPENING en continu alors que l'ouverture est déjà de 100%?

Observations:
- Lorsque le bug apparait, le contrôleur n'est pas complètement buggé, je peux encore modifier les paramètres avec le clavier. Par contre, le bug persite. Il faut que je ralumme le contrôleur pour sortir du bug.
- Lorsque le bug apparait, les mots OPENING - OPENING ne clignottent pas et les sorties sont activés, donc il essait réellement d'ouvrir plus que 100% d'ouvreture..
- Avec le code initial, j'avais déjà noté ce comportement étrange, mais ça n'affectait pas le fonctionnement normal des rollups. À un moment donné je voyais que l'écran était bloqué sur OPENING. Hypothèse: Le bug existait déjà, mais le contrôleur pouvait s'en sortir lorsqu'un changement d'état survenait. Avec le nouveau code, il faut impérativement que l'ouverture ou la fermeture soit terminée avant qu'un autre action s'effectue sur les rollups.
- Côtés paramêtres, voici ce que j'ai:
`

//Create greenhouse object
#define TIMEPOINTS            1
        //# of timepoints
#define ROLLUPS               2          //# of rollups
#define STAGES                4          //# of cool stages (for rollups)
#define FANS                  0          //# of fans
#define HEATERS               0          //# of heaters
//********************PINOUT**************************
#define ROLLUP1_OPENING_PIN   0    //connect this pin to the opening relay (west motor) (Si pin 0, alors LED D16)
#define ROLLUP1_CLOSING_PIN   1    //connect this pin to the closing relay (west motor) (Si pin 1, alors LED D15)
#define ROLLUP2_OPENING_PIN   2    //connect this pin to the opening relay (east motor) (Si pin 2, alors LED D14)
#define ROLLUP2_CLOSING_PIN   3    //connect this pin to the closing relay (east motor) (Si pin 3, alors LED D13)
#define HEATER1_PIN           OFF    //connect this pin to the heater relay
#define HEATER2_PIN           OFF    //connect this pin to the heater relay
#define FAN1_PIN              OFF  //Connect this pin to the fan relay
#define FAN2_PIN              OFF  //Connect this pin to the fan relay
#define ALARM_PIN             7    //Connect this pin to the safety buzzer

//*************COORD GÉOGRAPHIQUES*********************
#define TIMEZONE              -5
#define LATITUDE              46.03
#define LONGITUDE             -73.56

//TIME AND DATE ARE SET AT FIRST UPLOAD ONLY
//For subsequent RTC programming :
//1. Uncomment (by erasing the "//") the line #define RTC_TIME_SET and #define RTC_DATE_SET and upload to set time
//2. Put back the comment markers("//") and upload again
//ALWAYS SET TO WINTER TIME

#define RTC_TIME_SET
#define HOUR_SET            20
#define MINUT_SET           58
#define SECOND_SET          0

//#define RTC_DATE_SET
#define DAY_SET             29
#define MONTH_SET           8
#define YEAR_SET            2018
//*******************************************************************
/*
TIMEPOINTS PARAMETERS - SYNTAX RULES:
- type variable(TYPE) :   SR, CLOCK or SS (sunrise, manual, sunset)
- hour variable(HOUR) :   SR/SS types => 0 -24 to 24
                          CLOCK type => 0 to 24
- min variable(MN_MOD) :  SR/SS types => -60 to 60
                          CLOCK type => 0 to 60
- heating temperature variable(HEAT) : 0 to 50
- cooling temperature variable(COOL) : 0 to 50
[... 79 lines cut ...]
#define R1_S2_TARGET        50
#define R1_S3_MOD           2
#define R1_S3_TARGET        75
#define R1_S4_MOD           3
#define R1_S4_TARGET        100
//*******************************************************Rollup 2 (stages)

#define R2_S0_MOD           0
#define R2_S0_TARGET        0
#define R2_S1_MOD           0
#define R2_S1_TARGET        25
#define R2_S2_MOD           1
#define R2_S2_TARGET        50
#define R2_S3_MOD           2
#define R2_S3_TARGET        75
#define R2_S4_MOD           3
#define R2_S4_TARGET        100

//*************************************************************************
`

Hypothèses:
Le bug est probablement dans le code, mais pourrait aussi venir d'un glitch électrique qui met le controleur dans un état weird. J'ai donc tenté de reproduire le bug avec les moteurs déconnectés, le problème n'est pas réapparu. J'ai essayé de reproduire le bug avec les moteurs connectés. le bug est revenu. J'ai fait ça 2 fois avec le même résultat. Étrange.. Mais je pense toujours qu'il y a plus de chance que le bug soit vraiment dans le code.

Loup as tu une idée de ce qui pourrait causer cela? Sinon, est-ce que tu vois quelque chose d'autre que je devrais noter ou bien un test/flag à ajouter dans le code en temps réel pour identifier ce qui le fait bugger? Toute piste de solution est la bienvenue :-)  ... surtout avec le froid qui se pointe!

yoangp, 2018-09-06:
Ça se peut tu que le bug apparait lorsque les rollups sont ouverts à 100%, mais que la température continue de monter? Sur les photos que j'ai envoyés, la température a augmentée après que les rollups aient été ouvert à 100%.

LoupHC, 2018-09-06:
Vite de même, je dirais que le bug se situe au niveau de la calibration automatique des rollups, qui effectue une rotation complète pour s'assurer que les moteurs sont collés sur la limit switch.
As-tu vérifié si le bug persistait plus que 2-3 minutes?
Sinon tout va bien, c'est voulu! 
Si oui, je jetterai un oeil là-dessus, en attendant tu peux désactiver la fonction de calibrage automatique dans l'onglet paramètre en remplaçant :

#define R1_RECALIBRATE
#define R2_RECALIBRATE
#define SAFETY_DELAY                  30 (nombre de minutes entre chaque calibrage)

par 

//#define R1_RECALIBRATE
//#define R2_RECALIBRATE
//#define SAFETY_DELAY                  30

LoupHC, 2018-09-06:
Concernant le fait qu'il ne redescende plus si la température baisse, c'est que le signal est figé le nombre de secondes correspondant à une ouverture complète.

yoangp, 2018-09-07:
J'ai commenté le calibrage automatique et ça semble fonctionner. Il y aura un retour à faire dans le code à la fin de la saison pour voir s'il est possible de remettre le calibrage automatique. Ce qui est certain, c'est qu'il y a quelque chose qui ne fait pas la bonne affaire dans la fonction de calibrage automatique avec ce nouveau mode d'ouverture un après l'autre. Je dis ça car l'écran montrait que les deux côtés étaient en phase d'ouverture pendant le calibrage, il faudrait que ça soit juste un côté à la fois qui se calibre.

Malgré tout, un gros merci Loup pour ton temps sur cette modif du code, je suis prêt pour le temps froid :-)