LoupHC/controleur-CAPE #54: Subvention InnovAction - 1er octobre 2018

yoangp, 2018-09-25:
Commentaires généraux:

-Loup, dans la demande tu devrais te présenter comme coordonnateur du projet. Ça te donne un statut important dans le projet même si tu n'as pas de formation spécifique en électronique/informatique. (Suggestion de ma conjointe) À voir si tu pourrais demander le même taux horaire qu'un ingénieur pour les heures de coordination.

yoangp, 2018-09-25:
J'ai mis un début de brouillon du formulaire (appelé Brouillon Yoan) sur Dropbox/subvention 2019/paperasse.  C'Est juste des idées, on s'en rejase..

yoangp, 2019-01-04:
Brouillon de calcul de l'aide financière:
https://docs.google.com/spreadsheets/d/1exxTFzlRIwsBK15juQJ0n2sMos91u_hbtw-HYLdaNtg/edit?usp=sharing

yoangp, 2019-01-13:
Allo Loup,

Désolé pour le délai. Après avoir parlé à Geneviève Legault la semaine passée, elle a dit que la dernière fois environ 50% des projets ont été financés sur un total d'environ une centaine de demandes. Aussi pour les délais de traitement, elle a dit qu'elle a participé à l'envoie d'une subvention pour le volet 1 au deadline d'octobre dernier et qu"il n'y a pas eu encore de réponse. Donc, il faut planifier plusieurs mois de délais avant de savoir si ça va marcher ou pas.

Après que je lui aie raconté ce qu'on compte faire, son commentaire le plus important était qu'il faut qu"il y ait un aspect nouveau/innovateur et que le fait que les grosses serres ont probablement déjà la technologie qu'on essaie d'implanter n'est pas une bonne affaire. Elle a dit qu'on peut essayer de dire qu'il y a un gap technologique pour les petites et moyennes fermes. L'autre affaire c'est les priorités du secteur, c'est bien beau qu'on dise que c'est important les serres pour les petits maraichers, mais la subvention vise tout le secteur agricole, autant conventionnel que bio, animal que végétal et qu'il y a des priorités officielles qui ont été écrites dans les dernières années. Apparemment Frédéric Thériault de Tournesol a participé à l'établissement des priorités en maraichage biologique. Mais le bio ça reste 1 à 2% des productions végétales du québec, donc peut-être pas tant prioritaire overall. Aussi, ça ne va peut-être pas passer si on semble trop être dans un mode de pré-production. Il faut que ça reste quand même au niveau de la recherche même si c'est une adaptation technologique. En résumé, elle pense que notre projet ne colle pas à 100% avec ce qui est visé par la subvention, mais j'ai l'impression qu'on peut s'essayer quand même.

Elle a dit qu'pourrait réviser notre demande, pour cela je pense qu'il faudrait lui donner 2 semaines. Je propose qu'on essaie de finir notre brouillon d'ici vendredi prochain.

Je vais donner un gros coup sur la rédaction de la subvention la semaine qui s'en vient, dès lundi soir je devrais avoir de quoi à te partager.

Est-ce que tu pourras appeler Climax en début de semaine?

LoupHC, 2019-01-14:
Salut Yoan!
Pas de problème pour appeller Climax! Par contre, c'est moyen encourageant ce qu'elle nous dit là...
est-ce qu'elle a noté des éléments à notre avantage sur lesquels on devrait miser?

yoangp, 2019-01-16:
Réponse de Geneviève Legault:

"Après discussion avec les collègues, la demande devrait bien prouver que la technologie n’existe pas ou qu’elle ne répond pas au besoin et le type d’adaptation qui doit être fait pour que cela réponde au besoin des entreprises de petite et moyenne envergure.

Produire une technologie plus abordable n’est pas un justificatif, il faut démontrer la problématique au niveau technique.

Bonne chance dans vos développements,

Geneviève Legault, agronome, MSc."

closed 2019-01-16 by yoangp