LoupHC/controleur-CAPE #56: Brainstorming - Nouveau design, prototype 2019?

yoangp, 2018-12-13:
Allo!

Pour m'amuser, j'ai fait le schéma suivant pour essayer de voir s'il y aurait une façon de rendre le contrôleur vraiment pas cher tout en offrant une possibilité de communication/alarmes sur smartphone. Le schéma regroupe toutes sortes d'options, il ne faut pas penser qu'il y aurait tout ensemble. Par exemple, il y a les deux options de design Arduino Yun ou Arduino Mega + Raspberry Pi.

https://docs.google.com/drawings/d/1-hGnpsdTrqrXmDoJYeqsXhl5zh_nJ-BTi--XsoNR4tE/edit?usp=sharing

Commentaires:
- Le projet Farmbot utilise l'équivalent d'un Arduino Mega (Pwm, interrupts, IO,..) et un Rapsberry Pi (serveur web, wifi)
- Le projet OpenSprinkler fonctionne sur Raspberry Pi (ou autres..), mais ne semble pas nécessiter d'interrupts ou de pwm..
- Je n'ai pas vu l'Arduino Yun dans un projet commercial, mais ça semble vraiment simple vu qu'en un bloc ça peut tout faire (IO, PWM, Interrupts, serveur web, wifi, etc.) et que c'est fait à la base pour être vraiment facile de faire un lien entre les IO du Yun et un site web sur le serveur web ou d'envoyer des messages via l'internet...
- J'aime vraiment l'idée de faire de quoi vraiment pas cher, avec le mimimum de montage.
- L'idée d'envoyer des sms via un shield GSM est intéressante dans le sens que le risque de problèmes est minimal (s'il est possible de trouver un shield qui fonctionne bien.,,). Pas de chaine de communication complexe qui passe par l'internet. Mais ça reste pas mal basic et le futur sera aligné plus sur les solutions via internet plutôt qu'un contrôle par sms qui fait penser à la technologie des années 2000.
- Je pense qu'on peut couper par 2 le coût des drivers de moteurs si c'est un H-bridge avec mosfets plutôt que des relais. Mais c'est plus complexe car il est souhaitable d'utiliser des sorties PWM pour faire des démarrages/arrêts de moteurs smooths.

LoupHC, 2018-12-13:
Hey! Belle recherche Yoan!
J'ai plein de petites nouveautés à vous partager aussi! Je reviens d'une formation avec Climax-Conseil et j'ai jasé pendant longtemps de gestion climatique avec Sébastien et Dany. Mais avant de me lancer là-dedans, mes commentaires :
- choix mega+RPi vs Yun : je privilégierais le MEGA parceque c'est le processeur qu'on utilise déjà avec les Otomates déjà existants, tout simplement. Aussi, le Yun a moins de mémoire SRAM et faudrait couper ou refactoriser pas mal de code pour rentrer le software existant sur le atmega du Yun (et même le processeur linux est petit).
- opensprinkler : à ce sujet, j'ai appris quelque chose d'assez intéressant récemment, le raspberry pi d'opensprinkler donne la possibilité de se connecter à un téléphone en acces point. Autrement dit, même sans connexion au réseau internet, on peut s'y connecter avec un téléphone en établissant un genre de wifi local!
- intégration du wifi : déjà je pense que pour mieux répondre aux besoin de tout le monde on est mieux de prendre pour acquis qu'aucune serre n'a de connectivité wifi pré-installée à priori. Ensuite, si on réussit à connecter le contrôleur d'une façon pas chère et abordable, j'ai quand même quelques appréhensions. Je suis probablement juste chicken envers quelque chose que je ne connais pas vraiment, mais j'ai l'impression que toutes les compagnies qui offrent des services via internet sont jeunes, évoluent vites, peuvent disparaître du jour au lendemain ou modifier leurs tarifs sans prévenir. Si on vise quelque chose de low-maintenance, je m'en méfierais... Cependant ça donne plein de possibilités intéressantes comme contrôle à distance, mais en même temps, est-ce qu'on connait beaucoup de fermiers qui passent beaucoup de temps loin de la ferme?
Un des pièges dans lequel (intuition) faudrait pas tomber, c'est créer une dépendance entre les utilisateurs et nous. Si un jour on se retire du projet, faut pas que les gens se retrouvent avec des bidules inutilisables parceque plus personne ne s'occuppe de la maintenance back end. Mais comme je dis, c'est surtout que j'y connais pas grand chose donc je suis peureux mais je suis vraiment ouvert en fait!
- simple is beautiful, AMEN!
- messagerie SMS : probablement la connexion la plus fiable (quoique) et la plus accessible pour les serres, si on trouve du hardware qui fait la job, je dis oui!!! Et toute la maintenance est assurée par les compagnies de téléphone (qu'on paye évidemment) mais ça nous enlève de la job!

- une autre possibilité : les devices qui acheminent l'internet via l'alimentation électrique. Vieille technologie super fiable apparemment. Et pas trop cher, genre 70-100$? Un RPi serait un bon candidat alors parcequ'ils ont un port ethernet intégré et peuvent se brancher au arduino via usb sans autre hardware nécessaire. Aussi le raspberry pi a le bluetooth intégré alors ça se programme facilement via un téléphone.
- une autre alternative : le LoRa pour communiquer entre la serre et un ordi branché au réseau internet, ou un ordi qui agit comme serveur. Ça marche pas à plus d'1 km mais c'est quand même à considérer!
- côté H-bridge faut se méfier parceque c'est sensible aux surtensions et au problèmes de dissipation thermique. C'est donc un potentiel weak spot si on design mal notre affaire, mais toi Yoan, t'auras pas trop de mal à figure out ce côté-là je suis certain. :)
Faut juste penser limiteurs de courant, heatsink, varistors/TVS et remplacement en cas de chip défectueuse.
Pour ce 3e point j'aime beaucoup le format DIP, j'avais trouvé cette chip prometteuse mais je l'ai brûlé par manque de protections : https://www.digikey.ca/product-detail/en/stmicroelectronics/L6206N/497-4211-5-ND/725435
C'est un h-bridge double donc on peut doubler ses specs en utilisant les 2 h-bridge en parallèle. C'est 10$ la chip mais ça a le potentiel d'avoir une durée de vie de 100ans si on design ça bien et encore moins cher qu'un relais+socket!
Côté prog, j'ai beaucoup de reformattage à faire pour intégrer l'option pwm dans le contrôle des moteurs, donc je laisserais quand même possible l'option on/off à court terme, mais c'est entièrement possible et on peut caller la shot d'avance en intégrant une switch ou un jumper sur le circuit pour envoyer le signal aux sorties PWM plutôt que digitales.
En passant, j'ai demandé un budget de 5000$ pour cet hiver pour prototypage. À auto-financer par la suite en vendant des prototypes mais quand même, ça pourra nous donner du lousse!

LoupHC, 2018-12-13:
- sinon alternative au format DIP, si on veut plutôt acheter un module, je m'imaginerais des headers femelles dans lequel on emboîte le module, comme le module d'horloge dans le contrôleur actuel.
- aussi, parlant de chips, avec un simple switch (MCP23008 à MCP23016), on pourrait contrôler 16 sorties plutot que 8 sans changer quoi que ce soit au code!

yoangp, 2018-12-18:
Suite/réponse Yoan:

-Plateforme hardware Arduino Mega + Raspberry Pi: Je trouve dommage qu‘il n‘y ait pas de meilleure option dans un kit complet. Il me semble que juste organiser nous même la communication entre le mega et le pi est un bon projet en soi. Mais bon, sinon il y a l‘option Arduino mega + shield wifi, mais je pense que ça serait limité pour les interactions avec le web. En tous cas, le projet farmbot open source utilise quelque chose qui ressemble à l‘arduino mega + r pi, donc on pourrait utiliser  ce qu‘ils ont fait comme inspiration.

-Acces point via le raspberry pi: Oui, c‘est ça que je voulais faire. L‘idée c‘est d‘utiliser à la base un module raspberry pi qui a déjà du wifi intégré (ex raspberry pi modèle 3 a+ à 30$). Ensuite, établir un lien de communication entre le pi et l‘arduino. Ensuite installer un serveur web sur le pi. Ensuite, coder un interface web, qui est en fait comme un site web sur le serveur web du pi. L‘avantage; même sans connexion internet, tu arrives avec n‘importe quelle bébelle (laptop, tablette, smartphone), tu te connecte au réseau local du pi, tu ouvres un browser pis là il y aurait un interface ultra simple qui permet de voir ce qui se passe sur l‘otomate (l‘équivalent de ce qui est affiché sur le lcd de l‘otomate 2018) et des boutons et champs de saisie pour faire l‘équivalent du clavier de l‘otomate 2018). Vu que ça serait dans un browser, on serait assez à l‘abri des changements technologiques à venir dans les prochaines.

-Messagerie sms: ça serait tellement simple, mais plus que je cherche, moins que ça avance. Les modules pas cher semblent utiliser une vieille technologie sur le réseau 2G. Ça serait mieux pour le futur d‘être sur le réseau 3G ou 4G, mais là les modules sont vraiments plus cher et vraiment overkill pour une simple petite fonction d‘envoie de messages textes. Je suis en train de me dire qu‘une alternative serait de passer malgré tout par le web ( si le web n‘est pas dispo à la serre, utiliser un vieux smartphone à vitre cassée obtenu gratuitement, le mettre en fonction hotspot, avoir une sim card avec un plan de données de base (15$/mois)., je pense que ça pourrait marcher, je me demande juste si la connexion serait fiable 24/7).Ou internet via le filage 120vac comme tu dis. Bon, admettont qu‘on a une connexion internet, on utilise un service genre pushingbox.com. Ça coûte rien, pi ça forward des messages via courriel, mms, etc.

-Utiliser un vieil ordi comme serveur: je ne suis pas chaud à l‘idée. Je suis déjà passé par là lorsque je travaillais à hydro québec. Pour du service 24/7 ça marche vraiment mal car il y a toujours plein d‘affaires qui se passent à cause du système d‘exploitation. Ex,  updates automatiques qui ne sont pas vraiment automatiques, etc. Se lancer dans notre propre version de linux sur de vieilles marchines partout au québec, brrr, euh non merci! Pas simple, pas fool proof, probablement pas super 

-simple is beautiful : je suis 100% d‘accord avec toi Loup. C‘est juste que c‘est pas évident de trouver l‘affaire simple qui va marcher 10 ans sans maintenance et qui va avertir quelqu‘un sur son cellulaire que de quoi ne marche pas dans la serre.

-En réponse à pourquoi se casser la tête pour envoyer des notifications sur smartphone si la majorité des fermiers sont toujours à la ferme? : Les cas de figure que je vois:
-Même en habitant sur la ferme, si c‘est toi le proprio tu va te promener pas mal pour plein d‘affaires. Achat de matériel, points de chutes, marché, sans compter les affaire perso comme épicerie, visites de famille, garderie, etc. Imagine, tu es parti de la ferme acheter de quoi en plein mois de mars, il n‘y a pas d‘employés à la ferme car trop tot en saison, la serre tombe en panne. Même avec une sirène de 120dB à la maison, ça va pas faire grand chose. Ça prend pas bien des heures avant que les semis de tomates capotent s‘il fait -10 degrés dehors et que la  serre ne chauffe plus.

-H bridge: Je pense que ça a le potentiel d‘être plus fiable et moins cher que relais + sockets. Pour la fragilité aux surtensions; on a déjà ps mal de stock fragile aux surtensions, dont nos power supply qui fonctionnent eux aussi avec des transistors de puissance. Donc de toute façon, il faut que tout notre système soit protégé des surtensions. Pour la dissipation de chaleur, les transistors du module que j‘ai chez moi peuvent prendre du 100A selon la datasheet. Donc les quelques ampères des moteurs 24V ne devraient pas être un gros problème. C‘est sur que dans le cas ou quelque chose bloque les moteurs ça va monter sérieusement les ampères, mais avec un mécanisme de protection, genre fusible ou autre, ça devrait faire l‘affaire avant que les transistors de puissances brulent par surchauffe  Pour l‘histoire des pwm, c‘est plus de trouble dans le code et pour les i/o (faut que ça soit une sortie pwm et pas juste une sortie digitale ). Par contre, les softs starts produits grace au pwm font que la demande de courant des moteurs au démarrage est vraiment beaucoup plus douce. On peut démarrer les moteurs de 0 à 100% de la vitesse en genre 5 secondes plutot que 100% en 1/10 de seconde s‘il n‘y a pas de pwm.

- Cool pour le prêt de 5000$ à la capé, tiens moi au courant si ça marche.