LoupHC/controleur-CAPE #58: Aide d‘étudiants pour faire avancer le projet?

yoangp, 2018-12-18:
Ça serait peut-être le temps d‘essayer d’utiliser l’otomate pour lancer un projet de session pour les étudiants en génie électrique/informatique de l‘université de sherbrooke! On pourrait leur donner comme mandat de coder une partie de la communication entre le mega et l‘arduino ou de coder un interface via le serveur web du Raspberry pi...

Si vous me donnez votre accord, je pourrais regarder ça. Pour être passé par là, ça ne coute pas grand chose. Il faudrait peut-être fournir un peu de matériel (arduino, r pi, etc.), mais on pourrait s‘en tirer pour vraiment pas cher d‘après moi. Les étudiants ne sont pas rémunérés. Normalement, ils ont un nouveau projet pour la session d‘hiver qui commence en janvier. Une compagnie, ex capé, peut proposer un projet à réaliser et si ça fonctionne il y a environ 5 étudiants qui vont passer pas mal de temps là dessus pendant 15 semaines.

LoupHC, 2018-12-18:
Nice super bonne idée!
Si on se lance dans la micro-batch de prototypes cet hiver, j'ai demandé un 1000$ de budget prototypage si ça passe ça pourrait passer là-dedans.

yoangp, 2018-12-19:
En date du 19 décembre, il est trop tard pour soumettre un projets pour la prochaine session à l'Université de Sherbrooke.
En attente d'une réponse du CEGEP de Sherbrooke. Ils ont aussi des projets de fin d'études.

yoangp, 2019-01-04:
Réponse du CEGEP de Sherbrooke:

"Le processus de choix de projets de fin d'études débute dès le mois d'août. Il est donc trop tard pour cette année.
Si les délais occasionnés vous sont acceptables, je vous invite à nous recontacter dès le mois de juin.
Merci de votre attention."

Nicolas Huppé
Coordination TGE
819-564-6350 -4092

yoangp, 2019-01-04:
Réponse de l'université de Sherbrooke:

"Les soumissions de projets doivent se faire avant le mois de novembre de chaque année sur le lien suivant : https://www.usherbrooke.ca/gelecinfo/fr/divers/prop-prj/ "

Ghania Sarah FARHI, Ph. D.
Coordonnatrice académique
coordgegi@usherbrooke.ca
Département de Génie Électrique et de Génie Informatique
Faculté de génie
Université de Sherbrooke
Sherbrooke (Québec)

Canada J1K 2R1
Tél: (819)821-8000, poste 63844
Local: C1-3059