LoupHC/controleur-CAPE #60: Plan d'action 2019

LoupHC, 2019-01-14:
Bon!
Suite à une discussion qu'on a eu Yoan et moi, on a abandonné l'idée de faire subventionner le projet avec Innov'Action, qui répondait pas vraiment à notre situation et nous aurait probablement plus limité qu'autre chose dans ce qu'on voulait faire.
À moins qu'on trouve du financement autre, il nous reste une alternative : on auto-finance le projet en majeure partie!

Voici comment je pense qu'on pourrait procéder :
- on planifie le temps nécessaire pour arriver à une deuxième itération (finale?) de l'Otomate
- on répartit ça en fonction du temps qu'on peut mettre à contribution au projet pour se faire un calendrier
- on évalue l'impact sur le prix des contrôleurs en fonction du nombre d'unités produites si on rémunère le temps de développement à mettre
- on réajuste nos objectifs en conséquence au besoin
- on présente le dossier à la CAPÉ

Les objectifs mentionnés à date: 
- améliorer le design existant essentiellement pour réduire le filage et la complexité d'assemblage
- améliorer les fonctionnalités et le rendre plus facile à programmer
- arriver à un circuit standard "universel" à faible coût que les gens peuvent assembler eux-même
- mettre en place un système qui réduit au maximum les besoins de soutien technique 
(idée proposée : avoir quelques circuits de rechanges à échanger contre un circuit défectueux, qui pourrait être revendu comme réusiné après réparation)

Je propose ce document comme point de départ  : 
https://www.dropbox.com/ow/msft/edit/home/autoconstruction_private/autoconstruction%202019/projet-Otomate/Plan%20d'action%202019.xlsx

yoangp, 2019-01-15:
Super bon point de départ pour 2019!

Ajouts:

- Une méthode efficace de travail serait d'établir le minimum à faire pour avoir une prochaine version testable puis vendable. Quitte à laisser des améliorations pour une version 3, 4, etc.
- Jusqu'à ce que je trouve un emploi temps plein, je peux mettre 1 jour/semaine sur le projet.
- Comparer la version 2 de l'Otomate avec les autres produits sur le marché.

yoangp, 2019-01-16:
J'ai fait plein d'ajouts dans le fichier Excel sur DropBox dont un budget prélininaire de vente, comparaison de coût de contrôleurs et features de chacuns.