LoupHC/controleur-CAPE #63: Manque librairie pour schéma KiCAD

yoangp, 2019-01-17:
Loup,

Il manque une librairie dans les fichiers KiCAD (image ci-dessous), est-ce que tu pourrais l'ajouter dans les fichiers KiCAD sur GitHub?

[image]

LoupHC, 2019-01-17:
Snaps... je me mets là dessus

LoupHC, 2019-01-17:
C'est bizarre... il me dit qu'il manque les librairies, mais en vrai il arrive à ouvrir le fichier quand même...

LoupHC, 2019-01-17:
Shit je pense qu'ils ont changé le système de librairies pour les footprints... par une méthode plus compliquée qu'avant... mais les fichiers sont encore là.
Je vais les archiver sur github ça va être plus simple à l'avenir.

LoupHC, 2019-01-17:
J'ai mis à jour KiCad (passé de 4.0 à 5.0) et maintenant j'ai plus de problème avec les footprints. Par contre, c'est plus compatible avec KiCad 4.0...
J'ai reploadé ça dans une branche à part : KiCad-edit-2019
On pourrait mettre nos modifications là-dessus en attendant d'avoir qqchose qui vaut la peine d'être updaté!
Si tu mets à jour ta version est-ce que ça te pose encore problème?

yoangp, 2019-01-17:
Cool! Résolu de mon côté aussi avec KiCAD V5 et la version mise à jour du projet (KiCad-edit-2019)

closed 2019-01-17 by yoangp