LoupHC/controleur-CAPE #64: Plan d'action 2019 - nouvelles fonctionnalités

LoupHC, 2019-01-17:
J'ai retrouvé le sondage qu'on avait fait l'an passé par rapport aux priorités de développements, on demandait une note de 1-8 (un : très important, 8 : pas du tout).
- Monitoring : 5
- Contrôle via internet : 4
- Modulation selon température extérieure : 4
- Cycles de deshumidification : 3
- Fermeture en cas de pluie : 2
- Fermeture en cas de vent : 2

Voici les conclusions que j'en tire : 
- les fonctionnalités essentielles ++, ce sont celles qui sont liées directement à préserver la serre en bon état de marche (pas noyée dans l'eau, pas arrachée par le vent).
- en deuxième, ce sont celles directement liées à la conduite des cultures (bonne gestion de l'humidité, pas de coups de froid)
- en troisième, celles qui facilitent la vie (contrôle et monitoring via internet)

Un autre facteur qui peut nous orienter dans le choix de fonctionalités à prioriser, c'est "est-ce qu'un autre produit présent sur le marché peut jouer ce rôle?".
- Dans le cas de l'irrigation par exemple, ça pourrait être intéressant de la contrôler direct du contrôleur (et faisable), mais comme il n'y a pas d'interaction entre le contrôle du climat et l'irrigation, ça peut valoir la peine d'utiliser une alternative déjà existante et bien conçue (rainbird ou openSprinkler par exemple)
- Dans le cas du monitoring, il y a aussi des alternatives comme AcuRite qui sont vraiment pas chères et qui peuvent jouer ce rôle.

Je suis bien avancé du côté des overrides, ça devrait nous donner un gros éventail de possibilités. Cet hiver, côté fonctionnalités, je mettrais l'emphase là-dessus : station météo et gestion de l'humidité.
C'est pas mentionné dans le questionnaire mais l'interface a aussi beaucoup besoin d'être améliorée, ça serait ma deuxième priorité.
Vous en pensez quoi?

yoangp, 2019-01-17:
> Assigned #64 <https://github.com/LoupHC/controleur-CAPE/issues/64> to
> @yoangp <https://github.com/yoangp>.

- Je suis d'accord avec les priorités de développement.

- Loup, tu es vraiment confiant qu'on peut intégrer la station météo cette
année? Si c'était moi, je viserais d'intégrer en premier le kit vraiment
pas cher de Sparkfun dont tu m'avais déjà parlé (
https://www.sparkfun.com/products/8942). On parle donc surtout de vitesse
du vent et quantité de pluie pour la météo n'est ce pas? La direction du
vent est moins critique, bien que ça peut être utile de savoir de quel côté
la serre peut s'ouvrir même en conditions assez venteuses.

- Faire en sorte que les cycles de déhumidifications marchent pile poil est
aussi un must évidemment. Je l'ai appris à mes dépend l'été dernier avec du
blanc à grandeur, entre autre à cause de zéro gestion de l'humidité. J'ai
perdu 0,5jr/semaine à appliquer des produits pour controler le blanc sans
impact notable. Est-ce que tu as un capteur d'humidité en tête? J'ai pensé
à faire un montage bulbe sec, bulbe humide avec 2 capteurs de températures
qu'on utilise déjà, Danny disait que c'est la méthode la plus fiable (mais
il faut s'assurer qu'il y a toujours assez d'eau pour le capteur,
apparement aux 2 semaines environ pour un petit réservoir d'eau, facile à
oublier cette tâche répétitive...).

- Je vais ajouter cela dans les fonctions de l'Otomate 2019 pour le sondage
à venir.

LoupHC, 2019-01-17:
Pour le code, je suis pas trop inquiet, c'est le hardware qu'il faudrait absolument tester cet été...
Pour la station météo, je pensais réutiliser le code de ce gars-là :
http://cactus.io/projects/weather/arduino-weather-station
Et du coup, utiliser les mêmes capteurs serait le plus simple. 
Le capteur de pluie qu'il utilise est très bien fait, on l'utilise à la ferme avec le iGrow, et on peut programmer directement sur le module plusieurs degrés de sensibilité.
Pour l'anémomètre, la différence de coup est quand même importante, on parle de 150$ vs 80$ (avec un capteur de pluie en bonus).
Je me demande si la différence de qualité en vaut le cout, les commentaires sont ambigus pour les deux modèles...

LoupHC, 2019-01-17:
Pour la deshumidification, ya un gars qui était à l'atelier, Simon Leduc, qui a fait un capteur bulbe humide/bulbe sec avec arduino, faudrait le contacter pour avoir son code!
Sinon le plus simple à installer serait un capteur digital, mais moins fiable et plus cher comme tu dis. 
Je vais m'arranger pour que le code soit compatible avec les deux options.
La logique de contrôle évidemment serait la même.

yoangp, 2019-01-17:
Pour la sation météo, d'accord avec toi que le montage sur cactus.io et le capteur de pluie semble vraiment bien et pas cher. C'est juste l'anémomètre que je trouve cher à 150$, même si ça a surement à voir avec la qualité de la patente...

Je pense qu'on pourrais trouver un anémomètre (sans la direction du vent) pour moins cher, ex: https://www.amazon.ca/Sensor-Anemometer-Signal-Aluminium-Alloyed/dp/B07F6Z15N9/ref=sr_1_24?keywords=cup+anemometer&qid=1547765624&sr=8-24

yoangp, 2019-01-19:
Bon, j'ai repensé à la station météo et voici ma proposition; il y a deux setup possibles selon le budjet:
1. Kit Sparkfun pour 77$US (Économique)
2. Anémomètre et girouette Davis + Hydreon RG11 pour 214$US (Haut de gamme)

Les deux setup sont assez similaires:

Kit sparkfun:
Anémomètre : switch à on à chaque rotation
Girouette: array de résistances 0-120kohm
Rain bucket: switch on lorsque bucket se vide à 0.2794mm de pluie
Datasheet: https://cdn.sparkfun.com/assets/8/4/c/d/6/Weather_Sensor_Assembly_Updated.pdf

Davis instruments:
Anémomètre: Switch à on avec contact au ground à chaque rotation
Girouette: potentiomètre 0-20kohm
Datasheet: https://www.davisinstruments.com/product/anemometer-for-weather-monitor-or-wizard/
Hydreon:
Rain sensor RG11: switch on à chaque fois équivalent de .2mm de pluie (plus de précision disponible avec autre setings). Nécessite alim 12Vdc et 55mA
Datasheet: http://hydreon.com/wp-content/uploads/sites/3/2015/documents/rg-11_instructions.pdf

**Donc, en bref pour les différences:**
- Les deux anémomètres peuvent fonctionner comme entrée digitale 0-5V, il faudra juste avoir un paramètre de de convertion dans le code pour durée entre pulses par km/h de vent. Rien du côté hardware à part qu'il faut que le pulse soit une valeur à 0 (ground), à 1 le reste du temps car le modèle Davis est conçu ainsi, sur le modèle Sparkfun ça ne fait pas de différence.

- Les deux girouettes peuvent fonctionner sur une entrée analogique 0-5V, il faudra avoir une table de convertion dans le code pour entre x et y voltage = z direction du vent. Côté hardware, il faudra une switch entre deux valeurs de résistances selon la girouette qui est branchée.

- Les deux capteurs de pluie peuvent fonctionner sur une entrée digitale 0-5V, il faudra avoir un paramètre de convertion dans le code pour 1 pulse = x mm de pluie car c'est différent entre les deux modèles. Côté hardware, pas de problème pour une entrée digitale 0-5V. La grosse différence c'est que ça prend du 12V sur le board pour alimenter le Hydreon.. Donc on pourrait mettre un autre DC-DC converter, au moins celui là serait pas trop cher car genre 100mA serait suffisant. D'un autre côté il y a quand même souvent des pièces disponibles sur 12VDC, mais pas tant sur 5VDC ou 24VDC. Donc ça nous ferait une source 12V disponible pour d'autres affaires dans le futur (ex. un design de H-bridge prometteur (+ de 10A sur 24VDC pour les moteurs) fonctionne avec des inputs de contrôleur 0-5V, mais les drivers de mosfets ont besoin de 12VDC).

LoupHC, 2019-01-19:
La sonde de pluie optique a aussi la possibilité de fonctionner en mode actif quand il pleut, inactif quand il ne pleut pas, ce qui rend la programmation très facile à implémenter. Par contre, j'avoue que c'est intéressant de connaître les précipitations, comme avec une station météo normale.
Je pense aussi qu'on devrait chercher à être compatible avec plusieurs hardwares différents, ça devrait pas être trop de trouble à programmer!

LoupHC, 2019-01-20:
Nouvelles fonctionalités implémentées aujourd'hui :
- mode semi-automatique pour le rollups, les fans et les fournaises (complètement ouvert, fermé ou à X% pendant Y minutes ou indéfiniment)
- override en fonction de l'humidité pour les fans et les fournaises
- override en fonction de la pluie pour les rollups 
- override à heure fixe pour les rollups, les fans et les fournaises (de X heure à Y heure)

Programmable via l'IDE ou le tableau excel de programmation!

YOUPI!

yoangp, 2019-01-20:
Wow! Super cool Loup!! Pas de niaisage en fin de semaine!

jayserf, 2019-01-20:
for real!

way to go!