LoupHC/controleur-CAPE #65: Document explicatif du projet 2019

yoangp, 2019-01-20:
Document à présenter aux futurs partenaires du projet cette année..

Première ébauche:
https://www.dropbox.com/s/dvubsifsm5z65bd/Projet%20Otomate%202019.docx?dl=0

Quels sont vos commentaires?? :-)

jayserf, 2019-01-20:
> Assigned #65 <https://github.com/LoupHC/controleur-CAPE/issues/65> to
> @jayserf <https://github.com/jayserf>.

this is truly awesome!

i agree

LoupHC, 2019-01-21:
Wow c'est vraiment nice ça Yoan! Ce soir je commence à fatiguer mais je te donne mes commentaires demain c'est promis!

LoupHC, 2019-01-22:
J'ai ajouté quelques commentaires et écrit un petit paragraphe!

LoupHC, 2019-01-26:
Super! Moi aussi je pensais travailler plus là-dessus demain.
Je me met dispo toute la journée pour en discutter et travailler là-dessus, donc à l'heure qui te convient.
On pourrait faire ça sur hangout pour pouvoir travailler en parallèle.

yoangp, 2019-01-26:
Ok cool!

jayserf, 2019-01-29:
Looking great guys.

Did you compete and send to Yan yet?

I'm happy to read and comment also if it can help.

Good time-line and budget!

LoupHC, 2019-01-31:
Bon! J'ai fini de mettre au propre le budget et le document explicatif!
J'attend votre relecture avant d'envoyer ça (tu remarqueras Yoan, j'ai enlevé certaines parties qui étaient un peu redondantes)
https://www.dropbox.com/ow/msft/edit/home/autoconstruction_private/autoconstruction%202019/projet-Otomate/Projet%20Otomate%202019.docx
https://www.dropbox.com/ow/msft/edit/home/autoconstruction_private/autoconstruction%202019/projet-Otomate/Plan%20d'action%202019.xlsx