LoupHC/controleur-CAPE #66: PCB - Choix Écran

yoangp, 2019-01-20:
Allo,

J'ouvre cet issue pour toutes les discussions d'écrans.

Il y a différentes options de design qu'on peut suivre:
1. On réutilise au max le design précédant avec l'idée que dans une version future on mettra de l'effort pour avoir un interface cool via smartphone. Ça permet de pas perdre trop de temps sur cet élément.
2. On utilise un écran méga cheap, dans l'idée avec l'idée que dans une version future on mettra de l'effort pour avoir un interface cool via smartphone. Le but ultime étant de réduire au max le coût du produit à terme car l'écran et le clavier principal sera l'interface via browser web sur smartphone que tout le monde a déjà dans sa poche.
3. On met un écran touchscreen pour avoir juste 1 élément à intégrer. Désavantages: C'est une pièce qui coûte quand-même cher ( et possiblement pas moins cher que clavier et écran actuels). Pas super avec doigts méga sales de fermiers. Long à implenter.

Dans tous les cas je trouve qu'installer l'écran sur le PCB c'est pas compliqué, ça garde tout ensemble et avec un convercle de boîte transparent il est facile de voir que tout va bien dans la serre sans avoir à ouvrir la boîte.

D'autres options? Sinon, je vote pour l'option 1.

yoangp, 2019-01-20:
Copie du message de Jason sur la question de l'écran:

for the screen, I like it on the board. we can get a box with a window
perhaps...

i found some cool options that can fit more info perhaps and give loup a
sweet programming challenge!

https://www.aliexpress.com/store/product/3-5-480-320-HMI-Intelligent-Smart-USART-UART-Serial-TFT-LCD-Module-Display-Panel-for/900905_32910475160.html?spm=2114.12010612.8148356.1.38ab6decxGz04p

https://www.aliexpress.com/store/product/Real-OLED-Display-2-42-128-64-12864-SPI-I2C-IIC-Graphic-LCD-Module-Screen-LCM/900905_32920778860.html?spm=2114.12010612.8148356.1.61122ea9oDOHbP

found the number pad!
https://www.aliexpress.com/store/product/4x4-Matrix-Keyboard-Keypad-Module-Use-Key-PIC-AVR-Stamp-Sml-4-4-Plastic-Keys-Switch/2130127_32831963425.html?spm=2114.12010615.8148356.15.64ca172cI3QwR5

LoupHC, 2019-01-21:
Pour la question du couvercle transparent... faut penser aux UV aussi éventuellement... Comment on protégerait le reste du circuit?
Une solution à laquelle j'ai pensé c'est un print custom sur le boitier qui masquerait tout sauf le rectangle de l'écran,,, Ça peut-être un sticker qu'on colle à l'intérieur ou un print directement sur la boite... faudrait soumissionner polycase pour voir combien ça couterait...
Sinon on pourrait penser à une coquille qu'on pourrait visser par dessus le circuit à l'intérieur de la boîte. Jason est-ce que tu nous prêtes ta super découpeuse laser pour faire ça??? Steplait steplait steplait!

yoangp, 2019-03-10:
Pour LCD: https://fr.aliexpress.com/item/Special-promotions-lcd-20x4-LCD-module-Blue-screen-IIC-I2C-2004-5V-LCD-for-arduino/1621301904.html?spm=2114.search0204.3.37.48dc1af1PrP1iK&transAbTest=ae803_3&ws_ab_test=searchweb0_0%2Csearchweb201602_2_10152_10151_10065_10344_10068_10130_10342_10547_10343_10340_10548_10341_10084_10083_10618_10139_10307_5711213_10313_10059_10534_100031_10103_10627_10626_10624_10623_10622_10621_10620_5711313_5722413%2Csearchweb201603_25%2CppcSwitch_5_ppcChannel&algo_pvid=d2552fce-8736-4457-b640-e49cb00a3e77&priceBeautifyAB=0&algo_expid=d2552fce-8736-4457-b640-e49cb00a3e77-5

Trous dans les coins de 1/8in de diamètre, faut trouver spacer avec vis qui fitte = tread vis #4-40 impérial ou M3 métrique. Distance entre les trous : 55mm centre-à-centre hauteur et 93mm centre-à-centre largeur.

Hauteur des spacers pour boite YH 100804:
-Hauteur de l'écran au dessus du space = 11mm
-Hauteur dans la boite à partir de dessous le pcb = 102mm
-Hauteur du PCB = environ 2mm

Donc, hauteur des spacers = max 102mm-11mm-2mm = 89mm ou 3,5in. Trop cher chez Digikey.

yoangp, 2019-03-10:
Datasheet LCD choisi (même dimension que module chinois) Numéro de lcd : 2004A, semble standard...

LoupHC, 2019-03-11:
À date le plus haut que je trouve c'est du 4cm : 
https://fr.aliexpress.com/item/50-pcs-lot-blanc-Nylon-standoff-M2-m2-5-M3-M4-femelle-femelle-nylon-en-plastique/32870711165.html?spm=a2g0w.search0104.3.176.59ff3597Hujcbl&ws_ab_test=searchweb0_0%2Csearchweb201602_1_10065_10068_10130_10547_319_10059_10884_317_10548_10887_10696_321_322_10084_453_10083_454_10103_10618_10307_537_536_10902%2Csearchweb201603_51%2CppcSwitch_0&algo_pvid=71309e62-0c1a-4bc4-aee4-95733168310b&algo_expid=71309e62-0c1a-4bc4-aee4-95733168310b-24

LoupHC, 2019-03-11:
standoff en alluminium (80cm) :https://fr.aliexpress.com/item/Nouveau-10-PCS-Standoff-Ronde-En-Aluminium-Alliage-Entretoises-M3-95mm-60mm-80mm-90mm-100mm-pour/32834465655.html?spm=a2g0w.search0104.3.1.4d825170q1L47t&ws_ab_test=searchweb0_0%2Csearchweb201602_1_10065_10068_10130_10547_319_10059_10884_317_10548_10887_10696_321_322_10084_453_10083_454_10103_10618_10307_537_536_10902%2Csearchweb201603_51%2CppcSwitch_0&algo_pvid=503429eb-f290-47d6-8548-657ed4e3b110&algo_expid=503429eb-f290-47d6-8548-657ed4e3b110-0

yoangp, 2019-03-11:
> standoff en alluminium (80cm) :https://fr.aliexpress.com/
> item/Nouveau-10-PCS-Standoff-Ronde-En-Aluminium-Alliage-
> Entretoises-M3-95mm-60mm-80mm-90mm-100mm-pour/32834465655.
> html?spm=a2g0w.search0104.3.1.4d825170q1L47t&ws_ab_test=searchweb0_0%
> 2Csearchweb201602_1_10065_10068_10130_10547_319_10059_
> 10884_317_10548_10887_10696_321_322_10084_453_10083_454_
> 10103_10618_10307_537_536_10902%2Csearchweb201603_51%
> 2CppcSwitch_0&algo_pvid=503429eb-f290-47d6-8548-657ed4e3b110&algo_expid=
> 503429eb-f290-47d6-8548-657ed4e3b110-0

Ceux en alu de 80cm semblent intéressants et pas trop cher si achetés sur
Aliexpress. Je vote pour de quoi dans ce genre là!