LoupHC/controleur-CAPE #67: PCB - Choix clavier

yoangp, 2019-01-20:
Discussion sur le clavier à utiliser...

Options de clavier:
1. Garder le même clavier que l'Otomate 2018 afin de limiter les développements, dans le but qu'il y aura l'écran tactile d'un smartphone dans une prochaine version.
2. Trouver un autre modèle de clavier. Je sais Loup que tu as proposé un autre clavier, mais c'est quoi la raison exacte? Moins cher, plus robuste?
3. Touch screen sur le pcb, voir l'issue sur l'écran pour plus de détails.

Il faut aussi penser au positionnement du clavier, options:
A) Sur le pcb, à un endroit qui limite le risque qu'un fermier casse de quoi avec ses gros doigts sales ;-) Désavantages: Il faut ouvrir la boîte pour faire une action ou changer un paramètre, ça peut arriver quand même souvent. Peser fort sur le clavier peut engendrer des fissures dans les soudures ou les traces du pcb à long terme, donc défectuositée difficile à identifier et à réparer.
B) Au boût d'un cable. Donc possible de fixer le clavier dans le fond de la boîte ou sur le dessus de la boîte si le dessus est transparent. Idée de montage: Si on réutilisait le clavier de l'otomate 2018, il est déjà sur un flat cable et autocollant. Il serait possible d'installer le pcb directement sur le couvercle de la boîte (à l'en vers pour voir l'écran au travers du couvercle transparent) et de faire passer le flat cable dans un petit trou pour l'avoir accessible sur le dessus de la boîte (similaire à ce qui a été fait lors du dernier atelier).

Des commentairs, autres options? Sinon, je vote option 1B

LoupHC, 2019-01-21:
Mon concern avec le clavier membrane depuis le dernier atelier... c'est que j'ai peur que l'adhesif décolle après un certain temps... me semble qu'un beau clavier de téléphone ça serait swell et pas mal plus robuste  (pour 4$ au lieu de 2$)!

Un problème aussi si on passe le flat cable par le couvercle, c'est que ça crée un stress mécanique à chaque fois qu'on ouvre le couvercle. Il est assez court faque c'est pas évident de passer ses doigts pour le débrancher... c'est comme ça que le premier contrôleur de la clé des champs était fait... c'était pas génial...

Je penche plus aussi pour l'option B. Le moins de taponnage sur le circuit le mieux! Et ça diminue les chances de faire entre de l'humidité même si ça risque inmanquablement d'arriver... 

Si on s'inspire de ce modèle avec les connecteurs RJ45 entre le circuit est le clavier, ça donne un form factor intéressant je trouve  :
https://www.amazon.com/gp/product/B06XGSYP1C?pf_rd_p=1581d9f4-062f-453c-b69e-0f3e00ba2652&pf_rd_r=AWH4F8RB61G8WFX39ZDH
On a déjà utilisé un connecteur RJ45 étanche dans le dernier modèle, on pourrait l'utiliser dans l'optique de sortir le clavier du boitier : 
https://www.aliexpress.com/item/Free-Shipping-M20-8-core-waterproof-RJ45-network-connector-can-fixed-on-boxes-10pcs-lot/1923243933.html?spm=2114.search0104.3.1.5e884e55C8LYuy&ws_ab_test=searchweb0_0,searchweb201602_1_10065_10068_10130_10547_319_317_10548_10696_453_10084_454_10083_433_10618_431_10139_10307_537_536_5736111_5736011_10059_10884_10887_100031_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=87646215-6df9-4205-87a4-1bd0736307a6-3&algo_pvid=87646215-6df9-4205-87a4-1bd0736307a6
Après faudrait encapsuler tout ça dans une petite boite , qui peut-être petite et cheap, même pas étanche... c'est sur que ça fait plusieurs étapes d'assemblage, et c'est ce qu'on essaie d'éviter au max... c'est à bien y penser... mais c'est quand même moins d'étape que le boitier d'interface qu'on a fait au dernier atelier et pour une résultat final bien meilleur à mon avis!

jayserf, 2019-01-29:
> Assigned #67 <https://github.com/LoupHC/controleur-CAPE/issues/67> to
> @jayserf <https://github.com/jayserf>.

Très d'accord avec cela.

Nice idea.

yoangp, 2019-02-16:
Concernant le clavier pour la nouvelle version de l'Otomate, est-ce que vous pouvez me trouver un keypad plug-and-play? J'ai fouillé sur Aliexpress, Digikey, etc. Je n'ai rien trouvé d'intéressant. Le modèle que tu proposais Loup avec connecteur RJ45 n'est pas dispo en ce moment sur Aliexpress. Perso, je mettrait vraiment l'ancien clavier sur une plaquette de plastique protectrice devant le PCB, à l'intérieur de la boîte, comme on a déjà discuté (ou sinon sur le devant de la boîte avec une rallonge cable plat 8 fils). Pour avoir utilisé le clavier de 2018 pendant un an, j'ai trouvé qu'il faisait vraiment la job et que les touches répondaient bien.

yoangp, 2019-02-17:
Décision finale: Utiliser ancien clavier matrix 4x4 de 2018 avec chip mcp23008 sur le pcb. Le clavier sera dans la boîte sur la plaque de plastique ou à l'extérieur de la boîte avec une extrension du flat cable.

closed 2019-02-17 by yoangp

LoupHC, 2019-02-17:
Attends attends!
Je ne trancherais pas aussi vite.
Si on change la chip I/O expander du clavier je dois changer ma librairie et ça peut potentiellement être long et pénible.
Est-ce qu'on pourrait pas plutôt utiliser cette chip : 
http://www.ti.com/lit/ds/symlink/pcf8574.pdf
C'est celle qu'utilise le module jaune associé au clavier.

LoupHC, 2019-02-17:
En l'installant de façon à utiliser l'addresse 0x26.

yoangp, 2019-02-17:
Ok pour utiliser la chip pcf8574, Digikey en on en masse de dispo en format dip.