LoupHC/controleur-CAPE #68: PCB - Switchs manuelles en cas de bug de l'arduino

yoangp, 2019-01-20:
L'idée est la suivante:

- En conditions normales, toutes les actions du fermier pour bypasser le mode automatique passent quand même par l'arduino. Il est possible via le clavier et l'écran de forcer l'ouverture ou la fermeture des roll-ups/ventilation/fournaise pendant x temps ou x% ou indéfiniment. Loup, t'as déjà codé tout ça, donc super!

- Sur le PCB, il y a des switchs pour passer en mode réellement manuel, mais c'est utilisé juste si n'importe quoi dans le circuiterie lâche (sauf les relais et l'alimentation 24VDC) ou que l'arduino capote.

Assez straight forward à implanter ces switchs sr le pcb, je propose le même montage que ce qui avait été fait pour les switchs manuelles de l'Otomate 2018, sauf que ça sera sur le pcb avec des switchs style: https://www.digikey.ca/product-detail/en/nidec-copal-electronics/AS2E-2M-10-Z/563-1588-ND/5086845

Des commentaires? Sinon je crois que ça c'est réglé.

LoupHC, 2019-01-21:
Est-ce qu'on peut quand même avoir des mini-switchs à bascule? C'est ben plus le fun à actionner! ;)

Au niveau du code je suis rendu à l'intégrer dans l'interface, ça j'en ai pour un bout encore, mais ça sera fait je vous jure!

yoangp, 2019-01-21:
Switch a bascules 2x plus cher, ça vaut tu la peine? Surtout que si les
switchs coutent pas cher, on peut aussi en mettre pour la ventilation,
fournaise, etc.

jayserf, 2019-01-29:
> Assigned #68 <https://github.com/LoupHC/controleur-CAPE/issues/68> to
> @jayserf <https://github.com/jayserf>.

No comment....

Hard decision.

External switch means more wiring.
Internal means opening the box.

But if the interface is easier, people will actually use the override and
learn how to use the Keypad vs just defaulting to the big external switches
(observation from my farm)