LoupHC/controleur-CAPE #69: PCB - SMT ou through hole

yoangp, 2019-01-20:
En gros, pour continuer sur le commentaire de Jason, est-ce qu'on y va pour full SMT ou full through hole?

Pour moi, ça se résume à:
SMT:
- Moins cher (il faudrait évaluer la différence en $ mais surement assez important)
- Moins durable, mais le l'arduino mega est déjà full SMT..
- Moins facile à réparer, mais quand même possiblement réparable selon Jason.
- Fait plus de sens si on veut mettre un coating de protection contre l'humidité, mais ça veut dire qu'il n'y a pas de possibilité de réparer la chose.

Through hole:
- Plus cher
- Plus durable
- Il reste potentiellement juste les capteurs de courant pour les moteurs de roll-up qui sont difficile à trouver pas SMT ou sur socket intéressant.
- Plus facile à raparer si quelque chose brûle, mais difficile à dire si dans la vrai vie c'est vraiment applicable. On avait avancé l'idée que si quelqu'un a un board de brûlé, il pourrait nous l'envoyer en échange d'un board neuf et nous pourrions faire la réparation et revendre comme réusiné. Par contre, est-ce que ce scénario tient la route économiquement? Si on offre un service genre garanti an an sur les pièces, alors là on perd un peu moins d'argent si on peut réparer ce qui est brisé et le revendre. Si la pièce n'est pas garantie et qu'il faut que le fermier achète un nouveau board plein prix, c'est quoi son incitatif pour nous envoyer le board brulé? On peut bien payer le shipping du board brûlé en se disant qu'on va récupérer ce coût sur la revente du board réparé, mais si on voit que le board est full scrap lorsqu'on le reçoit (tombé dans l'eau, foudroyé à mort, etc.), alors il y a rien à faire avec.

Des commentaire? De mon côté, je pense qu'il serait pertinant de faire une petite recherche rapido pour estimer la différence de coût entre les deux. Si SMT c'est moitié moins cher et que ça réduit significativement le coût, ben là il va falloir bien y penser.

LoupHC, 2019-01-21:
Personellement j'ai jamais été foutu de souder autre chose que des résistances surface mount sur un board... et encore en sacrant pas mal... Les transistors je les brûlent 1 fois sur 2, les ICs on en parle pas j'ai hurlé en essayant de souder moi-même les sondes de courant sur un board une fois!

L'arduino MEGA est SMT c'est vrai! Mais au moins il est produit en série et on peut se dire que virtuellement on aura toujours la possibilité d'en trouver un autre quelque part pour le remplacer. Notre board non!
Aussi si je comprend bien, du moment qu'on a des conducteurs qui doivent rester exposés, comme nos connecteurs à vis par exemple, on doit laisser tomber pour le conformal coating? C'est pas possible de traiter seulement une partie du circuit si j'ai bien compris... à valider!

Perso je vote vraiment vraiment vraiment plus pour le through-hole! Surtout pour la possibilité de pouvoir rentrer les chips dans des sockets! Même pas de soudure impliquée! Réparer de l'électronique au tournevis, c'est pas génial ça?
Limite, je vois ça comme un argument de vente!
Si on offre une garantie (ce qui serait aussi potentiellement un excellent argument de vente!), faudrait bugetter le coût réel pour nous d'appliquer la garantie. Si on estime que réviser un board et le remettre en état ça nous prend tant de temps, faut facturer ce temps dans le prix de base, ou sinon le facturer au moment du switch. Genre on offre le service d'échanger le board pour un montant X, qui inclut le temps nécessaire pour le réparer et le shipping, plus un pourcentage de la valeur du board neuf pour compenser la dépréciation ou quelques pertes totales de temps à autres. Si les gens peuvent payer 200$ pour une remise à neuf, c'est quand même moins cher qu'un board neuf, ça couvre nos frais et ça évite de jeter de l'électronique inutilement!

Au niveau du coût, si on se fie aux couts de l'an passé, le prix des 3 boards+assemblage+pièces c'était environ 150$. 150$ sur 650$ de hardware et sur 900$ de prix de vente.
Si avec du surface mount on réduite le prix de 30%, ça fait quand même juste 5% de diminution du prix de vente total...

Je pense que j'ai un conflit personnel à work out avec le SMT... ;)

LoupHC, 2019-01-21:
Aussi un autre argument pour la durabilité, l'espacement entre les pattes... apparemment plus deux pôles négatif/positif sont proche, plus il peut y avoir de la corrosion par électrolyse sur le circuit. Avec du TH on a au moins 3x la distance entre les pattes qu'une pièce SMT!

yoangp, 2019-01-22:
J'approuves tes arguments Loup, allons y avec through hole. Il reste donc juste la question des capteurs de courant, les seules pièces SMT. Je sais que tu avais regardé pas mal de pièces sans trouver un autre format de chip intéressant. Je vais quand même regarder un peu de mon bord et si je trouve rien, ben on reste avec ce qu'on a.

yoangp, 2019-01-22:
Hmm, les shield de courant pour arduino sont tous fait avec le même genre de capteur de courant SMT, sinon through hole c'est pour du gros courant avec un fils qui passe dans un trou - pas pour nous. Le socket à soic-8 a l'air fou, gros et coute cher. Bon, je pense que je penche pour laisser le même chip et me casser la tête sur autre chose. 

En passant, est-ce que tu as trouvé le courant maximal que peut prendre la chip? J'ai vu que le voltage de sortie va saturer à 5V si le courant est plus haut que 5A, mais j'aimerais savoir si la chip va exploser à 10A.. D'un autre côté, si on continue avec le power supply Meanwell 24V 6,25A, le risque est pas trop grand que ça arrive. Le power supply est fait pour jamais dépasser 100% de l'ampérage max.

LoupHC, 2019-01-22:
C'est assez rugged comme chip d'après ce que j'ai vu. C'est un capteur magnétique donc il n'y a aucun contact entre le courant mesuré et le signal envoyé, avec une isolation de 2400V! Ça veut aussi dire que le output peut pas dépasser la tension d'allimentation de la sonde, donc on est safe d'être toujours dans un range acceptable.
Il n'est pas indiqué de tension maximale dans le datasheet mais il y a une configuration de la chip qui permet de mesurer jusqu'à 50A donc on est au moins tranquille jusque là!

Une alternative si on veut pas souder la chip en surface mount c'est de mettre un header femelle pour emboiter un module comme celui-ci : 
https://www.aliexpress.com/item/NEW-30A-Hall-Current-Sensor-Module-ACS712-model-30A-In-stock-high-quality/754724680.html?spm=2114.search0104.3.1.36d279a5M6BCM5&ws_ab_test=searchweb0_0,searchweb201602_1_10065_10068_10130_10547_319_317_10548_10696_453_10084_454_10083_433_10618_431_10139_10307_537_536_10059_10884_10887_100031_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=a728d383-2808-48c5-b98d-996d60fa318f-0&algo_pvid=a728d383-2808-48c5-b98d-996d60fa318f
On pourrait le faire tenir au board avec un ti-wrap pour qu'il tienne plus fermement en place. 
Qu'est-ce que t'en penses?

LoupHC, 2019-01-22:
Shit oui mets-en! C'est à voir s'ils peuvent le faire, si oui, on devrait s'en commander une couple de modules d'extra!!

yoangp, 2019-01-22:
C'est certain qu'on pourrait avoir des modules d'extra, méga facile à réparer par la suite. Et fini le SMT sur le board principal :-)

Jason t'en pense quoi de l'idée que j'ai lancée pour les pcb soic-8 à dip-8 avec PCBway??