LoupHC/controleur-CAPE #70: PCB - Protection surtension et inversion de polarité

yoangp, 2019-01-20:
Loup, est-ce que tu considère que cette problématique est suffisament robuste sur le modèle Otomate 2018?

Voici mes idées:

- Vu qu'on pense avoir des DC-DC converters 24V à 12V et 24V à 5V, il serait surement suffisant d'avoir la protection d'inversion de polarité sur l'entrée 24VDC.

- Sur l'Otomate 2018, la protection de surtension était sur le 5VDC.. Si la protection de surtension est sur le 24VDC, on protège du même coup le 12V et le 5V right? C'est sur que si le DC-DC converter 24V à 5V lâche sans raison et fait brûler l'arduino avec c'est poche. D'un autre côté, si la protection est juste sur le 5V et qu'une surtension passe sur le 12V ça risque de faire sauter le capteur de pluie Hydreon qui marche sur 12V. On pourrait mettre un DC-DC converter 5V à 12V pour garder le 12V protégé, mais ça pas l'air d'un super bon design, toujours mieux de mettre des DC-DC converter vers une tension plus basse pour réduire la demande en courant, ça fait aussi ben des converters en série. Ou bien on protège aussi le 12V séparément, mais c'est ben du trouble pour pas grand chose il me semble.

Donc, la question c'est : Est-ce qu'on déplace la protection sur le 24VDC ou on la garde sur le 5VDC ou on fait d'autre chose? J'ai pas encore mon idée de faite la-dessus, je dirais de garder la protection sur le 5V, mais ça m'achale que le 12V ne soit pas protégé et qu'on puisse faire sauter le capteur de pluie à 59$US c'est plus cher que l'Arduino!

N.B. En passant, j'ai vu que ça prend souvent une charge minimum de 10% de la capacité avec un DC-DC converter. Si c'est un risque pour nous, on pourra toujours mettre une résistance comme load minimale, ça a l'air d'être ça qui est utilisé habituellement pour des petits DC-DC converters. Certains DC-DC converter de qualité n'ont pas de minimum load.

LoupHC, 2019-01-21:
Pour l'inversion de polarité, en magasinant un peu on peut probablement trouver un régulateur de voltage qui a ce genre de protection intégrée. En même temps une diode schotky c'est pas cher payé et ça donne une tranquilité d'esprit supplémentaire.

Pour les surtensions je dirais que pour le arduino on doit être très prudent parcequ'on alimente le mega en 5V direct, et que son voltage maximal est de 5.5V. Le capteur de courant a probablement un régulateur de voltage intégré (je pourrai vérifier, j'en ai un chez nous!) donc on peut peut-être être un peu plus permissif au niveau des fluctuations de tension momentanées.
Dans les deux cas ça prend des condensateurs!

Ce que Daniel m'expliquait aussi, c'est que les circuits simples pour protéger des surtensions sont pas très précis, et peuvent fluctuer selon la température. Malheureusement on peut pas vraiment avoir une protection assez précise pour permettre 5V de passer, mais bloquer à 5.5V, de façon fiable tout le temps. À quelque part, on est un peu obligé de faire confiance au régulateur de voltage. Par contre! Pour limiter le risque on peut protéger le régulateur de voltage en amont. Comme ça on est certain que le régulateur reste à l'intérieur de ses specs en tout temps.
Faque quant à moi je mettrais seulement une protection entre l'alim 24V et les régulateurs 12v et 5v. Genre un gros varistor avec une fuse ou un crowbar circuit, comme le circuit déjà designé pour la dernière version.

Une de mes préoccupations aussi, c'est comment on va protéger les régulateurs des back-emf des moteurs... J'avais lu là-dessus un peu et ça disait que les régulateurs mettent un peu de temps à s'ajuster quand ya une variation rapide de tension et ça peut se répercuter sur la ligne de l'autre côté.
Dans le cas du arduino, vu qu'on a .5V de jeu, c'est risqué non?
On peut probablement faire un circuit filtre pour prévenir le problème mais ça implique des calculs qui dépassent mes compétences... et qu'il faudrait valider à l'oscilloscope!
Est-ce qu'on ferait pas mieux de rentrer le 12V dans le circuit par un transfo séparé, et de le rabaisser à 5V pour le contrôleur?

Le 24V des moteurs et des relais peut-être un peu plus freestyle quant à moi c'est moins téteux que des processeurs! Peut-être juste les varistors comme sur les drivers de moteurs feraient la job.
Avec un limiteur de courant pour pas griller les transistors!

Pour la charge minimum, c'est quoi l'effet contraire sinon?

yoangp, 2019-02-17:
Bon, pour la protection d'inversion de polarité sur le 24V, je trouve que le plus simple c'est de garder le design avec diode vu qu'on va descendre tout le reste sur le 12V. Mais je garde l'idée d'un P-Mosfet au cas ou...