LoupHC/controleur-CAPE #71: PCB - Provenance des pièces (Aliexpress/Digikey/Autre...)

yoangp, 2019-01-21:
Est-ce que ça vaut la peine d'acheter des pièces du PCB en chine puis les shipper au fabricant de PCB en chine plutôt que de dire au fabricant de pcb d'achter eux-même les pièces chez Digikey?

Aliexpress c'est sûr que c'est moins cher, mais surement aussi moins fiable.
Digikey ça me semble moins de trouble si on a pas à gérer l'approvisionnement des pièces au fabriquant de pcb.

LoupHC, 2019-01-21:
Pcbway peuvent acheter des pièces localement en Chine, on avait fait ça pour les gros connecteurs verts entre autre.
Je pense que tout ce qui est quincaillerie générique (switchs, fuse holder, connecteurs, résistances, sockets, condensateurs) ça vaut la peine de l'achèter en Chine.
Pour les pièces particulières (diodes, régulateurs de voltage, varistor) on devrait passer par digikey

yoangp, 2019-01-21:
Comment ça avait fonctionné avec PCBway? Est-ce que vous avez dit à PCBway:
telles pièces doivent venir de Aliexpress et telles autres pièces de
Digikey et arrangez-vous pour faire venir tout ça vous même à votre
compagnie pour l'assemblage?

LoupHC, 2019-01-22:
De souvenir on leur a juste envoyé les liens des pièces à acheter sur Aliexpress, et ils se sont occupé du reste!