LoupHC/controleur-CAPE #73: SOFTWARE - interface

LoupHC, 2019-01-23:
Je suis en plein dans le pimpage de l'interface et j'ai besoin de vos conseils et idées!

Jusqu'à maintenant, j'ai réussi à faire deux améliorations majeures : 
- contrôlers les rollups, fans,  fournaises en mode semi-automatique (avec ou sans timer) à partir de l'interface
- changer la décimale lorsqu'on modifie un paramètre, en appuyant sur la touche '0'. Ex, si je modifie la température cible, avec les flèches je monte/descend de 0.1 à chaque touche, si j'appuis sur '0', je passe à +-1, si j'appuis encore sur '0', je passe à +-10 et si j'appuis encore sur '0' je retombe à +-0.1.
Ça  fait vraiment du bien!

- Prochainement, je pensais intégrer dans le menu "Action", des raccourcis vers les menus auxquels on accède généralement avec le code à 4 chiffres. Au lieu de se rappeller du code, les gens pourraient garder 9 menus en mémoire et les activer d'un seul clique. 
Les menus peuvent modifier un paramètre ou lancer une action selon le cas, ils sont tous référencés par un numéro à 4 chiffres.
Trouvez vous que c'est une bonne idée? Ça donnerait une possibilité de customization intéressante je trouve!

Avez-vous d'autres suggestions d'améliorations à apporter à l'interface?

yoangp, 2019-01-23:
Tu travailles fort sur l'interface dis donc! :-)

Commentaires (certains ont l'air négatifs, mais c'est voulu constructifs!):
Personnellement, je trouvais que 1dégré d'incrément c'était suffisant comme précision et step de température. On sait qu'on veut rester dans un range 5degrésC à 30dégrésC environ. Des steps de 10 degrés c'est pas assez précis et ça oblige de savoir qu'il faut peser sur 0 pour switcher de mode pour aller mettre des incréments de 1degrésC. De plus, la température à différents endroits de la serre peut varier de plusieurs degrésC selon la position du soleil, l'entrée d'air extérieur, la distribution égale ou pas du chauffage dans la serre, etc. Des steps de 0.1 degré c'est bien au-delà de la précision réelle qu'il est possible d'atteindre.

Par contre, des steps de 0.5 degrés ça pourrait permettre d'introduire un certain ordre de priorités autour d'une température cible. Ex. 24.5degréC démarrage des fans pour entrer de l'air extérieur, 25degréC début d'ouverture des roll-ups.

Je trouve aussi que ça ajoute un niveau de complexité pour le fermier cette possibilité de swticher d'un mode à l'autre. Si tu veux tout de même garder ce feature. je proposes que tu mettes 0.5, 1 et 5 plutôt que 0.1,1 et 10.

Je sais que je t'en ai déjà parlé, mais quand j'ai utilisé le contrôleur d'irrigation HydroRain, j'ai trouvé ça merveilleux à quand point la programmation était simple à utliser et rapide à apprendre. Je pense qu'il y a juste 4 boutons et une roulette de sélection de la chose qu'on désire configurer. C'est sur qu'il y aura plus de paramètres à programmer sur le contrôleur, mais je te renvoie tout de même le "Simple is beautiful" ;-) Je peux te garantir qu'il y a une tonne de fermiers qui ont zéro intérêt à passer plus que 5min à apprendre comment le contrôleur marche. À la limite, il faudrait leur mettre dans les mains sans guide d'utilisation et qu'ils soient capable de le programme tout de suite juste de même.

Pour les raccourcis, je comprends ton idée, mais dans la réalité si les paramètres profond du contrôleur sont modifiés 5à10 fois dans l'année c'est probablement le gros max. Ensuite, ils vont pitonner des choses, mais ça va se limiter à changer quelques températures ou heures.

Moi, je pense qu'il faut que tu regarde ton interface avec des yeux neuf. Demande à quelqu'un qui connais rien de ton controleur de changer la consigne de température des roll-ups, ou le % d'ouverture des roll-up du stage 1, etc. Elle (la personne je veux dire, pas nécessairement une fille :-)) va surement te dire qu'elle a besoin d'informations pour faire l'action. Si elle trouve ça trop compliqué c'est qu'il y a un problème de design.

J'écris à voix haute...
Un redesign de l'interface pour rendre cela mega easy.
- Faire en sorte que toute action demande un code de 3 chiffres (ça donne quand même 999 posibilités)
- Ensuite tout ce qui peut être fait sur le contrôleur dépend d'un code.
Voici comment c'est présenté: Un des bouton est bien identifé en rouge avec 
"1. Peser ici pour entrer code"
À côté, une liste bien claire de tout les codes, séparés selon que ça soit une action ou un paramètre à changer ou bien un autre écran d'infos à voir. (L'été passé j'étais souvent mélangé entre les bontons a,b,c,d quand ça fesait un mois que j'avais pas utilisé l'interface)
Ensuite ça peut être quelque chose du genre comme 3 boutons: "Augmenter", "Diminuer", "Entrer".
Si t'as plus qu'un paramètre à entrer dans l'écran en question, le "Entrer" te mène au prochain paramètre. Le paramètre en cours de modif est clignotant. (une grosse partie fonctionne déjà comme cela)
Donc, 4 bontons bien identifiés et une liste de paramètres.

En tout cas, c'est pas obligé d'être de même, mais l'idée c'est qu'il faut que ça soit méga simple, le moins de fonctions d'interfaces différentes à apprendre...

C'est quoi ton feeling là dessus?
On s'en rejase ;-)

LoupHC, 2019-01-23:
Autre question existentielle : 
- les timepoints, est-ce que tu trouves ça trop abstrait comme concept? Est-ce qu'on devrait pas les catégoriser plus tôt comme : 
T-nuit
T-dif
T-jour
T-pré-nuit
J'essaye de me débarrasser du plus d'abstractions possibles...

LoupHC, 2019-01-23:
Aussi, j'ai la bonne nouvelle à t'annoncer que j'ai maintenant du code qui me permet de faire presque tout ce que je veux avec l'interface! 
Des architectures de menus, de l'affichage, de la modification de paramètres numériques, choisir parmis plusieurs nametags...

LoupHC, 2019-01-23:
Je garderais le code à 4 chiffres par contre, il y a une logique derrière : 
premier chiffre = type de paramètre (timepoint, rollup, fan, fournaise...)
deuxième chiffre = #d'item
troisième et 4e = paramètre à modifier
Vu la longueur de mon fichier de code, faut que je garde de l'ordre là-dedans au minimum

Par contre les codes plus usuels on peut les numéroter 0001, 0002, 0003

LoupHC, 2019-01-27:
Tu vas me trouver tannant... j'ai encore changé d'avis!
Je pensais à ton idée d'utiliser un code à 2 chiffres, qui pourrait ensuite appeller un code à 4 chiffres défini dans le code.
Pis là j'étais comme... pfff ça va être du trouble!
Et là je me suis dit, tant qu'à ajouter une couche de programme, au lieu de créer un numéro qui appelle un numéro, pourquoi pas créer un nametag qui appelle un numéro, là ça prend plus son sens selon moi!
Autrement dit, d'un point de vue de programmation, c'est pas plus compliqué pour moi de faire un menu de texte qui appelle le numéro de programmation nécessaire. 
Ex. 
-set Rollup1 (affiche successivement les paramètres du rollup1)
-set Rollup2 (idem pour rollup2)
-set Device1 (idem pour device1)
-set Device2 (idem pour device2)
-set T nuit (idem pour timepoint 3-4)
-set Tjour (idem pour timepoint 1-2)
-Overrides (modifie les paramètres des overrides (autre menu)
-#code (entrer un code pour modifier un paramètre isolément)

Boom... changement de paradigme!
Qu'est-ce que t'en penses?

jayserf, 2019-01-29:
Very elegant and easy to understand!

Nice one!

LoupHC, 2019-01-29:
Les résultats de ma petite expérience comportementale (merci à Ève qui a tenu jusqu'au bout malgré l'envie de rage quit plusieurs fois) : 
- les menus déroulants, c'est en général plus intuitif que les codes
- elle aimait mieux se faire guider à travers les paramètres un à la suite de l'autre plutôt qu'entrer uns par uns les codes correspondant aux paramètres à modifier. 
- remarque intéressante : on utilise le clavier numérique seulement pour rentrer les codes, mais pas pour modifier les valeurs de paramètres, c'est contre-intuitif! À la place, on devrait utiliser les flèches pour naviguer à travers les chiffres (pour changer de décimale) et utiliser le clavier pour modifier la valeur de la décimale sélectionnée. C'est plus intuitif selon elle que le concept d'appuyer sur zero pour modifier la décimale.

yoangp, 2019-01-29:
Intéressant, une autre option serait d'entrer des valeurs comme au guichet
automatique, Exemple:
"Entrez la valeur : 00.0"
Taper le chiffre 2 donne:
"Entrez la valeur : 00.2"
Taper le chiffre 5 donne:
"Entrez la valeur : 02.5"
Taper le chiffre 0 donne:
"Entrez la valeur : 25.0"

LoupHC, 2019-01-30:
En tout cas guys, je m'embarque dans quelque chose de gros! Je vais vraiment avoir besoin de votre avis là-dessus!
J'ai fait un diagramme fonctionnel de la future interface du contrôleur, je pense enfin avoir rien oublié!!
Ça va être une belle interface à menu classique, mais la plus épurée possible (j'ai essayé).
J'ai essayé de m'imaginer le workflow que j'avais l'an passé quand je gossais sur le iGrow, ça donne ça : 
https://www.draw.io/#G1ctQ59Fbii_Kc-W_18nohk9aDOSgipiFA

C'est un gros chunk mais vos commentaires sont plus que bienvenues! Dites le moi si ya des bouttes pas clairs. Les embranchements représentent différentes options sous-jacente à un menu.

LoupHC, 2019-01-30:
Idéalement il serait possible de naviguer dans ces menus-là dans les deux sens, c'est à dire en descendant ou en remontant les chaines de menus. À la fin d'une chaîne on reviendrait au menu principal ou au front screen (avec la température, les % d'ouverture...)

LoupHC, 2019-01-31:
Au cas ou ça bug sur draw.io
[image]

LoupHC, 2019-01-31:
En fait, je dirais que je suis pas mal rendu là. J'évalue qu'il me reste un bon 40h pour me rendre ou je veux...
Mon système d'overrides fonctionne comme je veux (pour la pluie, l'humidité, un override programmable à heure fixe par sortie, mode semi-automatique), pour l'override vent ça vaut pas la peine que je m'y mette avant d'avoir un setup physique où le tester.
Il me reste les fusibles électroniques à programmer aussi, mais sans setup physique c'est difficile aussi.
C'était pas mal mes gros chantiers pour cet hiver!

yoangp, 2019-02-02:
Pour commencer je vais essayer de mieux comprendre toutes tes fonctions.

-Dans Ajustement, C'est quoi l'ajustement pour T24h? Est-ce que tu peux mettre une valeur cible de T24h?
-Pour les ajutements météo, ça serais-tu possible de faire remonter cela d'un niveau car c'est de quoi qu'il va falloir changer quotidiennement, donc idéalement le moins de pitonage pour changer la valeur. Ex: "Menu Principal"/"Ajustements Météo"/ Soleil ou Mi-Mi ou Nuage. L'affaire c'est la T24h qui est problématique à garder dans ce menu. T'as tu une sugestion? 
- Rien à dire pour "Config menu" a part que le nom n'est pas super intuitif. Mais j'ai pas vraiment de meilleur nom. "Paramètres fixes"? "Config initiale". Oui, "Config initiale" je pense que c'est bon, c'est clair qu'il faut le faire une fois au début puis il n'est plus nécessaire de s'en occuper.
- C'est quoi le menu "Capteurs", est-ce juste pour spécifier quel capteur est branché? Si oui, ça pourrait aller avec le menu config initiales...
- C'est quoi T jour/Dif?
- C'est quoi T nuit/Pré-nuit
- Ça serait super d'avoir une façon de désactiver tous les paramètres qu'on veut pas utiliser, afin de ne pas avoir à se casser la tête pour changer la température lorsque le contrôleur n'est pas installé sur une serre à tomates. Ex.: Moi mon tunnel multiusage, je l'utilisait juste avec les roll-ups avec une seule temp réchauffement et une seule temp refroidissement. C'est basic, mais ça fait déjà mille fois mieux la job que d'ouvrir à la main et c'est pas compliqué à gérer. Mes suggestions; Soit que dans le menu config initial, il soit possible d'activer ou de désactiver certaines fonctions (ex.: Pré-nuit: ON/OFF). Sinon, dans le menu en tant que tel. Ex de l'écran lorsque tu arrives dans le menu Pré-nuit:
"Congif Pré-nuit"
"ON/OFF"
"Clock/Sunrise"
En fait, je viens de voir que tu as mis un paramètre Activer/Désactiver sur les Roll-ups, fan,fournaise. Dans le fond, il serait peut-être possible de mettre ça sur les menus : Tjour/Dif, Tnuit/Edit, Tjour/Pré-nuit.
- Ça se peux-tu qu'il devrait y avoir Sunset dans tes settings et pas juste des Sunrise ;-)
- Temps de rampe, si c'est pas utilisé, ça pourrait être par défaut à zéro... C'est dans le fancy ce paramètre!
- Paramètres roll-ups: Je sais pas si tu as pensé à ça mais ça prend une façon de clearer la jupette qui ne fait pas parti des paliers de refroidissement. Je pense que le plus simple, serait d'ajouter le temps pour passer la jupette comme premier paramètre (j'ai déjà ajouté une boîte pour ça dans ton schéma draw.io). Plus de détails: Lorsque la serre est complètement fermée, ça ne peut pas être flush avec la jupette, il faut que ça descende plus bas que le haut de la jupette pour s'assurer que le vent et le froid ne rentre pas trop dans la serre. Donc, si le roll-up descend plus bas que le bas de la jupette, ça fausse le premier palier d'ouverture si on en tient pas compte. Avec tes paramètres de 2018, il était possible de faire avec mais ce n'était pas propre. Dans le cas de mon tunnel, j'avais 4 paliers de 25% et le premier palier ouvrait juste une mini ouverture en haut de la jupette, le palier 2 ouvrait donc presque 1/3 de la surface d'ouverture. Tout ça pour dire, que ça serait vraiment moins mélangeant avec un paramètre spécial pour la jupette. Est-ce que c'est assez clair comme explication? On peut s'en parler au téléphone aussi..
- Pour les Enable-diasble des overrides, je mettrait ça en premier. Ex.: "Pluie"/"Enable-Disable"/"%". Si on sélectionne "Disable", il faudrait sortir tout de suite du menu, sinon c'est mélangeant de spécifier un % sur un paramètre disablé.
- A quand endroit tu spécifie "Sonde temp extérieur connecté/pas connecté" ? J'ai vu le commentaire, mais dans quel menu ça se retrouve exactement?
- Dans "Force close"/"Délai(min)", est-ce dans ce menu que tu peux mettre un temps infini quand il faut mettre à off de manière permanante?
- Il faudrait un menu pour indiquer s'il y a ou non un anémomètre (et lequel), un capteur de pluie (et lequel), etc. Mais en fait, peut-être le fait que l'override soit "Enable ou diasble" fait la job. En tout cas, ça fait bizarre que le capteur de temp intérieur soit à part. Est-ce que tu l'as mis à part car il est utilisé partout (roll-ups, fan, fournaise)?

Bon, je pense que tu as quelques piste de réflexion ;-)

LoupHC, 2019-02-02:
Nice!!! wow ça c'est du feedback! Thanks man!
- mettre Ajustements en haut de liste : bonne idée, c'est vrai que c'est la fonction la plus utilisée 
- l'ajustement T24h, c'est pour moduler la T24h vers le haut ou vers le bas en fonction de la charge en fruit. Ça modifie tous les paramètres de températures de tous les timepoints de +-X*C
- Hmm effectivement, c'est le genre de trucs qui tu définis juste une fois.
- Le Dif, c'est une période de chauffage avant le lever du soleil pour réveiller les plants avant que la chaleur commence à monter brusquement. Ça permet de contrôler la longueur des entre-noeuds, ça limite la condensation, etc. On pourra activer/désactiver cette fonction.
- La pré-nuit c'est une technique pour augmenter le calibre des fruits en faisant dropper drastiquement la température au coucher du soleil, et de remonter graduellement par la suite.On pourra aussi activer/désactiver cette fonction.
- Tu penses que ça pourrait être pertinent d'utiliser juste la T jour? Je comprend pas trop comment tu utilisais la température de réchauffement avec les rollups... seulement la température de refroidissement et l'hysteresis est pris en compte normalement. 
Faudrait juste pas que ĉa soit possible de désactiver la T nuit ET la T jour par accident, parceque là ça ferait crasher le programme...
- lol oui, blame it on the copy/paste
- pour le temps de rampe, c'est pas aussi évident quand tu automatises seulement des rollups, mais avec des fournaises et de la ventilation, sans ramping, ça peut créer des changements de température assez drastique et ça stress les plants. Mais c'est vrai qu'on peut le mettre a zéro par défaut. 
- la manière la plus straight forward selon moi serait de jouer sur les pourcentages. Ex. tu pourrais avoir ton premier palier à 40% puis le deuxième à 60%, puis 80%, puis 100%.
- enable des overrides : bonne idée de le mettre en haut!
- temp extérieure : nulle part vraiment, en même temps on pourrait inclure un système de reconnaissance software qui détecte automatiquement si une sonde d'extra est connectée.
- par défaut la durée du bloquage est infinie, en jouant avec les flèches on peut faire varier la vitesse entre 1min et 24h.
- la seule raison pour ça c'est que le code est compatible avec deux sondes différentes pour la température intérieure, mais seulement une sonde de pluie (pour l'instant). S'il y a un seul choix supporté, j'irais avec la reconnaissance automatique.
En tout cas merci pour les commentaires je vais dormir là-dessus!

LoupHC, 2019-02-22:
Nouvelle utilisation du clavier à bouton (idée) : 
- On supprime les boutons ABCD
- On garde seulement les boutons 1234567890*#
Le bouton * devient le bouton SCROLL et permet de passer d'un menu à l'autre.
Le bouton # devient le bouton SELECT et permet de sélectionner un menu ou de confirmer une nouvelle valeur
Les numéros servent à entrer la valeur du paramètre, guichet automatique style, ou à appeller des fonctions toutes faites ex : 
1 permet d'appeller la fonction d'override manuel du premier rollup
2 permet d'appeller la fonction d'override manuel du 2e rollup
3 permet d'appeller la l'override du 3e équipement
...
9 permet de sélectionner la météo du jour
0 permet d'ajuster la T24h
Trouvez-vous que ça a de l'allure?