LoupHC/controleur-CAPE #74: PCB - Alimentation 120V ventilation

yoangp, 2019-01-25:
Quand même pas mal comme patente:

https://www.amazon.ca/Iot-Relay-Enclosed-High-power-Raspberry/dp/B00WV7GMA2/ref=sr_1_1?ie=UTF8&qid=1548384118&sr=8-1&keywords=powerswitch+tail

Pas acheter de boîte, pas de montage à faire et super easy à installer.

Mais c'est juste 12 ampères. Y'a peut-être d'autres modèles.

yoangp, 2019-01-25:
Les plus gros fan jet que j'ai trouvé sur le net jusqu'à maintenant c'est avec un moteur de 1HP (16A) pour des grosses fan de 30 pouces. Le max serait des fan jet de 1/2HP (10A) avec un Powerswitch tail par Fan jet.

LoupHC, 2019-01-25:
Nice!
C'est bête n'empêche... c'est les même specs que les relais dans le boitier...

LoupHC, 2019-01-25:
Sinon avec un module comme ça on peut avoir une sortie à 20A, faut faire des connexions mais au moins ya pas de conducteur exposé: 
 https://www.amazon.com/dp/B01B60RZPA/ref=sspa_dk_detail_1?psc=1

jayserf, 2019-01-29:
That's a sweet find. Both of them.

I guess this is to not have 120 going to relays in our control box?

yoangp, 2019-01-29:
That’s right, no 120v in the box mean less danger for farmers without
electrical knowledge and maybe easier later to certify if we what to sell
the thing elsewhere than Québec.

jayserf, 2019-01-30:
Got it. Cool