LoupHC/controleur-CAPE #75: Sondage pour suivi atelier 2018

yoangp, 2019-01-31:
Allo Loup,

J'ai rajouté une option pour donner de l'aide via un webinaire, plutôt que visite sur place ou atelier de mise à niveau (j'ai ajouté cette colonne dans le fichier Excel). J'ai déjà 3 personnes qui trouvent que le webinaire serait le mieux.

J'ai rejoint 5 personnes,mais ça va prendre encore un petit bout de temps pour faire le tour de tout le monde. J'ai laissé plusieurs messages vocals que je suis pas sur d'avoir un retour d'appel. Je vais peut-être m'essayer pas courriel si j'ai pas de feedback téléphonique. Au moins pour savoir le minimum par rapport aux questions du sondage qu'on a.

Finalement, je vais mettre mes données dans le fichier en ligne lorsque j'aurai contacté tout le monde, car j'ai des problèmes d'internet et Excel en ligne me joue des tours...

LoupHC, 2019-01-31:
Hein oui bonne idée!
Ça serait moins de trouble à organiser qu'un atelier. Et un bonne alternative à une visite personnelle pour quelqu'un d'un peu patenteux!

LoupHC, 2019-01-31:
De mon côté pas encore commencé les appels, je m'y met demain!

yoangp, 2019-02-01:
J'ai commencé à envoyer des courriels à ceux qui répondent pas au téléphone, voici le message que j'ai envoyé si tu veux faire de même de ton côté:

Allo xx et yy!

Loup et moi faisons un petit suivi des contrôleurs de serre de la CAPÉ autoconstruits en 2018 pour voir si un coup de main peut être donné au besoin et nous enligner sur la prochaine version du contrôleur que nous sommes en train de concevoir.

J'aimerais bien vous parler au téléphone environ 10min (819-434-6801 en tout temps), mais sinon une réponse par courriel suffira. 
Voici nos quelques questions:

1. Est-ce que votre contrôleur de la CAPÉ est installé?
Si oui, que contrôlez vous? Roll-ups, fournaise, fan?
Si non, pourquoi?

2. Avez-vous fait un update du code ou modifié le circuit pour corriger les petits bugs suite à l'atelier?

3. Votre objectif en 2019 avec votre contrôleur? Installer dessus les roll-ups / fournaise / ventilateurs?

4. Pour utiliser le plein potentiel de votre contrôleur, seriez-vous intéressé à l'une des options suivantes?
- Une visite en personne payante - mais à coût moindre qu'on technicien de serre
- Un atelier de mise à niveau cet hiver à un endroit au Québec à définir
- Un webinaire de mise à niveau cet hiver pour avoir un complément d'information pour l'installation des roll-ups, fournaise, ventilateur et voir comment faire vous même les quelques corrections aux bugs.

5. Autre commentaire? Bug sur le contrôleur installé / amélioration à faire / etc.

C'est tout! Merci d'avance pour votre temps!

Yoan et Loup

yoangp, 2019-02-02:
J'ai mis mes infos recueillies jusqu'à mainteant dans le fichier en ligne.

yoangp, 2019-02-08:
Je pense que j'ai fait le tour de mon côté pour le sondage, il y a 3 personnes que je n'ai pas pu avec de réponse (Arnaud Mayet, Olivier Painchaud et Vazouk Poisson). Pour l'instant, au total on a juste 6 boîtes qui ont été mis en fonction; Lève tôt, Tessa, Nues mains, des libellules, rouge tomate (en cours d'installation), Tournesol (juste drivers utilisés). La plupart de celles par intallées c'est qu'ils n'ont pas encore de serre. Il faudrait ajouter le contrôleur de Carya et le mien dans la liste pour faire le tour. Je vais faire une ligne pour moi et est-ce que tu peux remplir celle de Carya?