LoupHC/controleur-CAPE #76: PCB - Protection physique du circuit

yoangp, 2019-02-01:
Détails à venir

yoangp, 2019-02-01:
Peut-être au lieu de mettre un collantsur le panneau de plastique protecteur, le texte pourrait être gravé au laser? Ça se fait-tu facilement Jason avec une machine comme vous avez à votre travail? Est-ce que la puissance du laser peut être ajustée pour par passer au travers du plastique?

LoupHC, 2019-02-01:
Avec des overrides switchs comme celles-là : 
https://www.aliexpress.com/item/Promotion-5-Pcs-AC-ON-OFF-SPDT-2-Position-Latching-Toggle-Switch-6A-125V/32813324272.html?spm=2114.search0104.3.16.3e294c7e5lXxmF&ws_ab_test=searchweb0_0%2Csearchweb201602_1_10065_10130_10068_10547_319_317_10548_10696_453_10084_454_10083_433_10618_10139_10307_537_536_10902_10059_10884_10887_321_322_10103%2Csearchweb201603_51%2CppcSwitch_0&algo_pvid=371541e5-9381-4e86-bb65-069384ba223b&algo_expid=371541e5-9381-4e86-bb65-069384ba223b-2
On pourrait facilement faire passer la switch de l'autre côté de la protection en plastique et même la bolter après!
Ça donnerait quelque chose de très solide je pense!

yoangp, 2019-02-01:
On a juste à les mettre dans les coins, pas besoin de spacers ;-)

yoangp, 2019-02-17:
Remplacer fusible sur le 5VDC par [circuit breaker 1.5A ](https://www.digikey.ca/product-detail/en/e-t-a/1410-L110-L1F1-S01-1.5A/302-1409-ND/3987396)

Exemple varistor vraiment solide:
[Varistor 100joules](https://www.digikey.ca/product-detail/en/littelfuse-inc/V20E20P/F11291-ND/4285775)

Ça fait environ 20$ de plus pour par avoir à s'occuper d'un fusible brulé. En même temps, j'ai comme l'impression que le DC-DC converter 5V ne sera pas en forme après qu'il ait eu 33V en sortie, donc s'il faut changer le régulateur, un fusible de plus ou de moins, bof. Laissons le fusible pour l'instant je dirais.

yoangp, 2019-02-17:
LM317:  https://electronics.stackexchange.com/questions/266054/lm317-explosion

Ça l'air que ça ne fait pas une super protection de court-circuit d'utiliser un lm317 tout seul car il va dissiper toute la puissance. Il faudrait y ajouter une autre affaire pour réellement déconnecter l'alimentation lorsque la sortie est shortée.. Mais ça peut faire une bonne première ligne de défense le temps qu'une fuse saute genre.. Qu'est ce que tu en penses?

LoupHC, 2019-02-18:
Je suis d'accord que si on ajuste de 24v à 12v et de 12v à 5v... ça fait pas mal de circuiterie à traverser avant d'arriver au contrôleur. Et les modules dc-dc que tu m'as montré ont une protection intégrée contre les courts-circuits non?
Peut-être que c'est suffisant!

yoangp, 2019-02-18:
Low dropout regulator pour doubler protection sur le 5V, tout en limitant la baisse de la tension:
https://www.digikey.com/product-detail/en/texas-instruments/TPS7150QP/296-8050-5-ND/373253