LoupHC/controleur-CAPE #77: Webinaire hiver 2019

yoangp, 2019-02-02:
But: Passer de l'info sur comment réparer bugs du contrôleur 2018 et comment installer fan et fournaise + autres questions des gens.

À faire:
- Trouver un contrôleur pas modifié et pas loin de chez Loup ou Yoan pour le webinaire
- Voir quel système de webinaire on utilise. Ça marche pas pire avec le compte Via de la CAPÉ, mais il faut utiliser des téléphones mobiles car son via l'internet terrible. Est-ce que ça va bugger avec la vidéo en plus.. à tester. Sinon, Jason a suggéré un autre système (Zoom Video Communications). Mais ça change pas que si les gens ont une connection internet pourrite, ils sont mieux avec téléphone pour poser des questions lors du webinaire.
- Faire un powerpoint pour le contenu
- Faire une pratique du webinaire (possible à distance??)
- Une fois que c'est plus avancé, afficher une date

LoupHC, 2019-02-05:
Ok de quoi on devrait parler pendant le webinaire en question : 
- problématique du power supply qui lâche : comment le corriger
- comment renforcer les connexions
- comment tester la continuité
- comment brancher une fournaise (terminaux contacteur)
- comment brancher un équipement sur le 120VAC (parler de calibre de fil, specs de relais)
- utilisation de l'IDE pour télécharger le programme
- autre chose?

yoangp, 2019-02-05:
Il serait bien de parler aussi de toit ouvrant. À part Coeur de légumes, il y a Simon Leduc qui veut installer un toit ouvrant similaire. À Tourne-sol aussi ils ont un toit ouvrant sur leur nouvelle serre, même si Reid utilise un iGrow... Tout ça pour dire qu'il y a quand même quelques grosses serre à toit ouvrant. De son côté, Simon, voulait 2 roll-ups, 1 fan, 1 fournaise et 1 toit ouvrant (sur 2 sorties). Est-ce que ton code s'adapte facilement pour faire ça?? En fait, je viens juste de penser que le toit ouvrant pourrait être en parallèle avec les côtés ouvrants. Donc pas de sorties additionnelles nécessaires. Est-ce comme ça que ça fonctionne pour Coeur de légume?

Est-ce qu'on était d'accord sur le version du code qui devrait être mise à jour dans les contrôleur de 2018? Comme j'avais dis, je pense qu'il faudrait y aller pour le minimum de changements pour éviter d'autres nouveaux bugs, je vois ceci qui est essentiel:
- Correction bug lorsques plusieurs paliers au même pourcentage
- Côtés qui s'ouvrent pas en même temps pour réduire les drops de tensions au maximum (exemple, Yan a encore des problèmes de roll-ups stallés sporadiquement, même avec le bypass de la résistance) surtout qu'il y a des gens pour qui les moteurs tirent pas mal plus de courant et le power supply de 6A est limite lorsque 2 moteurs partent en même temps.
- Je mettrais incrément de 0.5 degré C, plutot que 0.1 degré C lors de l'ajustement de la température avec le clavier, sans changer rien d'autre au code pour ça.

Pour l'instant, c'est ce qui me vient en tête...

LoupHC, 2019-02-05:
C'est tout à fait possible de mettre deux sorties en parallèles sur le même output! Les transistors sont vraiment overkills. Par contre les temps d'ouverture/fermeture seront identiques.
Pour l'instant le code de base permet de gérer 4 devices : 2 rollups et 2x fan OU fournaise
Il y a une sortie d'extra mais pas prise en charge avec l'interface, par contre c'est possible de l'utiliser directement dans le code.
Recommandation de Reid à ce sujet par contre : 
- un toit ouvrant, c'est un gros risque! C'est beaucoup plus sécuritaire si on peut inclure un anémomètre. certains voudront peut-être attendre qu'on ait implémenté cette fonctionnalité avant de le mettre en fonction.

Les corrections dont tu parles sont assez simples (pour l'affaire des pourcentages je vais voir si c'est rétrocompatible). Pourrais tu m'envoyer ta version du code pour l'histoire des deux rollups en alternance?

yoangp, 2019-02-08:
J'ai mis un premier draft de la présentation powerpoint sur Dropbox.
Je vais voir comment organiser la question du filmage du webinaire. Est-ce que tu peux t'occuper spécifiquement de la préparation du montage qui sera requis pour le webinaire vu que tu auras une vrai boîte avec toi? J'aimerais qu'on s'arrange pour que je n'aie pas à aller 2 fois à Victo en une semaine. Si on pense manquer de temps de préparation, je peux venir vendredi avant l'atelier en fin de journée, mais à voir selon notre niveau de confiance. Qu'est-ce que tu en dis?

yoangp, 2019-02-08:
Pour l'affaire de toit ouvrant, est-ce qu'il y a quelque chose de facile à faire pour le code d'ici l'atelier, ou ça sera obligatoirement une version du code à tester dans le futur?

LoupHC, 2019-02-08:
Tu peux venir vendredi soir ou samedi matin tôt comme tu préfères, le vendredi je vais revenir vers 7h par contre vu que je vais travailler sur les contrôleurs de Coeur de légumes pendant la journée.
Mais oui j'aimerais bien qu'on se fasse une petite générale avant, le live c'est intimidant quand même hehe.
Je pense qu'on peut monter la présentation à distance d'ici là, on s'appelle si y faut. J'ai fait une couple de schemas de filage sur draw-io je te les enverrai! Je jette un coup d'oeil à l'instant sur le draft!

Pour le toit ouvrant, c'est possible dès maintenant sans modifications dans le code si on branche les rollups en simultané. Je vais faire un schema pour ça aussi je sais que Coeur de légumes en veulent un aussi!

yoangp, 2019-02-12:
Salut Loup, pour la rencontre pour le webinaire, je vais venir samedi matin,  je serai chez toi entre 8h30am et 9h30am (à moins que la météo à court terme soit vraiment pourrie pour samedi, dans ce cas ça serait vendredi soir).

yoangp, 2019-02-12:
Salut Loup,

Est-ce que tu sais qui est responsable du compte Via de la CAPÉ? Il faudrait s'assurer de ne pas entrer en conflis avec une autre activité pour le webinaire de samedi. Je voulais justement tester des affaires avec la compte et j'étais en plein dans une conférence! Il doit y avoir un calendrier d'utilisation à quelque part...

LoupHC, 2019-02-12:
Je nous ai réservé la plateforme!

yoangp, 2019-02-12:
Comment as-tu fait cela? Il faudrait que je la réserve pour mes tests de
filmage..

yoangp, 2019-02-14:
J'ai fini mon travail sur le fichier dans Dropbox qui est maintenant nommé : Otomate V1 - Contrôleur de serre - Feuille des fonctions et paramètres.xlsx. J'ai mis aussi une version pdf, mais gène toi pas si tu veux qu'on change encore quelque chose dans les descriptions.

yoangp, 2019-02-15:
Allo Loup,

Je pense avoir fait mon tour de la présentation Powerpoint.
- j'ai mis en bas à gauche qui parle à chaque slide
- j'ai regroupé les sujets qui seront filmés ensemble vers la fin afin de faire une seule interruption dans le webinaire
-J'ai ajouté des infos par-ci par là dans la présentation

Regarde tout cela si tu peux et on s'en reparles en personne ce samedi am!
Je compte être chez toi tel que prévu entre 8h30am et 9h30am.

LoupHC, 2019-02-24:
Le lien du webinaire : https://www.youtube.com/watch?v=BzGnGOEXX9k&t=1599s

closed 2019-02-24 by LoupHC