LoupHC/controleur-CAPE #80: PCB - Override des moteurs

LoupHC, 2019-02-17:
J'ai essayé de trouver un nouveau circuit de switchs simple pour dealer avec le nouveau setup de relais ce matin.
J'ai rien trouvé à date... à part avec des switch dp4t et autres affaires weirds...
Je me demande à quel point on se donne du trouble pour enlever un relais du circuit. Celui qu'on a, on sait qu'il marche, et qu'il est facile à utiliser.
Et puis l'enjeu de perte de puissance, c'est assez spécifique au problème de peak current des moteurs, au pire si les gens veulent utiliser les overrides, on inclue une note qui indique "start motors one at the time" ou quelque chose du genre.

S'il y a un problème avec le contrôleur, l'alternative la plus efficace, c'est de mettre les deux moteurs à "ouvrir" et de partir faire ses affaires pendant la journée, plutôt que de mettre un moteur à "ouvrir", attendre 2 minutes, et mettre le 2e à "ouvrir".
Aussi, c'est pertinent de pouvoir bloquer un moteur pendant que l'autre roule en automatique, si en le bloquant on bloque aussi le relais d'inversion de polarité on n'est pas plus avancé. 
Je dis qu'on garde le circuit à 2 relais par moteur et le même filage de switchs.
Qu'est-ce que t'en penses?

LoupHC, 2019-02-22:
Qu'est-ce que t'en penses?

yoangp, 2019-02-22:
Oupss, j'avais pas vu passé celui là..

Oui, on peut faire comme tu dis. J'avais comme dans l'idée de mettre un 8ième relai juste comme spare si un des 7 autres lâche. Pis là je me suis dit que si on remet le circuit des moteurs à 4 relais, mais qu'il est possible de faire fonctionner le tout avec juste 3, ben un jumper pourrait permettre de retirer un des relais et s'en servir comme back-up.. Mais ça a l'air trop compliqué il me semble. C'est certain qu'on peut mettre aussi un 9ième relai de spare sur le circuit, mais là là, il va y en avoir du relai là-dessus!!

En bref, je peux commencer par remettre le circuit des moteurs sur 4 relais.