LoupHC/controleur-CAPE #82: Capteur de lumière

yoangp, 2019-02-17:
Capteur de lumière proposé par Loup:

https://www.aliexpress.com/item/Light-transmitter-sensor-0-10V-0-5V-4-20mA-RS485-200000Lux-65535Lux-industrial-light-intensity-illumination/32860403692.html?spm=2114.10010108.1000014.1.62062b0aNN9fJl&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.13338.107646.000000000000000&scm_id=1007.13338.107646.000000000000000&scm-url=1007.13338.107646.000000000000000&pvid=c67f15a6-feac-4b98-9bad-15a3bcdaa4c6

Loup, on se donne tu le trouble d'une connection en RS485 pour avoir une donnée de luminosité précise digitale en lux, ou on fait simple avec entrée analog 0-5V?

LoupHC, 2019-02-17:
analogue!

LoupHC, 2019-06-01:
Les équipements sur le 12V (sonde de pluie + sonde de lumière) font pas tellement de bruit électronique, je ne pense pas que ce soit problématique de relier les grounds ensemble!
Dans le pire des cas, la zener qui protège les ios va clamper les surtensions induites sur la ligne, qu'est-ce qui pourrait arriver d'autre?

LoupHC, 2019-06-01:
Premières données enregistrées du capteur de lumière!!!
[image]