LoupHC/controleur-CAPE #86: PCB - Anémomètre

yoangp, 2019-02-18:
- Anémomètre [Sparkfun](https://www.sparkfun.com/datasheets/Sensors/Weather/Weather%20Sensor%20Assembly..pdf?_ga=2.169973664.1013965956.1550428730-272579373.1543974264)

ou

- Anémomètre [Davis](http://cactus.io/hookups/weather/anemometer/davis/hookup-arduino-to-davis-anemometer)

Les deux fonctionnent avec 2 pins de switch qui s'activent à chaque tour complet de l'anémomètre
Sur entrée digitale avec pull-up de 4.7k au 5V
La différence c'est que la vitesse du vent des deux anémomètres n'est pas pareille pour une rotation. À paramètrer dans le code selon l'anémomètre installé.

LoupHC, 2019-03-11:
J'aurais le goût de tester ces deux capteurs météo aussi tant qu'à faire des tests : 
https://abra-electronics.com/sensors/sensors-temperature-en/sens-ef10121-wind-speed-sensor-anemometer-three-aluminium-cups.html?sl=fr
Et
https://abra-electronics.com/educational-kits-trainers/educational-kits/arduino-educational-kits/sen0170-anemometer-kit-0-5v.html

Celui à 35$ est ridiculement pas cher et simple et celui à 100$ est en alluminium, potentiellement plus durable que ceux en plastique de Davis Instruments et Sparkfun

yoangp, 2019-03-12:
> en/sens-ef10121-wind-speed-sensor-anemometer-three-
> aluminium-cups.html?sl=fr
> educational-kits/arduino-educational-kits/sen0170-anemometer-kit-0-5v.html

Ok, Je jetterai un coup d’oeil pour voir si ces sondes seront
automatiquement compatibles..