LoupHC/controleur-CAPE #87: PCB - Lora

yoangp, 2019-02-18:
Module à implanter:

https://www.aliexpress.com/item/LoRa-E32-TTL-100-long-range-UART-SX1278-433mhz-100mW-SMA-antenna-wireless-module-3km/32918242151.html?spm=2114.search0104.3.199.4d2e6a28rDlO9I&ws_ab_test=searchweb0_0,searchweb201602_1_10065_10130_10068_10547_319_317_10548_10696_453_10084_454_10083_10618_10139_10307_537_536_10902_10059_10884_10887_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=3622036b-d182-48c1-9cd1-bf168ac30310-29&algo_pvid=3622036b-d182-48c1-9cd1-bf168ac30310

yoangp, 2019-02-18:
Mode | M1 | M0 | Explanation
-- | -- | -- | --
Normal | 0 | 0 | UART and wireless channel is good to go

yoangp, 2019-03-03:
Celui-là maintenant: https://www.aliexpress.com/item/868MHz-SX1276-LoRa-100mW-Serial-Port-Wireless-Transceiver-E32-868T20D-868-MHz-IoT-Module-RF-Transmitter/32953427158.html?spm=2114.search0104.3.1.58d233ccMZhOed&ws_ab_test=searchweb0_0,searchweb201602_1_10065_10130_10068_10547_319_317_10548_10696_453_10084_454_10083_10618_10139_10307_537_536_10902_10059_10884_10887_321_322_10103,searchweb201603_60,ppcSwitch_0&algo_expid=c2755db7-56d3-40dd-a2e5-6c051e4573cf-0&algo_pvid=c2755db7-56d3-40dd-a2e5-6c051e4573cf&transAbTest=ae803_4

jayserf, 2019-04-28:
A noter que ce modele est UART. 
En essayant de prototyper avec MySensors librairie, ca semble supporter juste du SPI pour LoRa avec chipset SX1276 :-(

Alors faites attention au compatibilite hardware vs librairies radio avant d'acheter.

j'ai commandé qqs uns de cela pour faire du dev plus tard cet été...
https://www.aliexpress.com/item/SX1278-LoRa-module-Anxinke-LORA-spread-spectrum-wireless-module-RA-02/32964176718.html?spm=a2g0s.9042311.0.0.51a64c4dz8aqey

yoangp, 2019-04-28:
> Anxinke-LORA-spread-spectrum-wireless-module-RA-02/
> 32964176718.html?spm=a2g0s.9042311.0.0.51a64c4dz8aqey

Ok, merci Jason.

Bon début de saisons!

Yoan

yoangp, 2019-05-14:
Bon, ben, j'ai fait une petite revue de la partie sur le module Lora. Jason, le module que tu proposes fonctionne en 3,3V. J'ai cherché un module équivalent avec le dc-dc converter 5V à 3,3V, level shifter  pour la communication et antenne inclus sans rien trouver de bien concluant.

Si on veut avoir un prototype cette année, voici ma proposition, je laisse tomber la partie Lora sur le prochain PCB. Avant d'intégrer cela à un potentiel Otomate V3, on pourra toujours tester que ça va vraiment marcher via le shield https://www.dragino.com/products/lora/item/102-lora-shield.html. Même juste avec ce shield, il y a une grosse job à faire de programmation pour faire fonctionner tout ça dans le code de l'Otomate d'après moi.

Je me suis donné 30h pour terminer le prototype d'ici le 1er juillet, incluant finir le pcb, commander les pièces et faire le montage, et il n'y a pas de façon de faire ça simple pour le module Lora intégré. Désolé!

yoangp, 2019-05-14:
Infos pour commencer avec tout cela éventuellement: https://www.youtube.com/watch?v=duwUwXt-hs8

jayserf, 2019-05-17:
a man I love that guy, awesome videos!

Yoan, no problem! doesnt have to be integrated I can always test with a
3.3V supply / buck or something like this (which i have)
https://www.aliexpress.com/item/New-Socket-Adapter-plate-Board-for-8Pin-NRF24L01-Wireless-Transceive-module-51/32642877857.html?spm=2114.10010108.1000010.1.4b80a995HRS2NL&gps-id=pcDetailLeftTrendProduct&scm=1007.13438.125071.0&scm_id=1007.13438.125071.0&scm-url=1007.13438.125071.0&pvid=ad316363-e60e-4d9b-9d21-fd8885099073

after checking the pins...

or just get a shield, thats a nice one you found.

Amazing for a v2 proto for july 1st. also ambitious :-)
good luck and i'd be happy to look at it any time. Hope the v2 is easier
than the v1 now that we have some experience!