LoupHC/controleur-CAPE #89: PCB - Choix des connecteurs

LoupHC, 2019-02-22:
On a vu l'an passé les problèmes qui peuvent être occasionnés par des connecteurs interchangeables (l'histoire de Reid qui a brûlé une diode du driver en inversant la sortie moteur et le input).
Donc pour partir du bon pied cette année je propose qu'on revoit notre code de connecteurs!

Il y a deux types de connecteurs qui se plugguent sur Aliexpress, [ceux-là](https://www.aliexpress.com/item/10-Pcs-Pitch-3-5mm-Angle-4way-pin-Screw-Terminal-Block-Connector-Green-Color-T-Pluggable/564809969.html?spm=2114.search0104.3.214.8afd2fafAyszP6&ws_ab_test=searchweb0_0,searchweb201602_1_10065_10130_10068_10547_319_317_10548_10696_453_10084_454_10083_10618_10139_10307_537_536_10902_10059_10884_10887_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=7b20744f-8807-4b70-b67c-1f44a4596277-35&algo_pvid=7b20744f-8807-4b70-b67c-1f44a4596277 (on va les appeller type 1) et [ceux-là](https://www.aliexpress.com/item/10pairs-Solderless-docking-2EDGRK-5-08-2p3p4p5p6p7p8p9p10p-pluggable-terminal-block-2-24P-two-wire-connection/32845728247.html?spm=2114.search0104.3.15.8afd2fafAyszP6&ws_ab_test=searchweb0_0,searchweb201602_1_10065_10130_10068_10547_319_317_10548_10696_453_10084_454_10083_10618_10139_10307_537_536_10902_10059_10884_10887_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=7b20744f-8807-4b70-b67c-1f44a4596277-2&algo_pvid=7b20744f-8807-4b70-b67c-1f44a4596277 (on va les appeller type 2).

yoangp, 2019-02-22:
Intéressant!
C'est sur que si on y va pour des connecteurs 2,3,4,5,6 pins X 2 types ça fait pas mal de possibilités de connecteurs pour éviter qu'un conneteur se ramasse à être branché au mauvais endroit.

L'autre affaire, que tu dis pas directement, mais que tu as surement déjà pensé, c'est qu'on pourrait quand même dire que si des entrées ou des sorties sont similaires, elles pourraient être sur le même type de connecteur vu qu'il y a pas de danger de faire sauter de quoi. Exemple, on pourrait dire que:
- Les sorties des 3 relais extra utilisent les même types de connecteurs.
- Les sorties des 2 moteurs utilisent les même types de connecteurs.
- etc.

L'autre affaire, c'est l'hiver. Je pense qu'on en a déjà parlé, mais c'est certain que si le contrôleur n'est pas utilisé pendant 4 à 6 mois, il va avoir une meilleure durée de vie dans la maison que dans la serre. Sans compter que si quelqu'un veut reprogrammer sur contrôleur au chaud dans la maison avec son ordi (mais là c'est vrai qu'il peut juste arracher l'Arduino du PCB de l'Otomate). Est-ce qu'on se donne le trouble de trouver comment les gens peuvent rentrer leur coutrôleur?  Le but de mon affaire, c'est que probablement que les connecteurs ne peuvent pas sortie des cable gland sans défaire le fil du connecteur. Et pour moi défaire les fils des connecteurs c'est pas une bonne option. Avec l'Otomate 2018, il était possible de défaire tous les connecteurs externes, dévisser la boîte et rentrer la boîte au complet. Là, je dirait que le plus évidant serait de défaire les connecteurs du PCB, dévisser le pcb et le rentrer dans la maison. Pour moi ça serait le best.

LoupHC, 2019-02-22:
Une solution potentielle pour tout déplugger en dehors de la boite pour pas trop cher : 
- jumeler ce type de [connecteur à vis](https://www.aliexpress.com/item/IP67-Waterproof-IP67-T-Shape-Junction-Box-Connector-3-Pin-250V-15A-Cable-Wire-Gland-Sleeve/32812065944.html?spm=2114.search0104.3.246.60c0620b4cibAi&ws_ab_test=searchweb0_0,searchweb201602_1_10065_10130_10068_10547_319_317_10548_10696_453_10084_454_10083_10618_10139_10307_537_536_10902_10059_10884_10887_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=7526e077-7735-4343-b6c7-86ec470916f6-33&algo_pvid=7526e077-7735-4343-b6c7-86ec470916f6)
J'en ai commandé un qui est arrivé ce matin! Les vis sont plus grosses, c'est vraiment mieux fait que les connecteurs étanches bleus.
- avec ce type de [connecteur dépluggable](https://www.aliexpress.com/item/10-pairs-2-3Core-BLACK-Waterproof-pigtail-20cm-long-each-male-and-female-male-connector-s/32761066190.html?spm=2114.search0104.3.216.60c0620b4cibAi&ws_ab_test=searchweb0_0,searchweb201602_1_10065_10130_10068_10547_319_317_10548_10696_453_10084_454_10083_10618_10139_10307_537_536_10902_10059_10884_10887_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=7526e077-7735-4343-b6c7-86ec470916f6-29&algo_pvid=7526e077-7735-4343-b6c7-86ec470916f6)
Ça reviendrait, avec le cable gland, à genre 6$ la sortie de sonde... pas si pire. PAr contre je sais pas s'ils ont du fil avec calibre suffisant pour avoir le même principe avec le fil de moteur par exemple...

LoupHC, 2019-02-22:
Honnêtement je pense déjà que s'il faut dévisser tout le circuit, c'est déjà trop de trouble et les gens le feront pas... les connecteurs internes je vois ça plus pertinent peut-être si quelqu'un doit faire un swap de circuit parceque le siens est défectueux... ou veut changer le filage sans tout démonter son setup...

yoangp, 2019-02-24:
Pour ce qui est d'enlever le circuit de la boîte, il y a peut-être de quoi qu'on peut trouver qui permet de déclipser ou dévisser à la main le pcb de la boîte vraiment rapidement (vis papillons pour pcb?). Ça n'a pas besoin de tenir super fort, la boîte ne bouge pas.

Une option, 4 vis comme ça dans les 4 coins, on même juste deux dans les coins opposés + 2 snaps dans les autres coins:
https://www.ebay.com/p/15-Pieces-M5-X-10mm-304-Stainless-Steel-Wing-Bolt-Butterfly-Screw-Bolt/1666263379

Sinon, je suis vraiment pas chaud pour les connecteurs waterproof à vis.
- Ça coûte cher par rapport à des cable gland juste pour entrer un fils rond ordinaire dans la boîte
- C'est une étape de plus à monter
- C'est un risque de plus de mauvaise connection car on introduit un connecteur de plus sur chaque câble.

De toute façon, en y repensant bien, mon guess c'est que la majorité des gens vont laisser le circuit en place tous les hiver. Ils vont avoir un millions d'affaire plus urgentes à faire que de rentrer le circuit dans la maison. Dans ce cas, on se casse peut-être trop la tête à faire que ça soit rapide et facile à enlever. C'est certain que si le circuit est mort (on met déjà pas mal d'effort pour que ça arrive le moins souvent possible), il faut que ça soit minimalement facile de remplacer le circuit. Je pense que enlever 4 vis et déplugger tout les connecteurs du circuit est assez simple selon moi.

En bref, mon gros panchant:
- 4 vis dans les coins du pcb
- Cable gland ordinaires
- Pas de connecteurs en plus.

Qu'en dit-tu? Si tu es fortement en faveur de remettre des connecteurs waterproof à vis, parlons-en avec Reid et d'autres geeks du réseau pour avoir leur opinion d'utilisateurs.

LoupHC, 2019-02-24:
La boîte que je t'ai envoyé a seulement un point d'attache à chaque extrémité, donc effectivement deux bolts papillons ça serait parfait.
Faut voir s'il y aurait pas moyen de dévisser le circuit ET la plaquette d'acrylique en même temps. Sinon ça fait beaucoup de pièces à gérer dans ses mains en meme temps.

Pour les connecteurs étanches, effectivement je pense que le mieux sera de demander l'avis des gens en temps et lieux... c'est un bon point que tu soulèves... rendu là c'est pas ben ben plus compliqué d'enlever le circuit vs dévisser tous les connecteurs.

yoangp, 2019-02-24:
D'accord avec toi pour le dévissage et la plaque d'acrylique. On pourrait reprendre l'idée de ce montage:
[image]

Le circuit est monté sur la plaque d'acrylique et c'est la plaque d'acrylique qui est vissé à la boite. Sauf que dans notre cas, ça serait plutout l'inverse, le pcb en dessous, l'acrilique sur le dessus avec spacers. Ou bien le PCB est vissé direct sur la boîte, le plaque d'acrilique est plus petite et tient sur des spacers ailleurs sur le pcb, ou trous dans l'acrylique pour dévisser les vis qui tiennent le pcb à la boite. En tk, pleins d'options à analyser. On est vraiment rendu à se faire un dessin du montage...

yoangp, 2019-02-28:
Proposition pour connecteurs pour ne pas avoir de risque de brancher à la mauvaise place:

Description   connecteur | Pins connecteur | Types connecteur
-- | -- | --
Temp + hum | 8 | 1
Capteur Pluie   + light sensor | 7 | 1
Girouette +   anémomètre | 5 | 1
24VDC | 2 | 1
Alarme | 2 | 2
Moteurs | 4 | 1
Relai extra | 4 | 2
Relai extra | 4 | 2
Relai extra | 4 | 2

yoangp, 2019-03-03:
Mise à jour:

Description   connecteur | Pins connecteur | Types connecteur
-- | -- | --
Temp + hum | 8 | 1
Capteur de   lumière | 3 | 1
Girouette +   anémomètre + Capteur pluie | 8 | 2
24VDC | 2 | 1
Alarme | 2 | 2
Moteurs | 4 | 1
Relai extra | 4 | 2
Relai extra | 4 | 2
Relai extra | 4 | 2