LoupHC/controleur-CAPE #90: Assemblage du boitier

LoupHC, 2019-02-22:
J'ai eu un flash en regardant ce modèle spécifique du (iGrow)[https://www.greenhousemegastore.com/equip/controls-measuring-tools/environmental-controls/igrow-1400-greenhouse-controller]
Le circuit est bolté sur le couvercle!
C'est intéressant comme idée, comme ça on peut garder le clavier à proximité du circuit même en ouvrant la porte. Faut juste s'assurer que les relais seront pas trop collés aux bords de la boite pour pouvoir ouvrir la porte.

yoangp, 2019-02-22:
Intéressant,

Il y a plusieurs choses à considérer.
- L'écran finalement n'as pas à être connecté directement au pcb, car je me connecte comme sur l'Otomate 2018, c'est-à-dire via le petit connecteur I2C sur le côté qui est pas full pratique via un flat cable 4 conducteurs à un header sur le PCB. Donc l'écran peut aussi être flippé ou positionné ou qu'on veut. En tout cas, c'est pas encore final, il faut y penser pour que ça soit le plus simple.

- Admettons qu'on five le PCB au couvercle, il faut penser comment les switchs manuelles vont être accessible, si c'est via des trous dans le couvercle, le problème c'est que les switchs sont vraiment moins longues que les relais, les condo, etc. L'autre affaire, c'est le paquet de cables, s'ils sont pris dans un (des) cables gland, comment on peut ouvrir la porte? Ça fait méchamment du filage de surplus à avoir dans la boîte pour que le couvercle s'ouvre...

- Pour revenir aux switchs, c'est peut-être une bonne affaire si elles sont accessibles juste dans la boîte, ça veut dire qu'il faut pas y toucher (sauf bris majeur de l'arduino ou bug informatique ou etc.).

- Il va falloir voir c'est quoi les options de spacers pour tenir le pcb/mettre une plaquette devant, etc. Je sais pas si c'est encore une bonne idée de mettre une rail comme en 2018. Une rail c'est pratique quant il faut pouvoir connecter plein d'affaires dessus. Là peut-être jusque 4 vis dans les coins ferait l'affaire.

- Pour les cables gland, le Farmbot a fabriqué un seul cable gland sur mesure... Genre de long rectangle en 2 morceaux en caoutchouc avec tous les trous pour les cables dedans.

- Ça prendrait une genre de vue 3D de la patente je crois pour se faire une bonne idée.. Éventuellement un dessin Sketchup. Mais, en tout cas, si au moins on se fait une bonne idée de comment ça pourrait s'assembler facilement, on saura comment placer les composantes sur le pcb.

LoupHC, 2019-02-22:
- J'aurais inversé le PCB face vers l'intérieur... mais à bien y penser, ton argument des cables me convainc complètement...
- Oui je pense aussi que c'est une bonne idée les switchs à l'intérieur du boitier!
- Perso je laisserais faire le rail pour mettre des vis aux quatre coins, et donc on pourrait adapter assez flush le PCB aux dimensions de la boîte.
- Pour un autre projet j'avais trouvé des spacers sur digikey qui fittaient parfaitement avec les boites de polycase! Je pourrai les retrouver au besoin!
- le dernier détail qui me gosse, c'est le clavier... est-ce qu'on le fixe dans la boîte? C'est risqué de faire entrer autant d'humidité à n'importe quelle heure de la journée dans le circuit je trouve...
Est-ce qu'on le fixe à l'extérieur de la boîte? Ça va être chiant de gérer le flat cable du clavier si on a à ouvrir la boîte...à cogiter...

LoupHC, 2019-02-22:
Oh j'ai une idée finalement!! On instale le clavier au dessus de la charnière!! Comme ça pas de torsion du flat cable... faut juste que la boîte soit assez mince pour que ça marche...

LoupHC, 2019-02-22:
Avec une boite comme ça ça serait limiter correct : 
https://www.polycase.com/wq-50

LoupHC, 2019-02-24:
Autre idée (plus étroite par contre) :

LoupHC, 2019-02-24:
[Problème réglé : ] (https://www.aliexpress.com/item/50pcs-lot-2-54mm-Board-Spacer-1-40P-17mm-Long-Pin-Header-Straight-Single-Row-Double/32355465352.html?spm=2114.search0104.3.9.4ea44eb1szxKYt&ws_ab_test=searchweb0_0,searchweb201602_1_10065_10130_10068_10547_319_317_10548_10696_453_10084_454_10083_10618_10139_10307_537_536_10902_10059_10884_10887_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=a042329a-93a0-4798-89b1-cc57835917e5-1&algo_pvid=a042329a-93a0-4798-89b1-cc57835917e5)
On pourra mettre du heat shrink autour pour éviter d'avoir trop de métal exposé. Je dis qu'on prend la grosse boîte!

yoangp, 2019-02-24:
Je suggère qu'on essaie Sketchup pour faire le design du montage mécanique de l'Otomate.

Procédure:
1. Créer un compte Sketchup cloud-based (gratuit): 
https://www.sketchup.com/plans-and-pricing/sketchup-free
2. télécharger le modèle courant de l'Otomate disponible sur dropbox (https://www.dropbox.com/home/autoconstruction_private/AutoConstruction%202019/projet-Otomate)
Document nommé Otomate V2.skp
2. Une fois danas Sketchup, avec le menu en haut à gauche, ouvrir un dossier, puis faire "Add model", puis "Browse", puis loader le modèle dans ton compte Sketchup.
3. Une fois terminer, il y a une option "Download" pour télécharger le modèle et ainsi mettre la dernière version dans Droxbox.

Pour l'instant il y a juste un rectangle de 150mm x 300mm qui ressemble à ça:
[image]

yoangp, 2019-02-24:
Pour revenir à tes commentaires Loup, je comprends pas ou tu veux mettre le clavier et ce que tu veux faire avec les spacers pour les headers? Je comprends pas non plus sur quoi tu base la recherche du format de la boîte et qu'est-ce que ça veut dire une boite plus mince? Plus de détails stp!

yoangp, 2019-02-24:
Pour les vis sur la boite, si je comprend bien il y a 2 insets en métal à ces places là (image ci-dessous) pour des vis en métal, est-ce bien cela et c'est bien ce que tu propose pour visser le PCB?

L'affaire c'est que si le pcb est collé dans fons de la boîte, les autres protubérances de platiques pour visser des vis ailleurs dans le fond de la boîte risquent d'interférer avec les soudures et les petits bout de pins des pièces through hole sur le revers du pcb.

[image]

LoupHC, 2019-02-24:
Imagine [cette boite](https://www.polycase.com/wq-50) face à toi qui s'ouvre comme un livre.
Si le clavier est branché dans le pcb à travers le couvercle va falloir le déplugger à chaque fois qu'on ouvre la porte, c'est pas pratique.
Maintenant imagine que la charnière est située en bas de la boite, qu'on déclipse par en haut. Comme la flexion se fait dans le sens de flexion naturel du flat cable du clavier, c'est beaucoup moins pire.
Par contre comme le flat cable est assez court (8,5cm), faut pas que la boîte soit trop épaisse sinon on n'arrivera pas à brancher le clavier même le couvercle fermé.

LoupHC, 2019-02-24:
Le header-spacer permet de gagner quelque cm de de hauteur au niveau du connecteur.
À 8,5cm(flat cable) + 2 cm(spacer) ça nous donne 9,5cm de jeu

LoupHC, 2019-02-24:
J'avais pas pensé aux bosses de soudure... au pire on met un washer?

LoupHC, 2019-02-24:
Si on visse le circuit après la feuille d'acrylique et la feuille d'acrylique au boitier avec des spacers ça règle aussi le problème!

yoangp, 2019-02-24:
Ok, je comprend ton idée pour le clavier... C'est bien, mais je vais y penser encore un peu pour voir si j'ai pas d'autrs options à te proposer. Il y a toujours une rallonge de flat cable qui est une autre option.

Pour la grosseur de la boîte, ça prend une surface interne de 450cm2 environ, donc la boite WQ-50 est pas mal trop petite d'après moi. Aussi, euh, ça serait bien si le circuit pouvait fitter dans la boite de métal à 16$ d'Addison.. Pour l'option pas cher...

LoupHC, 2019-02-24:
Boitier retenu  :
https://www.polycase.com/yh-100804

yoangp, 2019-02-27:
Référence pour clearances holes sur le pcb pour fiter avec vis 10-32 aux quatre coins du boîtier retenu:
http://faculty.etsu.edu/hemphill/tap_drill.htm
= Mouting pad avec trou de 5.5mm sur le PCB dans KiCAD

yoangp, 2019-02-28:
J'ai mis à jour le dessin de coupe du PCB avec les trous correspondant à ceux de la boîte. Est-ce que tu pourras double vérifier les dimensions que j'ai utilisé? C'est un peu weird avec les coins à 45 degrés.
Voici ma feuille de calcul pour les dimensions:

  | in | mm
-- | -- | --
Longueur boîte | 10,069 | 255,7526
Longueur entre   trous | 8,25 | 209,55
Longueur entre   deux (car coins en 45 degrés ) | 9,1595 | 232,6513
Bord du pcb   par rapport au trou |   | 10
  |   |  
Largueur boîte | 8,069 | 204,9526
Largeur entre   trous | 6,25 | 158,75
Longueur entre   deux (car coins en 45 degrés ) | 7,1595 | 181,8513
Bord du pcb   par rapport au trou |   | 10
  |   |  
Longueur pour   trous du milieu |   | 104,775
  |   |  
Largeur pour   trous du milieu |   | 79,375

yoangp, 2019-03-10:
Allo!

J'espères que tout va bien pour vous en ce début de mars! Moi j'essaie d'avancer la partie physique du contrôleur petit à petit, au travers de mon nouveau travail..

Voici mes réflexions pour le montage de l'écran et du claviers
- Finalement, je ne metterais pas de plaque de plastique pour protéger le circuit. Je ferais en sorte qu'il faut ouvrir la boîte juste s'il y a un problème majeur. C'est à dire; changer un fusible ou autre pièce défectueuse, activer les switches manuelles d'urgence.
- Voici les idées de placement de l'écran et du clavier sur la boite:
Sur les images ci-dessous, il y a une option ou je considère la boite no YH-100806 qui est plus épaisse pour mettre l'écran sur le côté de la boite. Si le clavier est sur le dessus du couvercle, il faut absolument une rallonge de câble. Ça semble quasi impossible de trouver un cable male-femelle 8 pins, donc il faut une rallonge femelle-femelle + afadpteur male-male, c'est un peu poche comme montage je trouve..
-En tout cas, dites-moi quelles options vous semble préférable ou si vous avez d'autres idées. Merci! PS: Il faut regarder les images à l'endroit pour avoir la bonne orientation de la boîte, les lignes sur la page devraient être à l'horizontal, désolé pour les sens mélangés.
[image]
[image]
[image]

LoupHC, 2019-03-10:
Je vote pour le clavier et l'écran à l'avant du boitier!
Si on veut s'épargner le paneau d'acrylique, on pourrait aussi faire imprimer un grand rectangle blanc sur le couvercle, avec une fenêtre transparente pour voir le clavier, qui est fixé directement au couvercle avec des bolts auto-scellantes et des spacers.
Pour le clavier, j'irais avec une rallonge, avec en masse de jeu, vu que le couvercle serait imprimé ça serait pas trop grave même si c'est un peu moche, quelque chose comme :
- https://www.digikey.ca/product-detail/en/adafruit-industries-llc/3633/1528-2642-ND/7918051
+
- https://abra-electronics.com/interconnects/connectors/connector-housing/1907-pololu-01-254mm-crimp-connector-housing-1x8-pin-10-pack-1907-pololu.html

LoupHC, 2019-03-10:
À partir de 25 boîtes, ça revient à 10$ par boîte pour l'impression, et 15$ par boite pour le perçage sur deux faces (couvercle et dessous).

yoangp, 2019-03-10:
- Ok pour clavier sur le couvercle avec rallonge... Moi ce que j'ai trouvé de meilleur pour l'instant c'est : https://www.digikey.ca/product-detail/en/te-connectivity-amp-connectors/A9BBG-0804F/A9BBG-0804F-ND/161772 (pas de montage à faire, le cable est déjà complet), et vu que tout ces connecteurs sont femelle-femelle ça prend aussi : https://abra-electronics.com/interconnects/connectors/headers/male/1065-pololu-double-sided-straight-male-snappable-header-1-row-x-40-pins-1065-pololu.html
- Ok pour impression couvercle
- Pour l'écran je vettrait des spacer sur le pcb, donc par de trous dans le couvercle et pas de fils

T'en penses quoi?

LoupHC, 2019-03-10:
Yes ça a ben de l'allure ça!
On pourrait mettre la charnière dans le bas de la boîte, pour pas tordre le flat cable quand on ouvre le couvercle, qu'est-ce que t'en penses?
Aussi je calcule ça... avec l'extension de 10cm et le flat cable du clavier de 8cm, on est un peu juste avec la boîte de 7 po de haut. La boîte de 5po de haut(YH-100804) ferait la job à mon avis c'est la même hauteur que le boitier qu'on avait au dernier atelier.

yoangp, 2019-03-10:
Design global choisi pour l'assemblage pcb/boiter/écran/clavier:

https://drive.google.com/file/d/1wt2NEwKWlfCWdSCmHsxkXo7DZFxuQrOY/view?usp=sharing

yoangp, 2019-05-19:
Allo Loup,

Est-ce bien un sample du https://www.icotek.com/en/kel-dpz-24/ que tu as reçu? Ça va faire l'affaire pour le nombre de trous et la grosseur des trous d'après toi?

LoupHC, 2019-05-19:
yep! On a pas loin de 30 trous là dessus!

LoupHC, 2019-05-19:
J'ai juste hâte de voir si le "stress relief" est adéquat mais à part ça c'est vraiment bien fait et solide

jayserf, 2019-05-23:
wow these are super cool!

best find ever! tres hate de pas avoir de cable glad si ca marche :-)