LoupHC/controleur-CAPE #91: PCB - Prototypage

yoangp, 2019-02-22:
Moi je pense qu'on va oublier quelque chose de critique et qu'il va falloir faire faire au moins 2 pcb comme en 2018, même si on est pas en production. C'est pour cela que je proposais de montrer un pcb à la main et le tester au fur et à mesure. L'autre affaire, c'est que dans ma tête c'est quand même un gros travail d'indiquer tous les petits détails que les monteurs de PCB way devront suivre, si en plus le design risque de changer significativement, il me semble que c'Est le double de travail. Bon, peut-être que je sous estime le temps de souder un seul pcb à la main..

Sinon, côté coût: Une batch de minimum 5 pcb 30cm x 15cm pas montés c'est 79$ avant shipping. Ton prixà 330$ est-ce que c'est sans le shipping? Et pour combien de pcb montés, 3 ou 5?

La drop du prix à partir de 20 unités c'est surment qu'à partir de ce volume ils font les soudures dans une machine spéciale.. Je ne pense pas qu'on devrait se lancer dans un volume aussi grand sans encore savoir quels sont les défaut de notre design.

LoupHC, 2019-02-22:
Le 330$, c'était incluant le shipping.
moitié pour imprimer 5 pcbs
moitié pour en assembler 2
Là je me rend compte qu'on a vraiment pas les mêmes chiffres je me demande si j'ai manqué des trucs, je vais réessayer.
Je pense que c'est important de demander l'assemblage, ne serait-ce que de un. La dernière fois on a dû déssouder tous les capteurs temp/humidité parce qu'on avait skippé cette étape là. Mais c'est vrai que si on a aucune pièce surface mount, techniquement on aura aucun problème à le faire nous-même.
Au pire peut-être qu'on peut leur dire de nous envoyer les 4 pcb d'extra des quils sont fait et on en assemble chacun de notre côté?

yoangp, 2019-02-24:
Ma proposition, c'est est-ce qu'on s'occupe d'organsier la procédure
d'assemblage plus tard pendant l'été, plutôt que tout de suite quand on a
besoin d'avoir un prototype fonctionnel à installer ASAP.

Voici donc ma suggestion:
1. Faire faire 5 pcb vide shippés dès que possible.
2. On les assembles nous même et on fait des ajustement au pcb en fonction
de nos tests.
3. Si possible, on peut déjà utiliser ces pcb comme prototype à faire
fonctionner dès le début de la saison.
4. Lorsqu'on aura le temps en cours d'été, faire une nouvelle demande de
fabrication de 5 pcb, avec 1 ou 2 unités montés.
5. Recevoir les nouveaux pcb, vérifier les problèmes de montage et swaper
les nouveaux pcb sur les serres de tests.

Comme ça, on a minimum 2 runs de pcb avant de lancer la production l'hiver
prochain.
Qu'est-ce que tu en penses?

LoupHC, 2019-02-24:
Je suis d'accord avec ce plan-là moi!